MarlinFirmware/Marlin #4882: Dual X Carriage tool change

pbrgle1, 2016-09-23:
I noticed that in the current version the tool change commands always take the position of the previous tool. This is a huge problem when the tool is in park position. When changing tools the next tool will crash in the previous one because it will want to take it's last position. In a previous version of marlin this worken well since the tool change command only changed the active tool status and did not take the position of the previous tool immediately. Please fix this for the next release. :)

thinkyhead, 2016-09-24:
I'll look at this soon. Obviously it's not good to do that when the tool is parked.

pbrgle1, 2016-09-24:
Another thing. Would it be possible to add an option to invert the second X direction? Simply inverting E2 (1st free extruder slot) doesn't do anything.

thinkyhead, 2016-10-02:
> Another thing. Would it be possible to add an option to invert the second X direction?

See `INVERT_X2_VS_X_DIR` in `Configuration_adv.h`

psyvision, 2016-10-28:
I'm seeing the same issue too.  I will dig about in the code and see if I can find anything to help this along!

psyvision, 2016-10-28:
We've been working on this over at the E3D BigBox forum as that now has IDEX support.

https://forum.e3d-online.com/index.php?threads/bigbox-dual-x-carriages.2184/page-14#post-24112

It seems that you must use T0 S1 or T1 S1 to prevent the head moving.  The S1 will pass true to the tool_change function's no_move variable.  In that thread AndyVirus is using a post processing script to replace all T0/T1 commands from S3D with T0 S1/T1 S1.

thinkyhead, 2016-10-28:
We have two options for a long term solution. Either we can revert the behavior and have Marlin report the altered position, as it does with the `S1` flag, or we can fix the issue so that the first tool is parked before the second tool tries to move into position. And, we have to ensure that the second tool doesn't crash into the first tool, either by not moving the new tool's carriage at all, or by moving it only as far as it can move without crashing into the other carriage.

I believe that the new behavior —moving the selected tool into place— was implemented in response to a discussion on a feature request, and it was decided that it was the best default behavior.

Since 1.1.0 is not finalized, we can continue discuss whether this is the better option.

psyvision, 2016-11-03:
I'm not sure what's best for this.  I use IDEX in full control mode as I prefer to script tool parking etc myself as part of my tool change code rather than relying on the firmware, therefore I am happy just using the S1 flag.

However, I guess the complete solution is that if you just ask for a tool change without the S1 flag then the expectation is that the firmware should know that the old tool needs to be parked and then bring the new tool into position - as you have said.

thinkyhead, 2016-11-03:
If the auto-move-into-place behavior is meant to apply in every mode, then I agree that even though the mode might not be "auto park", then it should move the carriage out of the way _somewhere_, and in that case it makes the most sense to auto-park it, otherwise it will always be in the way.

However, it also makes good sense to behave as if `S1` was given and to not move the newly-active nozzle when it's in `DXC_FULL_CONTROL_MODE` since the implication is that all movement is meant to be GCode-controlled. The problem with this is that the GCode needs to take account of the differing positions of the two nozzles.

Hmm… A good specification would help clear this up. As I think about it more, the purpose of `DXC_FULL_CONTROL_MODE` is implied to be that you could print two completely different objects, and in that case when you switch tools you really want to leave both nozzles where they are, and let the slicer (or GCode post-processor) worry about tracking the positions of each.

So, I'm leaning towards only moving the nozzles automatically on tool-change in `DXC_AUTO_PARK_MODE` and not in any other mode.

psyvision, 2016-11-03:
I'm definitely not authoritative on the matter but that sounds sensible!

thinkyhead, 2016-11-07:
Still working out the tool-change behavior. @psyvision @pbrgle1 Please join the topic #4694. I'll be posting code for testing there.

closed 2016-11-07 by thinkyhead