MarlinFirmware/Marlin #4884: UBL Quick-Start/User's Guide

kmbecker13, 2016-09-23:
Hello!  I'm planning to put together a UBL user's guide and quick-start checklist for the less experienced Marlin users who want all the shiny bells and whistles of the latest firmware, but may not understand exactly how to make it happen.  Creating this thread to get input from the community in terms of what you think would be useful to include/go over and also correct anything i may have wrong with my below gameplan.

Here's what i'm thinking:

- Guide will assume that the user is currently using some flavor of Marlin firmware and has access to all files (not just a hex file or something similar)

- Cover how to migrate key parameters from old Marlin configuration into the new one, including pins files.  Will give a 2 minute crash course on this.

- UBL Setup (detailed below)

- Triggering UBL on future prints/adjusting z-offsets

The UBL setup workflow would essentially be:

1. Load firmware, verify you have >500 bytes local variable space free
2. `G29A P1 M` to automatically probe your bed, pay attention during first probing - watch out for any crashes
3. `G29 P2 H3 B M` to zero your mesh height and fill-in any missing probe points
4. `G26 B55 H210` for a test mesh, adjust B and H values according to your printing needs
5. `G29 P4` at various probe points, using LCD dial to adjust the mesh as needed
6. `G29 P5` to see average/std. dev of the mesh
7. `G29 S0` once you are happy with the mesh to save it to EEPROM

I have a feeling I may be missing some M502/M500's in there somewhere...Could someone give some guidance on the best usage of those?

Roxy-3D, 2016-09-23:
> I have a feeling I may be missing some M502/M500's in there somewhere...Could someone give some guidance on the best usage of those?

Probably, these should be done any time you load a new version of firmware.   But for sure they need to be done when you bring up the UBL System the first time.  The reason is there EEPROM layout is changed and you need to initialize it.    

> Load firmware, verify you have >500 bytes local variable space free.

I think we should encourage a little bit more than this.  At least initially.   How about 750 bytes of local space free?  

> G29 P2 H3 B M to zero your mesh height and fill-in any missing probe points

The manual probing is pretty painful.   But it does serve a purpose.   A reasonable compromise is to  manually probe 4 or 5 points spread out over the unprobed region.   And then using this information, just filling in the area between those manually probed points in separate steps works very well and is fast to do.   Showing how to do that would be very helpful.

> G26 B55 H210 for a test mesh, adjust B and H values according to your printing needs

At the top of the G29.cpp file are the default parameters for the command.  It might be good to show the user that and how to modify them.    Of course I have them configured for me right now.  So, if the user knew how to get them setup for them selves that probably helps them.   But once in that state, what I typically give for a G26 is:   G26 C P O2.5     It really doesn't matter where you start it, it produces a nice effect.   But one thing to stress is that user watch the first pattern being laid down and if the nozzle starts getting too close to the glass, show them how to press and hold the Encoder Wheel to aboart the operation.   And then how to go edit those locations (using a G29 P4 R at that location) to give themselves more clearance.

> G29 P4 at various probe points, using LCD dial to adjust the mesh as needed

The most important thing here is that there will probably be several areas that need to be edited.  Showing the user how to start the edit of a region and terminate the edit is important.    The user can move the nozzle to the desired area and confirm where it is positioned with a G29 M.  The nozzle's current location will be in brackets on the Map.

 I typically do a G29 P4 R and stop it when I've edited the region.   But a G29 P4 R 4 will do 4 points and auto terminate.   It might be good to show them the various ways to control that behavior.   

Also, the way it is setup, you can just say G29 P4 R and very quickly click the encoder wheel to skip points that are OK as the nozzle dances around from Mesh Point to Mesh Point.   You almost don't even need to look at the LCD Panel once you get a feel for how much the numbers change when you twist the dial.  I just watch the nozzle and if a Mesh Point is OK, I just click the Encoder Wheel.  Editing a region is very easy when you are doing things that way.   It might be good to communicate that to the users so they can understand how quickly the Mesh can be corrected and dialed in.   Please try to figure out how to get that idea across to the users!

> G29 P5 to see average/std. dev of the mesh

This was mostly done for me.   It is interesting.  But make sure they understand what ever the numbers are, they don't really matter.  If the numbers are accurate, even a very non-flat bed will get a first layer printed every time.

I actually find the G29 K # more interesting because it lets me see how two meshes compare against each other.   But that is probably an exercise best left off the beginners guide.  That starts to be interesting when you have a couple different pieces of glass dialed in.

> G29 S0 once you are happy with the mesh to save it to EEPROM

One thing to stress is the Mesh is too much work to get setup that you really want to save it.   And the other thing to stress is when you save a Mesh, you are also saving the current state of the UBL System.   So if you do a G29 A   and then a G29 S0  the UBL system will come up with Mesh 0 active every time you power up the printer.    If you don't want that behavior, you should do a G29 D followed by the G29 S0.    In that case, the Mesh will still be stored in Slot 0, but when the system comes up, it will load the Slot 0 Mesh, but the UBL system will be deactivated.

kmbecker13, 2016-09-23:
Thanks!  Lots of good info in that reply.  750 bytes sounds good and similar to what i'm working with.

For the G26 command, could you explain a little more on what the 'C' parameter does? I haven't fully grasped that one and haven't been using it.

On the G29 P4, i've actually been doing specifically a "G29 P4" without any R after driving the nozzle over the problem point manually (getting it close enough works).  Is that the same as a G29 P4 R1?  Just curious if omitting the R has any unintended side-effects.

Roxy-3D, 2016-09-23:
> For the G26 command, could you explain a little more on what the 'C' parameter does? I haven't fully grasped that one and haven't been using it.

The UBL System tends to (there are a couple of exceptions) try to operate on Mesh Points close to some starting point.   As those points get visited (whether you are doing a P1 or a P2 or P3 or P4 or a G26) the already visited points fall off the list.   So the nozzle is always trying to find the next closest Mesh Point to work on.    The 'C' parameter modifies that behavior.  It is still looking for the next available point that is 'close' to the starting point.   But with the 'C' parameter, it puts some weight onto the current nozzle position.   That parameter will keep the nozzle from jumping from one side of the already processed area to the opposite side and then back again.   

> On the G29 P4, i've actually been doing specifically a "G29 P4" without any R after driving the nozzle over the problem point manually (getting it close enough works). Is that the same as a G29 P4 R1? Just curious if omitting the R has any unintended side-effects.

gordo3di, 2016-09-24:
If you need graphics or formatting for the manual let me know. I am very proficient in Illustrator, Photoshop, 3dsMax etc... Over 15 years experience. I know it sounds stupid but graphics/formatting can be a huge thing.

adamfilip, 2016-09-24:
Im in the same boat.. been a Graphic Designer for 16 years.. happy to help if needed.

Renatfg, 2016-09-25:
Roxy,
I can not figure out what is wrong.
Z offset configured correctly - after the command
G28 X
G0 Z0
I see that the nozzle is just at the table, i.e., all right.

Here are the settings and results G29 P1:

echo:Z-Probe Offset (mm):
echo:  M851 Z-1.90
>>> m100
SENDING:M100
1716 bytes of memory initialized.
>>> g29 m
SENDING:G29 M
Bed Topography Report:
(0,4)                                                       (3,4)
(0,200)                                                   (200,200)
       .           -1.23250     0.31999     1.85374

       .           -1.00750     0.51593     2.04499

       .           -0.90250   [ 0.62280]   2.11718

       .           -0.92313     0.61249     2.11905

       .           -0.97938     0.51968     2.03749
(0,0)                                                       (200,0)
(0,0)                                                          (3,0)

>>> g29 p2
SENDING:G29 P2
Manually probing unreachable mesh locations.
>>> g29 m
SENDING:G29 M
Bed Topography Report:
(0,4)                                                       (3,4)
(0,200)                                                   (200,200)
 -0.17000    -1.23250     0.31999     1.85374

  0.10000    -1.00750     0.51593     2.04499

  0.21000    -0.90250     0.62280     2.11718

  0.21000    -0.92313     0.61249     2.11905

  0.15000    -0.97938     0.51968   [ 2.03749]
(0,0)                                                       (200,0)
(0,0)                                                          (3,0)

>>> g29 p5
SENDING:G29 P5
# of samples: 20
Mean Mesh Height: 0.410917
Standard Deviation: 1.070752

Look at what happens in the video - nozzle jumps during printing of the circle:
(This can be seen in 12 and 40 seconds.)
https://www.youtube.com/watch?v=X4y5ym2fsdw&feature=youtu.be

thinkyhead, 2016-09-26:
Don't keep stringing it out. Drop that 'M' parameter asap.

sarf2k4, 2016-10-10:
I would like to ask how do I activate the UBL after everything has been probed? 
I did the "g29 a" to activate and load the UBL mesh, but it seems that my z motor are not moving according to the elevation from UBL, is this normal?

I transitioned backed to rc7, comparing free memory (from pronterface) on rc7 is 3.5k while on UBL with previously stored mesh has lesser than 2k left

Roxy-3D, 2016-10-10:
> I did the "g29 a" to activate and load the UBL mesh, but it seems that my z motor are not moving according to the elevation from UBL, is this normal?

Almost for sure...   You had the nozzle above the 'Fade Height'.   The default is 10mm.   Once the print gets up to 10mm, no correction is applied.   If your Z-Height was above that, you won't see any correction as you move the nozzle around.

> I transitioned backed to rc7, comparing free memory (from pronterface) on rc7 is 3.5k while on UBL with previously stored mesh has lesser than 2k left

Your free memory is going to vary depending on your mesh dimensions.  With a 10 x 10 Mesh, you are going to lose 400 bytes right off the top to hold the Mesh.   RC-7 has more optimized memory right now than UBL.   But as long as you have 500 or 750 bytes of free memory, you are going to be fine.

sarf2k4, 2016-10-10:
ahh... I see, no wonder the z axis didn't move, what about during printing, will it be moving during printing or is it just for up to lets say 3mm of layer height only?

Roxy-3D, 2016-10-10:
You get the full correction at Z = 0.0    At the first layer height of .2mm you would get 98% of the correction.   At 9.0 mm you would be getting 10% of the correction.    You get some correction until you are above the 'Fade Height'.   But the amount of correction you get is less and less until there is no correction at all (which happens at the 'Fade Height').

sarf2k4, 2016-10-10:
I see, so basically the layer of the printed object would have some area that are "thicker" compared to the other area and as it slowly fade it will just become a consistent horizontal layer right? 

During printing, at the start of gcode, I usually do this
```
heat up
g28
g29
prime
begin print
```
but with UBL I do this?
```
heat up
g28
g29 a
prime
begin print
```

Roxy-3D, 2016-10-10:
UBL is taking a different path than previous G29's.   If you activate G29 and Save a Mesh at some slot number,   it will automatically be active when Marlin comes up.   You can have the old way back by deactivating the UBL system and saving the Mesh at some location.  In that case, you will need to explicitly activate it after the G28.

But if you activate and save the Mesh, you can eliminate the G29 A line.   Your chosen mesh will be loaded and active.    And you can verify that with the first few lines of a G29 W command.   Or by moving the nozzle down to 1mm above the bed and moving it around.

sarf2k4, 2016-10-10:
Okay, I will give it a try shortly

Roxy-3D, 2016-10-11:
Can you create a Mesh with a G29 P1 O command?    The G29 W command will tell us what G29 thinks is happening in its world.  That would be interesting to see also.   As well as a map of your mesh:   G29 O

sarf2k4, 2016-10-13:
This is the results, Even loading the previously probed mesh would show something similar especially on the sanity check at the end of it as ok/passed

[image]

Roxy-3D, 2016-10-13:
Please do a G29 A   and then a G29 O  and then a G29 W      (You can actually stack the commands.  IE.   G29 A O W    but lets keep things simple right now until you see things working.)    Please verify you have a full correction mesh defined and that it says the UBL System is Active.

Then do a G29 S 1     To save the Mesh in slot 1.     When you do a G29 W  it should say the active mesh is 1.      Now, if you reboot the printer and do a G28 you should be able to do a G29 O and see that the UBL System is still active and that mesh 1 got loaded.   Similarly, if you do a G29 O you should see the same mesh numbers.   If you bring the nozzle down to 1mm in height and start moving around in the X and Y axis, you should see the corrections being applied.

fr33dimension, 2017-05-15:
I flashed Marlin 1.1 on one of my machines to be able to use the z endstop and zprobe together, mainly to use it on another machine that really needs bed leveling. I tried UBL before and liked it but it had some bugs and I truned it off. Now, when I probe the bed I have 4 points that the probe can't reach, they are on the left side since my probe is 43mm left to the nozzle. When doing a G29 P2 to manually measure these points, the probe goes all the way to the right of the bed X190 to measure, after completing the 4 points, the probe moves to x0 y200......looks like the whole thing is mirrored, but all my numbers in marlin seam to be okay.

Bob-the-Kuhn, 2017-05-15:
If the probe is to the left of the nozzle then the right side of the bed is where the probe can't reach.

We've just added a graphic at the very end of the guide in the wiki [guide in the wiki](http://marlinfw.org/docs/features/unified_bed_leveling.html)  that shows this.

To confirm move the head to the far right and do G29 P1 again.  The head should move left before it does any probing.

Make sure you've saved your mesh before doing the G29 P1 again.

mainframecn, 2017-06-27:
I was playing around with this and needed to do a fine tune of a single point (alu-bed little bit saggy in the middle). So moved the nozzle close to the middle and did a G29 P4 T. No matter what value i put in there, there is no Z change of that point. (UBL was active, mesh was saved. Checked with G29 W and G29 T. The mesh was adjust but no change in the Z on "tuned" points.

vido89, 2017-09-25:
@kmbecker13 Did you write any documentation on this topic I can help you with drawing/writing ?

closed 2018-02-06 by thinkyhead