MarlinFirmware/Marlin #4890: Another compiling issue with the Viki 2 LCD

cjryker06, 2016-09-24:
Hey guys, I was directed here by the users on the Robo forums. I am trying to setup my LCD and all is well until it gets to the LED color portion. I get a compile error stating a line in the code is not declared in scope. I'm very new to Arduino sketching. It compiles fine as //#define TEMP_STAT_LEDS, but with it uncommented, it does this:
__________________________________________________________________________________________________________

Arduino: 1.6.9 (Windows 10), Board: "Arduino/Genuino Mega or Mega 2560, ATmega2560 (Mega 2560)"

sketch\Marlin_main.cpp: In function 'void handle_status_leds()':

Marlin_main.cpp:8421: error: 'STAT_LED_RED' was not declared in this scope

           digitalWrite(STAT_LED_RED, new_led ? HIGH : LOW);

                        ^
Marlin_main.cpp:8422: error: 'STAT_LED_BLUE' was not declared in this scope

           digitalWrite(STAT_LED_BLUE, new_led ? LOW : HIGH);

                        ^
exit status 1
'STAT_LED_RED' was not declared in this scope

thinkyhead, 2016-09-24:
Which board are you using?

cjryker06, 2016-09-25:
Mega 2560 with Ramps 1.4 on top of it.

thinkyhead, 2016-09-25:
So, `BOARD_RAMPS_14` ?

thinkyhead, 2016-09-25:
It seems that these pins are in fact defined in Marlin `1.1.0-RC` and `RCBugFix`. Are you using an older version of Marlin?

cjryker06, 2016-09-25:
It has my Ramps properly established. I am using Marlin 1.1.0 RC7 working on migrating to RC bugfix. I assigned the pins according to the document from Panucatt for the Viki 2. Everything else seems to compile fine outside of the status led colors.

thinkyhead, 2016-09-25:
As you can see, the pins are defined in the RAMPS pins file. So it's odd that you are seeing that compiler error.

https://github.com/MarlinFirmware/Marlin/blob/RC/Marlin/pins_RAMPS.h#L267-L270

thinkyhead, 2016-09-25:
And indeed, starting with 1.1.0-RC7 and simply enabling `VIKI2` and `TEMP_STAT_LEDS` no error is generated. Is it possible you have more than one display type enabled? Please post your configuration files.

cjryker06, 2016-09-25:
It appears my pins don't match what you put, but I believe I changed them according to the setup.
      #if ENABLED(TEMP_STAT_LEDS)
        #define STAT_LED_RED   39
        #define STAT_LED_BLUE  37

thinkyhead, 2016-09-26:
> I didn't see the issue and that being linked.

The pins are defined based on which controller is enabled. The latest `RCBugFix` sanity checks to catch errors like that.

cjryker06, 2016-09-26:
For now I would Iike to stick with stable builds, so to be safe should I proceed with the pins changed according to the guide or default Marlin?

Sent via the Samsung Galaxy S®6 active, an AT&T 4G LTE smartphone

  LCD (#4890)

thinkyhead, 2016-09-26:
Be aware: stable builds may still have bugs. "Stable" just means they won't be getting changed (or fixed).

What do you mean by "default Marlin"?

Should you proceed with the pins changed? Just use whatever works.

cjryker06, 2016-09-27:
I meant the default pin layout for Marlin before I changed the Pins.h according to the Viki 2 guide. I just don't want to ruin an LCD.

Sent via the Samsung Galaxy S®6 active, an AT&T 4G LTE smartphone

  LCD (#4890)

thinkyhead, 2016-09-27:
You can use the newest Marlin (`RCBugFix`) pins files as a guide for how to set the Viki 2 pins in an older version of Marlin. I'm not sure how much they might have changed.

closed 2017-10-31 by thinkyhead