MarlinFirmware/Marlin #4897: 1.1.0 RC7 - X-Endstop triggering during movement

johannesgeh, 2016-09-25:
I updatet to Marlin 1.1.0 RC7 a few days ago to be able to use Manual Mesh Bed leveling.

Since then every print fails because the X-Endstop gets triggered (without the Extruder even being near it) after about a Minute or so of printing.
This also happens when I move the Extruder manually with the "Move Axis" option on the LCD.
It homes just fine and if I only move it back and forth a few times nothing happens, but when I repeatedly do it at some point the Extruder abruptly stops when moving towards the endstop.

I use ordinary Tact-Switches with the Coarse Pull-Up enabled. I had no problems with this setup in 1.0.2-1.

I hope that this is the right place to write this as I've never used GitHub before.
Thanks!

(I now downgraded to my old 1.0.2-1 and everything functions normally, so the endstop isn't the fault)

kmbecker13, 2016-09-25:
You should triple check your wiring, very likely that something is loose. I
had the same issue with my printer and it was quite confusing since it was
an intermittent problem.

jasoroony, 2016-09-26:
The Marlin developers may have removed the "triggered twice" check for endstops.  It is likely you are picking up crosstalk from other wires, which is being rejected by the old Marlin version.  Don't know why they would remove it, but it's changed fairly significantly now and this is their project to do with as they please.

thinkyhead, 2016-09-26:
The method of checking endstops hasn't changed. It pleased our caprices to leave it as it was.

emartinez167, 2016-09-26:
I thought the age of caprices ended a while ago... ;)

johannesgeh, 2016-09-26:
@thinkyhead I'll try isolating the endstop wires when I get back home, I already have ferrite beads on all stepper wires.

thinkyhead, 2016-09-26:
It's hard to debug a loose endstop or wiring because there's no "continuous" check of the endstop status. Seems like that would be a helpful debug feature to add at some point.

johannesgeh, 2016-12-19:
Rewired all Endstops, seems to work without any issues now.

closed 2016-12-19 by johannesgeh