MarlinFirmware/Marlin #4898: Bug, Move platform wrong

tintinz, 2016-09-25:
Hello.
Thanks for the amasing work you guys do!
I try to use my Ends stop as a Z-prob and have problem using "safe homeing"

If i restart my printer klick go home Y the platform move down 5mm and Y go home.

If I click X the platform start going up insted and ignor hitting the endstop.

if I start with Y the nX  works fine

also if I do a print and after the print try to home Y. The platform gos up and ignor the endstop... 

I have a Mbot cube 3D with ramps 1.4, moded. there is endstop sitting on the extruder and hole in the bed for the nozzle so the safe home is inportent ;-)

[Configuration.txt](https://github.com/MarlinFirmware/Marlin/files/491897/Configuration.txt)

Thanks

thinkyhead, 2016-09-26:
What version of Marlin are you using?

tintinz, 2016-09-26:
the readme says
## Release Candidate -- Marlin 1.1.0-RCBugFix - 27 April 2016
- RCBugFix

- RC6 - 24 Apr 2016
  - Completed support for CoreXY / CoreXZ in planner
  - Changes to positioning behavior
  - Various issues fixed. More details pending.
  - Throw error if compiling with older versions (<1.60) of Arduino due to serious problems with outdated Arduino versions
  - Please upgrade your IDE at least to Arduino 1.6.0. Thanks.

- RC5 - 01 Apr 2016
  - Warn if compiling with older versions (<1.50) of Arduino
  - Fix various LCD menu issues
  - Add formal support for MKSv1.3 and Sainsmart (RAMPS variants)
  - Fix bugs in M104, M109, and M190
  - Fix broken M404 command
  - Fix issues with M23 and "Start SD Print"
  - More output for M111
  - Rename FILAMENT_SENSOR to FILAMENT_WIDTH_SENSOR
  - Fix SD card bugs
  - and a lot more
  - see https://github.com/MarlinFirmware/Marlin/releases/tag/1.1.0-RC5 for details

thinkyhead, 2016-09-26:
A lot of changes since April! Try this one:

https://github.com/MarlinFirmware/Marlin.git

tintinz, 2016-09-26:
Thanks!
Do I need to do a new configuration file or is it possible to use the same?

thinkyhead, 2016-09-26:
@tintinz If you drop in your old configuration files and try to build, the latest Marlin will give you guidance on anything you need to change. Go for it.

closed 2017-10-31 by thinkyhead