MarlinFirmware/Marlin #4901: Hello All! What is meant by "const unsigned char start_bmp[574] PROGMEM". In this what is 574? I know it may be arrary size. But how come 574 only. Please reply.

Shrikant888, 2016-09-26:
Hello All!
What is meant by "const unsigned char start_bmp[574] PROGMEM". In this what is 574? I know it may be arrary size. But how come 574 only. Please reply.

Blue-Marlin, 2016-09-26:
Someone changed the graphic (and its size) but did not correct the arrays size.

START_BMPBYTEWIDTH * START_BMPHEIGHT = array size in byte

So should be 512 in this case.

START_BMPBYTEWIDTH = ((int)START_BMPWIDTH / 8) + (START_BMPWIDTH % 8) ? 1: 0

Marlin V1.0.x is very outdated! Don't use it any more..

Shrikant888, 2016-09-26:
Yeah Got it sir. Ok,Now since Marlin-Reprap allowed to add custom boot up message. So, Here is what I am requesting. I have created 128x64 logo and wanted to display first and then next will be "Reprap Marlin" and then home screen so on. What change should I make? Please tell according to Marlin V1.0.x.

Blue-Marlin, 2016-09-26:
Have a look at RCBugFix and port back.

Shrikant888, 2016-09-26:
Yeah I have gone through it RCBugFix just now. There is unit for custom logo. Ok, Now with Marlin V1.0.x. What should be array size for 128x64 bmp logo.?
I am new to this parts so asking.

Blue-Marlin, 2016-09-26:
128/8*64 = 1024

Shrikant888, 2016-09-26:
Ok. .I am trying now..

Shrikant888, 2016-10-03:
Please let me ask again sir. I have designed my own 128x64 resolution logo and I wanted to display it. So, process will be like first my 128x64 logo will have to come and then regular REPRAP logo should come. I am not changing REPRAP and then rest will be same. What changes I should make in dogm_lcd_implementation.h and in DOGMbits.h.. Please tell.

thinkyhead, 2016-10-04:
See the files in `Marlin/example_configurations/Hephestos_2` for an example of making a custom boot screen, with links to the tools used.

Shrikant888, 2016-10-04:
Hi all.. By the way I am using Marlin V1.0... Please tell...

Roxy-3D, 2016-10-04:
> Hi all.. By the way I am using Marlin V1.0... Please tell...  

If this is important to you, you need to invest some time so you can support yourself.   And really...   what ever work you do should be done using RCBugFix.

thinkyhead, 2016-10-05:
@Shrikant888 Yeah… Replacing the boot screen in 1.0.x is just a matter of figuring it out. It should be fairly "obvious" based on how it was done in later versions of Marlin. Study, trial-and-error, and perseverance. That's how to do it.

closed 2016-10-05 by thinkyhead

Shrikant888, 2016-10-06:
Oh ya well advices:)... I have changed the code and now seems to coming. I will tell now.

I have put my 128x64 logo and remaining REPRAP as usual. But after compiling. I am getting only 64x64 of my logo. I don't know why it is....! and amazing thing is when I run the same logo with bit map demo program its coming perfectly. Any one guess please what might be the solutions...