MarlinFirmware/Marlin #4902: About Boot up message custom...

Shrikant888, 2016-09-26:
Now since Marlin-Reprap allowed to add custom boot up message. So, Here is what I am requesting. I have created 128x64 logo and wanted to display first and then next will be "Reprap Marlin" and then home screen so on. What change should I make? Please tell according to Marlin V1.0.x.

thinkyhead, 2016-09-27:
No option to do that with Marlin 1.0.x. You'll have to hack it in. With 1.1.0 you just drop in your own custom boot screen and set an option, and it automatically replaces the regular Marlin one.

Shrikant888, 2016-10-03:
Please let me ask again sir. I have designed my own 128x64 resolution logo and I wanted to display it. So, process will be like first my 128x64 logo will have to come and then regular REPRAP logo should come. I am not changing REPRAP and then rest will be same. What changes I should make in dogm_lcd_implementation.h and in DOGMbits.h.. Please tell.

closed 2016-10-05 by thinkyhead

Shrikant888, 2016-10-06:
Oh ya well advices:)... I have changed the code and now seems to coming. I will tell now.

I have put my 128x64 logo and remaining REPRAP as usual. But after compiling. I am getting only 64x64 of my logo. I don't know why it is....! and amazing thing is when I run the same logo with bit map demo program its coming perfectly. Any one guess please what might be the solutions...