MarlinFirmware/Marlin #4906: Heated Bed Overdrive

to-the-nth, 2016-09-27:
For heated beds using copper heating elements, power falls off by ~30% by the time the bed reaches a final temperature of about 115C (which is typical for ABS).  In some cases, a power supply with extra current capacity is needed to deliver the cold start up current (which adds cost - sometimes crazy amounts).   Both of these can be addressed with a code hack.

Constant Power ~360W supply=25.5V 14.1A constant  160sec to 115C
[image]

Before a series of HW upgrades, my bed heat times were on the order of 10 minutes.  On my printer, I've added a second 24V 15A 360W Mean Well style switching power supply to drive a custom sticky copper tape heated bed.  The Mean Well style supplies are the eBay cheapy supplies commonly use to drive LED lighting.  Most of them use the same core design.  Here is a great[ link to the internal workings](http://imajeenyus.com/electronics/20151028_smps_variable_voltage/index.shtml).  With this overdrive hack, I get a power density of 0.45W/cm^2.  Some mains powered heaters are running in the 0.5W/cm^2 range.  Bed heat time dropped from 225sec to 160sec.  This hack requires enabling bed PWM (#define PIDTEMPBED).

So the idea is to crank up the voltage adjust on the power supply so that the power supply CAN  deliver full rated output power when the bed is near a max temp, as needed by the stock PID control loop.  In my case, I chose a cross over temp of 110C.  Above that temperature, the overdrive is off.  Below that temperature, the firmware hack uses the Marlin software bed PWM to limit the max bed power.  The  MAX_BED_POWER configuration #define is turned into a variable that is a function of bed temperature.  Copper has a temperature coefficient of about 0.0039/degC, so for lower temperatures we have to linearly reduce the amount of PWM and max bed power to stay within the rated limits of the supply.  You can see in the pic above how the PWM bed power is initially limited to about 77% of max, and then gradually increases to 100% at 110C. 

In practice, the measured values for the temperature coefficient were slightly different from the ideal for copper.  So the hack uses measured values of resistance at room temperature and at the 110C number.  Voltage and current measurements are easy, and are done at reduced supply voltage, then plugged into #defines.  You probably don't want to try this with a  stock RAMPs board power FET and polyfuse because of their huge power losses.  My RAMPs board is driving a custom solid state relay.

One concern is that the stock firmware bed PWM cycle time on the order 131msec.  At low temps the power supply doesn't get much in the way of averaging, but sees a train of relatively large current and power pulses.  Setting #define SOFT_PWM_SCALE 3 helps somewhat by changing the PWM cycle time to 16.3msec  at the loss of granularity.   Does anyone know why bed hardware PWM would not be infeasible? and if it has been implemented somewhere in Marlin history?

Would there be enough general interest in this hack to create a branch?  There are a handful of edits to cut and paste in if anyone wanted to try it.  My current code base is seriously down rev from trunk.

Constant Voltage supply=21.4V 15A starting at room temp  225sec to 115C
This is the same hardware without the overdrive hack.  Supply voltage had to be limited to keep from exceeding the max 15A current limit of the supply.
[image]

Rerouter, 2016-09-27:
I would also think bed hardware PWM if not already a feature seems to be a good one, as again cheap Chinese PSU, means you can hear the PWM effect on the supply, rather than it averaging out,

I think the other functionality could best be described as inrush protection/ soft start ramp up, In my head the cheapest (least amount of edits) way of approaching this would be to be able to relax the PWM limit once in PID mode vs bang-bang (which by default is a 10C window around setpoint)

but i suppose it could also be done by linearly interpolating between 2 limits vs input temperature, just sounds uglier to me.

AndyZuerich, 2016-09-27:
Hi,

I would like to give some comments on this, since I am tuning my heated bed at the moment, as well. I think in general, modifying the power supply internals, as described above is:
- something for experts who know what they are doing -> not many people will follow the same path, but will just buy another power supply or heated bed that better suits their needs.
- can be risky (fire hazard) if done wrong. Of course, the power supplies have their margins, but we never know which power/current is too much for the supply.

`Would there be enough general interest in this hack to create a branch? There are a handful of edits to cut and paste in if anyone wanted to try it. My current code base is seriously down rev from trunk.`

I doubt that enough people are trying to do the same. And even if so: I think it is too risky (fire hazard!) to encourage people to do so by implementing an according feature for this in Marlin.

Solid State Relay: I think there are better alternatives for expensive solid state relays for DC applications. There are Mosfets available which perform quit well and are much chaper than DC SSR's. I am using this one for my 12V heated bed:
Transistor unipolar (MOSFET) International Rectifier IRLB8743PBF N-Kanal Gehäuseart TO-220AB I(D) 150 A U(DS) 30 V
This Mosfet does not even get hot (without Heatsink!) with 25 Amps of current. Since you have a 24 V input, this Mosfet is possibly not suited (30 V breakdown). Additionally to the blocking voltage (24 V), you have to add the voltage overshoot during switch turn-off, caused by parasitic inductances.

Ok, this was my criticism, sorry for being so negative about your approach. Here are my additional (and possibly helpful) advices  and comments:
- I am using a buck converter (stepdown-converter) for my heated bed, with the Mosfet mentioned above. A buck converter is not complicated [(https://en.wikipedia.org/wiki/Buck_converter)] . Compared to the simple Mosfet on-off pwm, the buck converter has a freewheeling diode, input- and output capacitors and an inductor. The result is a smooth output voltage (with low ripple), and a smooth input current (low ripple). I am switching the Mosfet gate directly from the Arduino with 30 kHz PWM frequency. A separate gate drive was not necessary, since the Arduino could supply enough current to switch the Mosfet on within 300 nsec. In total, the buck converter is comparable to an adjustable transformer, but just for DC/DC conversion.

As far as I understood, the problem during your heatup-phase is that the current that the bed draws is, for lower temperatures, too much for your power supply, and you are using the MAX_BED_POWER feature of Marlin to limit the current during the heat-up phase. Well, what you do is not limiting the maximum current/power, but the maximum average power. Therefore, I would strongly suggest that you also use a buck converter. Then, you can use the full 360 W of your power supply, even if the bed resistance changes with temperature, but:
- without modifying the power supply. The buck converter stores energy in the C's and L during the off-phase of the switch. This gives you a real adjustable output voltage. For instance, when you have a buck converter with 24 V input and 12 V output, then the input current is also half the output current.
- you have less potential problems with electromagnetic interference, since the bed power is not rectangular pulses, but nearly DC (with a minor voltage/current ripple).

> One concern is that the stock firmware bed PWM cycle time on the order 131msec. At low temps the power supply doesn't get much in the way of averaging, but sees a train of relatively large current [...]

Concerning PWM frequency: I would not use or modify the `#define SOFT_PWM_SCALE 3` for increasing the switching frequency. I would directly use the analogWrite() of Arduino the generate the PWM pattern. There, you can modify the PWM frequency: `setPwmFrequency(HEATER_BED_PIN, 1);` This command changes the hardware PWM frequency to 31 kHz. If you like to have another frequency, you can do for instance `setPwmFrequency(HEATER_BED_PIN, 2);` for 4 kHz. Just be careful, the following PWM pins are then affected: PIN 6, PIN 7, PIN8 . Pin 8 is the default heated-bed pin. Pin 7 is not used (but easily accessible for testing purposes with PWM), and Pin 6 is used for one of the Servos. For more detailed PWM frequency settings, please read [http://forum.arduino.cc/index.php?topic=72092.0] . For my buck converter, the 31 kHz is perfect, since the size of the inductor can then be quite small. And, 31 kHz is outside of the audible spectrum. 4 kHz noise sounds like a train that is accelerating out of the station.

I think the final (and best) solution for our 24 V (or 12 V) heated beds would be a power supply that is adjustable with an external PWM signal from 0..24 V. Obviously, it does not make sense to convert 230 V voltage to 24 volts, and then adjusting the lower voltage (and huge currents) in a second stage to control the heated bed. Unfortunately, I don't know if such a power supply exists for a reasonable price. The other solution are 230 V beds with SSR's (here, they make indeed sense). But having 230 V directly on the printer is also something people don't want to have.

thinkyhead, 2016-09-27:
This would make a good subject for a blog post and a YouTube video. Maybe a "special feature" on marlinfw.org in the future, once we get the basic documentation completed. If there's a page on this subject at the reprap.org wiki, then it could be augmented. If not, it could be created.

to-the-nth, 2016-09-29:
BTW.  I was not proposing modifying the PSU.  These supplies come with an adjustment trim pot.  One of my supplies can be varied from 18V to over 28V.  I find this common design fascinating because even the component reference designators tend to be the same from manufacturer to manufacturer.  The same common PCB can be stuffed to build up a 15/20/30A and 5V, 12V, 24V, etc versions.  I included the link on internal workings as a reference for understanding the current limit control and overvoltage protection.  The self oscillation start up sequence was interesting.

@Rerouter - I'll report back on the PSU noise after trying Andy's 31kHz PWM.  It is kind of an inrush limiter.  The trick is figuring out the operating points for max power transfer without over loading while the bed is ramping up. 

@thinkyhead - Agreed.  Let's let it cook here for a while, and then see if there is enough traction to roll something out.  

@AndyZuerich - Andy my friend.   
-THANK YOU for the great suggestion for using analogwrite and setPwmFrequency and info about pin interactions.  That makes it too easy!  I was starting to dig into timer control registers.
-On the "SSR".  I am using IRLB3034pbf 40V 343A(silicon limited) FETs that I scored on a bulk order from Verical.com  I built it up to be opto isolated with a full gate driver and inductive clamp.  I was tempted to use a photo-voltaic opto (cool devices) but decided against that in case I wanted to PWM it at 100kHz.  Now it looks like that will come in handy at 31 kHz.
-I hear you about the buck converter. I went down the same path for a while.  But I think the problem statement we want is to have the supply deliver full rated 360W (in this case) to the bed, as the resistance of the bed varies.  

> For instance, when you have a buck converter with 24 V input and 12 V output, then the input current is also half the output current.

I think you want to say that the **_average_** input current is half the output current.  _The Peak input current is identical to the output current._  

Even so, there should be an advantage to using a buck converter in terms of reducing the peak current delivered by the PSU.  However, the peak current into the buck converter will still exceed the average PSU current rating (while maintaining full rated 360W to the bed), just not as much as PWMing the bed directly.  (Maybe I am too hung up on this peak current thing.  The output filter caps on the PSU should be able to filter out most of the ripple if the PWM runs at 31kHz.)

Consider the example of a 25.5V PSU delivering 14.1A to a bed load of 1.80 Ohms at 110C.  With both a buck converter and directly PWMing the bed,  the duty cycle is 100% and delivered power is 360W.  

Now assume we just applied power to a cold bed who's resistance is 66% of the high temp value or 1.2 ohms.  

For the direct PWM case, the voltage is still 25.5V and the bed current is 21.25A and bed power of 541W.  The  PWM duty cycle needs to be set at 66% to have the average bed power match the max 360W capacity of the PSU.  The average 14.1A PSU current is 66% of the peak 21.25A. 

For the buck converter, the PWM gets adjusted to deliver a constant voltage of 20.78V and 360W to the bed.  When cold, the bed voltage decreases by SQRT(0.66) and the peak current increases to 17.3A or by 1/SQRT(0.66) from the original 14.1A.  So for the direct PWM, the peak current varies inversely with the bed resistance, where for the buck converter, the peak current varies inversely with the SQRT of the bed resistance.

Andy, you could  adjust the PWM value on your buck converter based on the SQRT of the estimated/interpolated bed temperature to maintain constant max power during heat ramp.

AndyZuerich, 2016-09-29:
Hi again.

first thing: I think we should move this discussion to a reprap form, since many of the hardware details are not really specific to the Marlin firmware. Well, later on when things have settled, we still could continue in the marlin issues page for a concrete implementation of this overdrive feature.

@to-the-nth Ok, I see, there were some misunderstandings.

- The basic difference between your setup an mine is: I am using an ATX power supply and especially the 12V output. The ATX supply has 550W, which is more than enough for my heated bed. Therefore, I cannot use the full ATX power on my heated bed. Well, I could use a buck-boost converter instead of the buck to drive my bed. With the buck-boost, I would be able to set higher and lower voltages for the bed voltages compared to the 12 Volts. And then, using some resistance approximation over temperature would also make sense for me.
- Next misunderstanding: SSR: Ok, I think the IRLB3034pbf 40V 343A is a good choice for you.
- Next misunderstanding: With buck converter, I always mean some circuit that has input caps (instead of the constant voltage source that is typically drawn in the basic explanation schemes). Then, including the input capacitors, the maximum input current is really averaged out, and then the average input current is (except of a minor current/voltage ripple) equal to the peak input current.

A buck converter can then be seen as a simple switch with an input filter (caps) + an output filter (cap + inductor) and a freewheeling diode.

This brings me to the next advice: As I understand now, you have something as a buck converter, but without input filter and output filter. Removing the output filter (cap + inductor) might be ok. What you get is a rectangular pulsating bed power (which might introduce electromagnetic interference problems). Now the big "BUT":
When you misuse the power supply capacitors as your "Buck" input filter stage, this might be dangerous:
- The supply cables from the PSU to your Mosfet have for shure some inductance, and this could be deadly for the switch! A small calculation: Let's assume you have 200nH parasitic cable inductance, and you are switching a current of 14 Amps during the switching time of 100nsec. This is leading to a switch turn-off overvoltage of U_over = L * di/dt = 28 volts, i.e. your maximum voltage at the switch might be U_max = U_supply + U_over = 25 V + 28 V = 53V. What I want to say with this: Please beware of not killing your Mosfets! I think in any case, there should be an input capacitor directly at your switch input stage. And even there, you have to consider: Electrolytic capacitors have quite some series inductance and ESR. The cap's series inductance is ofter too large so that the cap does act as an inductor for a rise time of e.g. 100 nsec. I used a 100uF SMD cap, located geometrically as near as possible to the switching stage. And even then, I get a voltage overshoot of 10volts, which brings my 30 V switch close to its limit.
If you have a oscilloscope available, please try to measure the voltage at your switch during turn-off, just be be shure about your circuit's behavior.
- Additionally be careful with the higher switching frequency (e.g. 31 kHz) and using the PSU output  caps as your input filter. The schematic in your link to the internal workings of the PSU is too unsharp to see the details. But: If the output caps are not designed well enough to withstand a huge current ripple at a frequency of 31 kHz (+ Harmonics), they might overheat and get destroyed. -> Also here, using input caps directly on your switch would help a lot. 

In think for a possible continue of the discussion, we should open a new thread in the reprap forum...

Rerouter, 2016-09-29:
On reflection, the simplest hardware method to resolve this is to use a constant current led buck driver, this way you never exceed the supply and as the bed heats up, the power increases (I^2R)

to-the-nth, 2016-09-30:
Ok let's move the discussion.  reprap developers forum.  http://forums.reprap.org/read.php?2,710677

Roxy-3D, 2016-09-30:
@AndyZuerich 

> first thing: I think we should move this discussion to a reprap form, since many of the hardware details are not really specific to the Marlin firmware.

OK!  But I think there is a firmware aspect to this.   The bottom line is you want to match the Bed's impedance (over time) to the Power Supply's impedance in order to get maximum power transfer.   Right?   Is this logic going to be implemented in Hardware or Firmware?

closed 2016-09-30 by Roxy-3D