MarlinFirmware/Marlin #4907: Current RCBugFix - Cannot compile Delta XL Configuration

aljscott, 2016-09-27:
I pulled RCBugFix 4ac520435f1e42c1fe112c9ff2c84111f51a061d 
I cannot compile the Kossel XL Delta Configuration.

```

Marlin_main.cpp:308:17: error: expected initializer before 'homing_feedrate_mm_s'
 float constexpr homing_feedrate_mm_s[] = {
                 ^
Marlin_main.cpp: In function 'float get_homing_bump_feedrate(AxisEnum)':
Marlin_main.cpp:1443:17: error: expected initializer before 'homing_bump_divisor'
   int constexpr homing_bump_divisor[] = HOMING_BUMP_DIVISOR;
                 ^
Marlin_main.cpp:1444:13: error: 'homing_bump_divisor' was not declared in this scope
   int hbd = homing_bump_divisor[axis];
             ^
Marlin_main.cpp:1450:10: error: 'homing_feedrate_mm_s' was not declared in this scope
   return homing_feedrate_mm_s[axis] / hbd;
          ^
Marlin_main.cpp: In function 'void do_homing_move(AxisEnum, float, float)':
Marlin_main.cpp:2214:149: error: 'homing_feedrate_mm_s' was not declared in this scope
     planner.buffer_line(current_position[X_AXIS], current_position[Y_AXIS], current_position[Z_AXIS], current_position[E_AXIS], fr_mm_s ? fr_mm_s : homing_feedrate_mm_s[axis], active_extruder);
                                                                                                                                                     ^
Marlin_main.cpp: In function 'void home_delta()':
Marlin_main.cpp:2870:21: error: 'homing_feedrate_mm_s' was not declared in this scope
     feedrate_mm_s = homing_feedrate_mm_s[X_AXIS];
                     ^
Error compiling.
```

thinkyhead, 2016-09-27:
What version of Arduino are you using?

aljscott, 2016-09-27:
Definitely using arduino.cc - But I'll upgrade tonight and keep note of the versions.

thinkyhead, 2016-09-27:
Two things you might try, apart from updating to a newer Arduino…
- Change `float constexpr` and `int constexpr` to `constexpr float` and `constexpr int`, or
- Change them to `static constexpr float` and `static constexpr int`.
If the code still doesn't compile, perhaps I can find a workaround that supports both older and newer compilers.

aljscott, 2016-09-28:
@thinkyhead - sorted.
I had arduino IDE 1.6.2
I upgraded to 1.6.12 and also upgrade the core libraries.
Compiles now.

closed 2016-09-28 by aljscott

Renatfg, 2016-11-23:
Exactly the same error.

Arduino was updated to version 1.6.12, 1.6.13 and 1.6.14 hourly build.
The error remains.

Tried to rename as you pointed out variables, it is still not compiled. What can I do?

Grogyan, 2016-11-23:
Upgrade your Java

thinkyhead, 2016-11-26:
@Renatfg Arduino 1.6.8 and later should have no problem compiling the current code. If you still see an error, please post the Arduino output log here.

o0shojo0o, 2016-12-18:
Hi,
i have the same error ....

Arduino IDE: 1.6.13
Java 32bit: 8 update 111

 `Marlin_main.cpp:378: error: expected initializer before 'homing_feedrate_mm_s'

 float constexpr homing_feedrate_mm_s[] = {

                 ^

sketch\Marlin_main.cpp: In function 'float get_homing_bump_feedrate(AxisEnum)':

Marlin_main.cpp:1595: error: expected initializer before 'homing_bump_divisor'

   int constexpr homing_bump_divisor[] = HOMING_BUMP_DIVISOR;

                 ^

Marlin_main.cpp:1596: error: 'homing_bump_divisor' was not declared in this scope

   int hbd = homing_bump_divisor[axis];

             ^

Marlin_main.cpp:1602: error: 'homing_feedrate_mm_s' was not declared in this scope

   return homing_feedrate_mm_s[axis] / hbd;

          ^

sketch\Marlin_main.cpp: In function 'void do_blocking_move_to(const float&, const float&, const float&, const float&)':

Marlin_main.cpp:1742: error: 'homing_feedrate_mm_s' was not declared in this scope

     feedrate_mm_s = fr_mm_s ? fr_mm_s : homing_feedrate_mm_s[Z_AXIS];

                                         ^

Marlin_main.cpp:1754: error: 'homing_feedrate_mm_s' was not declared in this scope

     feedrate_mm_s = fr_mm_s ? fr_mm_s : homing_feedrate_mm_s[Z_AXIS];

                                         ^

sketch\Marlin_main.cpp: In function 'void do_homing_move(AxisEnum, float, float)':

Marlin_main.cpp:2663: error: 'homing_feedrate_mm_s' was not declared in this scope

   planner.buffer_line(current_position[X_AXIS], current_position[Y_AXIS], current_position[Z_AXIS], current_position[E_AXIS], fr_mm_s ? fr_mm_s : homing_feedrate_mm_s[axis], active_extruder);

                                                                                                                                                   ^

exit status 1
expected initializer before 'homing_feedrate_mm_s'`

thinkyhead, 2016-12-19:
@o0shojo0o It's not clear what version of GCC is being used to compile. Are you on a Linux system where GCC is separate from the Arduino IDE? That might explain why an older version of GCC is being used which can't handle C++11 features.