MarlinFirmware/Marlin #4909: K8400 endstop pins

urbhar580, 2016-09-27:
The K8500 Velleman Vertex only x- and y-max endstops, both located at the rear right side. There are no pins on the board for more than one endstop for X and Y. In the RC7 the pins are set in three files, RAMPS, 3DRAG and K8400. In the config.h the MIN pins are set active. 

The result is that during homing of X and Y the end stops are not activated. 

To fix the problem I did the following two changes:

In configuration.h:
//#define USE_XMIN_PLUG //Changed to comment
//#define USE_YMIN_PLUG //Changed to comment
#define USE_ZMIN_PLUG
#define USE_XMAX_PLUG  //Changed removed comment
#define USE_YMAX_PLUG  //Changed removed comment
//#define USE_ZMAX_PLUG

In Pins_K8400.h :
#undef X_MAX_PIN
#undef X_MIN_PIN     
#undef Y_MAX_PIN
#undef Y_MIN_PIN     

#define X_MAX_PIN	3
#define X_MIN_PIN     -1
#define Y_MAX_PIN	14
#define Y_MIN_PIN     -1

Maybe not the correct place to do it, but after this the homing worked correctly.

thinkyhead, 2016-09-27:
Is that only applicable to the K8500, a newer machine? … or does it also apply to the K8400?

We actually have a better facility for dealing with the situation of fewer endstop pins. In the appropriate pins file, 

```cpp
#undef X_MAX_PIN
#undef X_MIN_PIN
#undef Y_MAX_PIN
#undef Y_MIN_PIN
#define X_STOP_PIN  3
#define Y_STOP_PIN 14
```
The "stop" pins will be re-assigned to the min or max according to the configuration.

urbhar580, 2016-09-28:
My fingers slipped, it is the K8400. 
http://www.vertex3dprinter.eu
ok, I will insert that and test it.

thinkyhead, 2016-10-01:
Merged #4919 which should work here.

closed 2016-10-01 by thinkyhead