MarlinFirmware/Marlin #4911: Thermal Protection not Working when setting temperatures from LCD panel

AndyZuerich, 2016-09-27:
Please try the following for a problem / bug verification:
- enable the thermal bed protection in configuration.h
- leave your bed power supply off
- set a temperature of e.g. 100 degrees on your lcd panel
- wait for a while. Nothing happens (the power is off)
- then, do the same with some G code, or Pronterface. Set the temperature to 99 degrees
- wait for a while. Now, the thermal protection reacts and tells me that heating failed and that I should reset. I think the function `void Temperature::start_watching_bed() {` is not called when temperatures are set from the LCD panel. But it should be called. The Marlin user could be misleaded to believe that the bed (and also hotend) has a thermal runaway protection. But the protection is not active when we use the LCD panel.

thinkyhead, 2016-09-27:
Hmm. I remember specifically patching the LCD functions to call `start_watching_*` when this issue came up before. I'm surprised to hear that it might only be my imagination. I'll look into it.

thinkyhead, 2016-09-27:
Looks like one got missed:
```cpp
    //
    // Bed:
    //
    #if TEMP_SENSOR_BED != 0
      MENU_MULTIPLIER_ITEM_EDIT(int3, MSG_BED, &thermalManager.target_temperature_bed, 0, BED_MAXTEMP - 15);
    #endif
```

closed 2016-09-27 by thinkyhead