MarlinFirmware/Marlin #4913: Z_CLEARANCE_BETWEEN_PROBES  RCBugFix

Bettman66, 2016-09-28:
Z CLEARANCE BETWEEN PROBES does not work with bltouch.
G28 works beautifully, G29 does not work.
The z axis does not move up, after the first measurement point.

#define BLTOUCH
#define Z_ENDSTOP_SERVO_NR 0
#define Z_SERVO_ANGLES {10,90} // Z Servo Deploy and Stow angles
#define X_PROBE_OFFSET_FROM_EXTRUDER 45  // X offset: -left  +right  [of the nozzle]
#define Y_PROBE_OFFSET_FROM_EXTRUDER 0  // Y offset: -front +behind [the nozzle]
#define Z_PROBE_OFFSET_FROM_EXTRUDER -1.8   // Z offset: -below +above  [the nozzle]
#define Z_CLEARANCE_DEPLOY_PROBE   10 // Z Clearance for Deploy/Stow
#define Z_CLEARANCE_BETWEEN_PROBES 5 // Z Clearance between probe points
#define AUTO_BED_LEVELING_LINEAR
#define DEBUG_LEVELING_FEATURE
#define ABL_GRID_POINTS_X 3
#define ABL_GRID_POINTS_Y ABL_GRID_POINTS_X
#define LEFT_PROBE_BED_POSITION 45
#define RIGHT_PROBE_BED_POSITION 150
#define FRONT_PROBE_BED_POSITION 10
#define BACK_PROBE_BED_POSITION 200

with RC works bltouch great.

thinkyhead, 2016-09-28:
Let's take a look. Get the latest `RCBugFix`, enable `DEBUG_LEVELING_FEATURE` and issue the command `M111 S32` before doing `G28` followed by `G29 V4`. Post the log here and we'll be able to see why the raise isn't happening.

Bettman66, 2016-09-28:
[bltouch_log.txt](https://github.com/MarlinFirmware/Marlin/files/497122/bltouch_log.txt)

thinkyhead, 2016-09-30:
It looks like you have configured a "servo probe" but have not enabled the `BLTOUCH` option specifically for the BLTouch probe.

thinkyhead, 2016-09-30:
It looks like the moves are being attempted, but are not succeeding. Are you using 32x microstepping on your Z axis? What are your settings for `Z_PROBE_SPEED_FAST` and `HOMING_FEEDRATE_Z`? Do you see the same problem if you use a lower value for `Z_PROBE_SPEED_FAST`?

Bettman66, 2016-09-30:
I use 16x stepping, Z_PROBE_SPEED_FAST lower brought no change  (motor was slower and louder.).

working config rc : 
[Config_rc.txt](https://github.com/MarlinFirmware/Marlin/files/502990/Config_rc.txt)

not working config rc_bugfix:
[Config_rc_bugfix.txt](https://github.com/MarlinFirmware/Marlin/files/502991/Config_rc_bugfix.txt)

DatenThielt, 2016-09-30:
Getting the same issue but intermittently, sometimes it retracts, othertimes its hit or miss

thinkyhead, 2016-09-30:
Which endstop pins are your BLTOUCHes connected to? And are you using `Z_MIN_PROBE_ENDSTOP` also?

Notice that the `SERVO_DELAY` is overridden in `Conditonals_LCD.h` where `BLTOUCH` emulates a servo. It's lowered to 50. See if setting a higher value like 60 or 100 here helps.

If the probe is connected to the `Z_MAX` pin (the default) then possibly the raise move was interrupted because the probe was still registering as "triggered."

I'm working up a PR that adds conditions so the `Z_MAX` logic so `Z_MAX_PIN` won't _read on a raise_ in the case of when the min probe pin is usurping the `Z_MAX` pin.

DatenThielt, 2016-10-01:
Mine is connected to the Z_MIN, Not using Z_MIN_PROBE_ENDSTOP

I tested this changing the Servo Delay to 75, then 100 but neither effected it.

On an related note, when it does fail, it ALWAYS fails on the 6th to 7th probe (Using AUTO_BED_LEVELING_LINEAR) See video

https://youtu.be/EsS3PTQ8PHw

thinkyhead, 2016-10-01:
The BLTouch behavior was changed in recent weeks only to retract the BLTouch in-between probes. I can't see why a raise would be aborted, since the Z min endstop isn't checked for moves away from the bed. Do you see any difference if you do `G29 E`?  The E flag makes the probe stow and disable completely after each point, instead of just retracting. That might be more reliable.

I made a small patch to `RCBugFix` earlier but nothing likely to fix this issue. Still, grab the latest code before testing. I'd like to see the log output where you have 5 or 6 points succeeding before it fails.

Why does the BLTouch probe go into error condition? I couldn't tell what was setting it off.

So far the log indicates that the raise is being sent, but as your video shows, the move is not completing. Moves are not typically aborted, unless they are moves towards an active endstop that has been triggered. I can write code to test for that and add debugging. Please post your Configuration files also.

DatenThielt, 2016-10-01:
Tested with the latest but as you guessed, did not fix the problem. 

G29 E does seem to be more reliable but still gives errors (on G28 mainly G29 seems to be ok)

on Step 6, the BL Touch starts retracted, then Z goes down, the second the pin hits the bed it starts to move to step 7 then errors as the pin has not retracted.  

I have not got advance debugging on yet but config is attached  
[Configuration.txt](https://github.com/MarlinFirmware/Marlin/files/504408/Configuration.txt)

Jartza, 2016-10-01:
I can confirm this with the latest RCBugFix and BLTouch, for me even the first probe on G29 fails (using PROBE_DOUBLE_TOUCH) as after first hit to bed the Z-axis does not move up, but starts moving down again. The "bump" happens correctly with G28.

Jartza, 2016-10-01:
Also, strangely, after failing G29 and BLTouch reset, when doing homing again, the Z-axis moves down instead of up (the clearance making move).

Bettman66, 2016-10-01:
When I disable Z_SAFE_HOMING, G29 E works for me.

Jartza, 2016-10-01:
[bltouch_log.txt](https://github.com/MarlinFirmware/Marlin/files/504559/bltouch_log.txt)

Jartza, 2016-10-01:
https://www.youtube.com/watch?v=W00BZI9PcBE

Bettman66, 2016-10-01:
Hi Jari, can you test "G29 E" without Z_SAFE_HOMING ?

Jartza, 2016-10-01:
I get strange results and strange stop.

[bltouch_log.txt](https://github.com/MarlinFirmware/Marlin/files/504594/bltouch_log.txt)

Jartza, 2016-10-01:
Ahh. Z_SAFE_HOMING wasn't disabled. Strange.

Jartza, 2016-10-01:
Sorry. Had to revert back to commit 6335acbf5cc96f50109e987d8030b26702650a88 which I at least know did work, as I have some prints I have to get done. Will test with Z_SAFE_HOMING disabled later...

thinkyhead, 2016-10-02:
```
do_probe_raise(10.00)
Error:Printer stopped due to errors. Fix the error and use M999 to restart. (Temperature is reset. Set it after restarting)
```
Seems to indicate that the probe is showing up as "triggered" before being deployed.

bleugray, 2016-10-02:
having this same issue with the very latest rcbugfix.  i am using bilinear, doubletouch, z_safe_homing is enabled

thinkyhead, 2016-10-02:
Sorry @bleugray, the state of `G29` is precarious right now. The logs show that the raise move is being initiated, but then for some reason it seems to be getting interrupted, or in any case it isn't actually occurring. I will have to add yet more logging code to determine where the main cause lies.

bleugray, 2016-10-03:
let me know what i can do to help out testing.

Jartza, 2016-10-06:
Any news on this or more debugging needed?

genzo76, 2016-10-08:
Hello, now  I have try the new version (RcBugFix) have see most change, but the problem with "Z_CLEARANCE_BETWEEN_PROBES" not is solved.

genzo76, 2016-10-08:
I have a proposition for you, I have analyzed the code as much as I can.

The problem is due to the fact that when you make the raise of the plate, the current position is updated, but the movement has not actually occurred.

I do not know well the code, but for you it should be very easy to implement the solution that I thought:

You can use the same behavior that already exists for the Home of the Z, but in reverse.
I thought that when the probe has cut the first point, you can increase the coordinates of the Z axis until the switch Z does not turn off (a small offset may be useful to say when away after the endstop switch Z has disconnected).

In this way we will be sure that the deck is detached from the sensor.

I hope my solution will be useful.

Thanks for your work.

thinkyhead, 2016-10-12:
A feature of Marlin is that when a probe or endstops are enabled and triggered, the steppers simply stop, and then the position is incorrect. But there shouldn't be any interruption by endstops or probes when moving away from the bed. The stepper code makes sure only to test an endstop or probe when an axis is moving in the direction towards the endstop or probe. So when moving away from the bed, nothing should be getting triggered.

As you point out, the position is updated but the move doesn't take place. The only cause for a move to not occur once initiated is due to an endstop trigger, which means either the endstop/probe isn't being disabled, or the endstop flag isn't being cleared.

So far in looking at the code I haven't been able to find a causal link. I need to look through issues and see if it's just `BLTOUCH` being affected. It shouldn't be different from other probes, so it is a conundrum.

DatenThielt, 2016-10-12:
I am wondering if there is any kind of live debugging with breakpoints for the Arduino/Marlin (I come from a C# background and love visual studio) If I can find a way to do it I will run through the code tonight and see if anything sticks out whilst manually running through the code (I can see on google that there is a Visual studio Serial debugger for the Arduino)

Roxy-3D, 2016-10-12:
> So far in looking at the code I haven't been able to find a causal link. I need to look through issues and see if it's just BLTOUCH being affected. It shouldn't be different from other probes in this regard, so it is a conundrum.

I'm wondering if we should ask somebody with a BLTouch that is having problems to wire up an LED to its probe output.  All of these problems could be explained if the BLTouch is not sending the proper status to the AVR board.   I don't believe that is happening, but it certainly would explain all the flaky things being seen.   And it is very easy thing to do so we can concentrate the search else where.

Jartza, 2016-10-12:
> I'm wondering if we should ask somebody with a BLTouch that is having problems to wire up an LED to its probe output. All of these problems could be explained if the BLTouch is not sending the [...]

I don't believe so as I personally switched to older commit 6335acbf5cc96f50109e987d8030b26702650a88 which seems to work nicely.

timeshell, 2016-10-13:
@thinkyhead Further up you said:
"The E flag makes the probe stow and disable completely after each point, instead of just retracting. That might be more reliable."

Do you mean you are disabling the servo?  From what I've been told by ANTCLABS, the servo pin to the BLTouch should not be disabled.

thinkyhead, 2016-10-14:
@timeshell No, the servo pin doesn't disconnect. With `BLTOUCH` the servo is always left activated.

thinkyhead, 2016-10-20:
@Jartza I'm using that commit —or the date of around Sept 21— as the basis to compare all recent changes and find the one (or more) that's most likely to be involved. I'm about halfway through that process, so I should have some list of the most likely candidates soon.

thinkyhead, 2016-10-22:
Ok, so here's what I have. I made a new branch starting with https://github.com/MarlinFirmware/Marlin/commit/6335acbf5cc96f50109e987d8030b26702650a88 and applied all the changes from the current code to it. Then I went through and made commits with all the most innocuous changes, but leaving out the changes that seem most likely to be connected with this issue. Hopefully I didn't miss anything…

https://github.com/thinkyhead/Marlin/tree/rc_reverted_for_testing

Please test with this branch and see if it works as well as the code from Sept. 21. If it does work, then we should look at those few suspicious commits and figure out which one is the culprit. If it still fails, then I will revert some more changes and we can keep testing.

Jartza, 2016-10-22:
> Please test with this branch and see if it works as well as the code from Sept. 21. If it does work, then we should look at those few suspicious commits and figure out which one is the culprit. If it still fails, then I will revert some more changes and we can keep testing.

I just tested the bed levelling and the BLTouch probe is now lifted between probe points like it should (also the bump feature works), so at least that works now correctly. I didn't yet try printing, though.

Jartza, 2016-10-22:
RC_Reverted_for_testing, G28 & G29:

https://www.youtube.com/watch?v=81ONKZdOsl8

thinkyhead, 2016-10-22:
So from your testing, apparently the culprit has to be one of the changes in this commit that I've just added to the branch:

https://github.com/thinkyhead/Marlin/commit/65b6abf5549d9bda661c7ff90dd78bd6e628ebfd

So, try making some of those changes, one at a time, and see which one causes the failure.

Jartza, 2016-10-22:
> So, try making some of those changes, one at a time, and see which one causes the failure.

None of them, it seems. Applied them, one by one. Still working.

Jartza, 2016-10-22:
I even tried pulling the latest from your branch to make sure I didn't miss anything from the patch. Still works.

Jartza, 2016-10-22:
Now the difference between RCBugFix and RC_reverted_for_testing is pretty small:

[rcbugfix.diff.txt](https://github.com/MarlinFirmware/Marlin/files/545856/rcbugfix.diff.txt)

Jartza, 2016-10-22:
I'm now a bit baffled. Gonna try the absolutely latest RCBugFix and see if it works with it.

thinkyhead, 2016-10-22:
Hmm!

Jartza, 2016-10-22:
For reason or another, it now seems to work with the latest RCBugFix too!

Anybody else willing to confirm?

DatenThielt, 2016-10-22:
Give me half hour and I will be able to confirm

DatenThielt, 2016-10-22:
So this is now even more of an issue with me than it was before.  :\  I used to be able to level the bed maybe 1 out of 10 tries, this time it fails on the first probe.  :( tell me what you need.

DatenThielt, 2016-10-22:
Ok so, I have the exact same issue as before,  Fails 90% of the time on probe 7 out of 9 on the bed level.  and is intermittent with G28

Jartza, 2016-10-22:
This is strange. For me G28 & G29 -combination now works.

thinkyhead, 2016-10-24:
@DatenThielt Do you see the problem in the branch linked below? It's a version of Marlin where I've left out any recent changes that seem likely to be involved.

https://github.com/thinkyhead/Marlin/tree/rc_bltouch_debug

DatenThielt, 2016-10-24:
@thinkyhead Tested with the rc_bltouch_debug and G28  G29 and now working 100% of the time, So it seems it it something in one of those commits, Let me know if you need anything

thinkyhead, 2016-10-24:
@DatenThielt I've just applied a patch to `RCBugFix`. Please test and see that it works now.

DatenThielt, 2016-10-24:
So just tested the new RCBugFix  I am getting about 30% failure on G28 (3/10 tries fails),  G29 seems to be working perfectly though!

DatenThielt, 2016-10-24:
Just to elaborate, G28 failes on the 2nd probe, First probe extends, Z goes down and then retracts, then instantly extends again hitting the bed and causing an error (As no Z raise happened)

thinkyhead, 2016-10-24:
@DatenThielt Are you actually using the `BLTOUCH` probe option? It doesn't (any longer) instantly re-deploy the pin. In the latest code, the `BLTOUCH` is a special case. The probe should not redeploy until just before starting the downward probe. If it seems to be deploying early, perhaps your Z homing bump is set too small.

DatenThielt, 2016-10-24:
@thinkyhead I am using the BLTouch probe option with the latest RCMarlinBugfix as of 2 hours ago.  My Process.

Turn Printer On -> Prepare -> Auto Home -> (Fail) -> Prepare -> Auto Home -> (Success) I have included a video of this process.  Once again it only fails about 30% of the time. 

Video: https://youtu.be/zPD4lmpy-Wc
Configuration.h: http://pastebin.com/ccivsNY7

thinkyhead, 2016-10-24:
In the video it appears not to even register the second probe. The Z axis continues to move down. This indicates to me that the probe is malfunctioning.

thinkyhead, 2016-10-24:
I notice that your second probe speed is really, really slow.
Try making it faster and see if it makes any difference.

DatenThielt, 2016-10-24:
How would I increase the second speed sorry?

thinkyhead, 2016-10-24:
⏫

DatenThielt, 2016-10-24:
That seems to have done it!  Appears to be fully functional and reliable 👍

thinkyhead, 2016-10-24:
Good to know! My guess was that the BLTouch throws an error if it's enabled for too long without getting triggered.

[2 more replies]

closed 2016-10-24 by thinkyhead