MarlinFirmware/Marlin #4916: Z_DUAL_STEPPER_DRIVERS won't home

Kaiwol1990, 2016-09-28:
Since I've updated to RC7 my z axis won't home anymore. It's just moving up a liitel bit and then stops.
I cheked for Endstops etc. IF I move the axis over the manual controls it moves in both directions. If I change back to RC6 everithing works like a charme, so i think theres an Buck somewhere.

My Setup;
CoreXY
RUMBA Board
Dual Z
DRV8825 Drivers

Kaiwol1990, 2016-09-30:
So i have investigated a little bit more and it seems that the bug only occurs when i have 
#define Z_DUAL_ENDSTOPS active.

And if so the M119 command dosn't responde an z2 endstop, but if I change the following lines in Conditionals_post.h it shows up, but still dosn't home (it just moves up a little bit).
      ` #define Z2_MAX_PIN 36`
      `//#define Z2_MAX_PIN X_MAX_PIN`
      `#undef USE_XMAX_PLUG`

And if I use only one Endstop for Z_Max erverthing works fine.
I hope sombody can help, because im note very into C++ programming.

thinkyhead, 2016-10-02:
Do you see the same issue if you use `RCBugFix`? We might have fixed this issue in the interim since `RC7` was released.

Kaiwol1990, 2016-10-02:
Yes still the same if i try the `RCBugFix`

thinkyhead, 2016-10-27:
@Nocturnal42 seems to have nailed it. The issue has been patched in `RCBugFix` and another PR #5097 is pending to prevent the wrong endstop bits being set when Z2 is triggered.

Kaiwol1990, 2016-10-28:
Good work !

it works fine now if i use the same setting as @Nocturnal42. But i got still some strange errors:
1. it only works with the Y_Max pin, if i use the X_Max pin it still just moves up a little bit an won't home.
2. i don't get an Z2 endstop reading if i send an M119, unless i aktivate  
                  `#define USE_YMAX_PLUG`
in configuration.h

But other than that it works now for me. 
Thanks for the work

Nocturnal42, 2016-10-29:
Works with both for me. Have you tried it with `USE_XMAX_PLUG` enabled? You need to have the plug enabled in order to use something attached to that port, that would also be why it doesn't show in with a M119. Sounds like it needs some more sanity checking.

thinkyhead, 2016-10-29:
> But i got still some strange errors

@Kaiwol1990 Post your updated configuration if you still haven't solved them.

Kaiwol1990, 2016-10-29:
Yeah had  USE_XMAX_PLUG enabled and it's working well, also shows me z2 when i send an M119, but also shows xMax. So just an "cosmetic" bug.

Will try it later today again wit Y_Max to be sure it wasn't an simple mistake and also post my configuration files.

thinkyhead, 2016-10-29:
> but also shows xMax

That's odd, because in `Conditionals_post.h` we have:
```cpp
  #define IS_Z2_OR_PROBE(P) (P == Z2_MIN_PIN || P == Z2_MAX_PIN || P == Z_MIN_PROBE_PIN)
  . . .
  #define HAS_X_MAX (PIN_EXISTS(X_MAX) && !IS_Z2_OR_PROBE(X_MAX_PIN))
  . . .
  #define HAS_Z2_MAX (PIN_EXISTS(Z2_MAX))
```
…and then in endstops we have:
```cpp
void Endstops::M119() {
  SERIAL_PROTOCOLLNPGM(MSG_M119_REPORT);
  . . .
  #if HAS_X_MAX
    SERIAL_PROTOCOLPGM(MSG_X_MAX);
    SERIAL_PROTOCOLLN(((READ(X_MAX_PIN)^X_MAX_ENDSTOP_INVERTING) ? MSG_ENDSTOP_HIT : MSG_ENDSTOP_OPEN));
  #endif
  . . .
  #if HAS_Z2_MAX
    SERIAL_PROTOCOLPGM(MSG_Z2_MAX);
    SERIAL_PROTOCOLLN(((READ(Z2_MAX_PIN)^Z2_MAX_ENDSTOP_INVERTING) ? MSG_ENDSTOP_HIT : MSG_ENDSTOP_OPEN));
  #endif
  . . .
} // Endstops::M119
```

thinkyhead, 2016-10-29:
That's odd, because in `Conditionals_post.h` we have:
```cpp
#define IS_Z2_OR_PROBE(P) (P == Z2_MIN_PIN || P == Z2_MAX_PIN || P == Z_MIN_PROBE_PIN)
. . .
#define HAS_X_MAX (PIN_EXISTS(X_MAX) && !IS_Z2_OR_PROBE(X_MAX_PIN))
. . .
#define HAS_Z2_MIN (PIN_EXISTS(Z2_MIN))
```
So the `HAS_X_MAX` flag should be false and prevent the display of the "X max" in `M119`.

Nocturnal42, 2016-10-29:
When did you last refresh the RCBugFix tree? I believe there is a time gap between the two commits where you'd get one fix but not the other.

Kaiwol1990, 2016-10-29:
Realy seems to be strange. As i said im gone take a look later today, i'm not near my printer by now.
Maybe it's already fixed, hadn't implemented  #5097  at first.

Refreshed yesterday around 11 PM (GMT+01).

Kaiwol1990, 2016-10-29:
I may have found it, the lines you postet above aren't implemented in my Conditionals_post.h.
So i will download the newest Version later and try again.

Thanks for the help

Kaiwol1990, 2016-10-29:
So had time to get to my printer and it works, the new RCBugFix takes care of everthing.

Thanks for your help and commitment!

thinkyhead, 2016-10-29:
@Nocturnal42 `RCBugFix` gets "refreshed" on every commit or merge.

thinkyhead, 2016-10-29:
@Kaiwol1990 Good news! Thanks for the update.

closed 2016-10-29 by thinkyhead