MarlinFirmware/Marlin #4917: moving axis will restart

diver197, 2016-09-29:
I use the 256 subdivision drive module, set the parameters, the moving axis will restart

thinkyhead, 2016-09-29:
Too many steps per mm will prevent other functions from getting any time. If you have `WATCHDOG` enabled, this will lead to a reboot.

diver197, 2016-09-30:
Did not restart, but the work is not normal, is not supported by 256 subdivision？

thinkyhead, 2016-10-01:
Arduino 8-bit AVR simply cannot generate enough pulse signals to keep up with 1280 steps per mm (256*200/40). I've been working with a machine configured to ~1778 steps per unit (a geared-down SCARA robot: (200steps*32microsteps*100gear)/360°), and that was using up too much time to drive at reasonable speeds. I reduced microstepping down to 8x to get (200*8*100)/360=444.44 steps per degree, so it could run reliably with some overhead. We are finding that 32x microsteps is near the limit of what Marlin can support before the CPU simply can't keep up.

closed 2017-10-31 by thinkyhead