MarlinFirmware/Marlin #4920: Motors flickering

i23d, 2016-09-29:
Hi there. I am new at Github so I apologize in advance if i make few mistakes in the aproach to the problem, to obvious questions or grammer mistakes. 

Me and my friends build a custom 3D printer / CNC. 

The only unresolved problem that we have and it is unabeling us to print is the flickering of the motors on y and x axes. 
After we home it (on Printrun didn't tryed other programs) after few seconds you can hear and see that the motors on y and x axes are flickering a bit (not sure for the right verb?). You can see that the motor on y axes is flickering harder. The interesting part is that on the y axis we have dual motors and usually only one is flickering. The motor on the x axes always starts to flicker few seconds after the motor on y. 
We checked the wireing, and we tried to release the belt from the motors so we think it is not mechanical or electrilcal but firmware. 

To help explain the problem and give you more clues I uploaded a video on youtube. 
[https://www.youtube.com/watch?v=3B07D6gLGdQ](https://www.youtube.com/watch?v=3B07D6gLGdQ )
The flickering starts at 1.11 After homing on Printrun I have not touched anything. 

Hope someone has good ideas about this and how to find the problem. 
Thank you in advance.

thinkyhead, 2016-09-30:
What version of Marlin are you using?

i23d, 2016-09-30:
Marlin 1.1.0-RC7

norm8332, 2016-10-01:
Almost looks like the stepper drivers may be over heating and cutting in/out.

thinkyhead, 2016-10-02:
Stepper drivers will fail at something pretty low, like 190°C so they should be heat-sinked, or the board itself may have a layer for heat-sinking.

The wiring for stepper motors usually has the center wire pair swapped. So, if the two center wires were yellow-green at the driver end, it will be green-yellow at the motor end. I'm not sure how that applies to motors with 6 poles.

i23d, 2016-10-14:
The board and the power units are separated from the printer with their cooling so we doubt that the problem is overheating. In the meantime we checked the conectors, connect a new power unit,  don't turn on the heaters but the problem persist.

thinkyhead, 2016-10-24:
The thing is, when there are no movements underway, Marlin doesn't send any signals to the steppers. But, try the `RCBugFix` branch anyway to see if it behaves any differently. If that doesn't help, post your configurations so we can check to see if any of the other enabled features could be involved.

i23d, 2016-10-25:
Hi. 
We have found the problem and norm8332 was right :D
One colleague adjusted the current on 250 for some testing without telling anyone and didn't put it back. 
Thank you very much.

closed 2016-10-25 by i23d