MarlinFirmware/Marlin #4922: PID_FUNCTIONAL_RANGE value hardcoded

asm-wolf, 2016-09-29:
While trying to tune the PID loop on my printer by hand (I've newer trusted auto-tuners) I noticed that the loop behaves unexpectedly and overshoots with the derivative gain set low, even when the other gains are low (EG. p=0.01 i=0 d=0). I found the following in Configuration.h

`#define PID_FUNCTIONAL_RANGE 10 // If the temperature difference between the target temperature and the actual temperature
                                  // is more than PID_FUNCTIONAL_RANGE then the PID will be shut off and the heater will be set to min/max.`

I would not say that I'm particularly well versed in PID control theory, but have read books on the matter. From my experience coding and tuning PIDs by hand I've found that this is best set to the same value as the loop's proportional bandwidth (PID_FUNCTIONAL_RANGE = PID_MAX/Kp) rather than any hard-coded arbitrary value. Alternatively I've found that loops behave well if integration is disabled whenever the total output would be greater than the max output (I.E. PID_MAX). Is there a reason that I've overlooked that this is static instead of being calculated based on Kp?

I'm not certain how changing this will affect users' tuning, but using the example [here](http://reprap.org/wiki/PID_Tuning) of Kp = 19.56  and assuming PID_MAX = 255 this gives a functional range of about 13.0C which is in the ballpark of 10.

Rerouter, 2016-09-30:
asm-wolf, care to post some images of your temperature behaviour, I'm currently having a solid dig into the PID code, (and some upcoming fixes will remove / reduce the need for this constraint)

Also see how your values differ from the M303 autotune, e.g.  M303 E0 S215 C8 you dont have to use the values, but more something i can compare off,

asm-wolf, 2016-09-30:
The behavior seems to occur over a longer time frame than I can get pronterface to show on the graph all at once. Can you recommend a way to adjust this or some other software? (I'm running arch linux)

With the above example of P=0.01 I=0 D=0 it appears as an exponential curve with a little overshoot followed by some sinusoidal  oscillations that don't seem to dampen. I'd expect with no I/D gain and a small P gain that it would show an exponential curve that would approach a value below the set point with no oscillations. The autotune gives adequate performance with some sinusoidal oscillations that eventually dampen. It's not a major issue, but it would be nice to manually optimize this with lambda or manual tuning. I find the D gain that the ziegler nichols method arrives at is often aggressive and many first-order loops perform better with little or no D gain and well chosen P and I values, but with the above constraint on the active region this is not possible.

Thanks for the help!

AndyZuerich, 2016-10-01:
The general problem with the PID algorithm in Marlin is, that PID does not behave very well when there is a time delay (dead time + time constant) involved in the plant (hardware device). I don't know if you have realized this as well: When you use the auto-tuned PID parameters and modify it with the Ziegler-Nichols no-overshoot, then I would have expected that there is no overshoot in the step response. But my experience is, that Marlin shows still a lot of overshoot.

For a heater or hotend,  the plant behavior is typically a "FOPTD" (= first order + deadtime) process. I found that we could HUGELY improve the PID behavior by using something called a "Smith predictor". Please have a look at this article:
[matlab smith predictor](https://ch.mathworks.com/help/control/examples/control-of-processes-with-long-dead-time-the-smith-predictor.html).
This approach requires knowledge about the heater behavior:
- Approximated value for the lag time (time that passes from starting the heating process until the heat reaches the temperature sensor)
- The time constant of the exponential temperature curve
- the gain (output value -> steady-state temperature).

I think all of these variables could be easily extracted.

@asm-wolf : On your question with the PID_FUNCTIONAL_RANGE : Do you know any reason why this value/feature should be here at all? My impression is that we should remove the functional range constraint at all. When PID is implemented with anti-windup, there should be no reason for limiting the functional range. Removing this feature would implify the code, which is always a good thing.

Rerouter, 2016-10-01:
I've dug deeper into what explains your description, and it comes squarely down to limited output resolution, at the moment it is 7 bits of output resolution for the PWM, so far the only reason i can find for this was a bodge to line up with what it outputs to the host software, I will have to recheck what pins the controller boards use, but this can be increased to 8 bit easily, and with a bit more digging 16 bit if they use the right pins. 

So not the functional range limitation, but purely resolution, 
There are also non linear aspects of the system that make this limitation cause wierd oscillations,
for instance on my current hot end heater cartridge it is 3.2 ohm at 25C, and thanks to temperature coefficients is 5.5 ohm at 220C, or about 35W cold, about 75W hot, all this non linearity ends up adding to roughly 0.5W added to the system per PWM step, and equally the steady state temperature changes by roughly 5C per step on my geetech stock hot end, (kind of a big step change eh?) 

Improving to 8 bit would result in my system (which i will take to be fairly generic) to 2.5C per step, still not that great but an improvement, and 16 bit would take it to such a tiny divisor that it stops mattering, equally as the PID routines already use float, converting to a 16 bit value really doesnt change the performance at all. (this part i will likely need a main devs assistance with)

I'll submit a pull request that removes the functional range code tommorow, i have stipped it out personally, and tested it with no sign of issue, this goes along with the integral constrain removale i pull requested earlier

Blue-Marlin, 2016-10-01:
> There are also non linear aspects of the system that make this limitation cause wierd oscillations,
for instance on my current hot end heater cartridge it is 3.2 ohm at 25C, and thanks to temperature coefficients is 5.5 ohm at 220C, or about 35W cold, about 75W hot, all this non linearity ends up adding 

@Rerouter 
Please recalculate. A heater with a negative temperature coefficient would surprise me (and the rest of the world).

Blue-Marlin, 2016-10-01:
With a 'tuned' functional range one can minimise the initial overshot. Even a negative overshot is possible. I use it for minimising the heat up time.

Blue-Marlin, 2016-10-01:
A AC-heater can be switched with 2*net_frequency. The SSRs can be switched of only at 0V.
At DC, with a increased pwm_frequency you will switch the mosfet more often - it will be more often in the 'analoge'  state - it will be hotter.
For a RAMPS the bed-pin is the only one you could use with analogWrite. The other pins do use timers needed by Marlin. Ruining the stepper IRQ-timer is not a good idea.

Rerouter, 2016-10-01:
Sorry had my divider wrong, E^2/R, so 45W at cold, 26W at hot, which drops it down to about 0.3W per PWM step, the temperature change per open loop per step remains the same at roughtly 5C. so the issue still remains,

Looking at the GT2560 controller board, they use timer 3 for the 2 hotend outputs, and timer 0 for the heated bed, (0C3B and 0C3C, and the bed 0C0B), so flat out that leaves the heated bed at 8 bit, but both the hot ends are on a 16 bit timer, the PWM is done by hardware and looking at the registers we should be able to flick those 2 to 10 bits with only small modifications,

Looking at the ramps 1.4, it uses D8, D9 and D10, for its heater outputs, or 0C4C, 0C2B and 0C2A, only timer 4 on him is on a 16 bit timer, a shame, 

Still to this end, i am planning to bring it back to atleast 8 bits, and will have a dig into what is required to change the PWM hardware that can run at a higher bit level to actually use it, from the pins file we know if a heater is on a 8 or 16 bit timer, so should not be that hard, if the hardware doesn't specify, say adding later 32 bit platforms it can default to 8 bit, (or atleast that is my take on it, my firmware background is still limited)

asm-wolf, 2016-10-01:
@AndyZuerich When the loop is outside of the proportional bandwidth (as calculated above) the theoretical output due to the proportional gain is above 100% (by definition), so it does not make sense to use the calculated PID values at that time. This may cause the integral term to wind up a lot and as @Blue-Marlin says this can cause overshoot. Setting this as the functional range will ensure a smooth transition from being 100% on to being PID controlled. There may very well be a better way of doing this, but I've found this works well. The overshoot you see when you use the Ziegler-Nichols no-overshoot rules is likely due to the constraint on the functional range causing the heater to be on for too long. I've not used a smith predictor before but it should help in this scenario. Personally I'd try and tune the loop using something other than Ziegler-Nichols first, but I don't think many users would bother with that.

@rerouter Would you mind sharing your config when you get something working as I have the GT2560 board also.

asm-wolf, 2016-10-01:
@Rerouter I just noticed your statement on resolution:

> So not the functional range limitation, but purely resolution

FWIW I don't think that this is purely the cause for overshoot on my hotend. With the gains that I used the heater has 0.01*10=0.1 counts upon entering the functional range and turns off immediately as a result, but still overshoots. With the functional range larger this can be prevented. Also FWIW I use this gain value as an exaggerate example — not for practicality — as regardless of the underlying logic I would not expect overshoot with such tiny gains.  Upon further consideration I don't think that the derivative action is relevant here like I first thought. The low resolution is certainly an interesting issue though and I agree that this may still cause problems.

Rerouter, 2016-10-01:
Ok pull request created #4929 for the removal of the functional range, I'll leave it to the devs to discuss if they wish to use it, 

Now working on 8 bit. 

@asm-wolf, the board is already in the system, use "MOTHERBOARD 7", for any other settings i would have to know that you are using the same system for my settings to apply (G2S Pro)

asm-wolf, 2016-10-02:
@Rerouter Looks good. Is there anything that stops the integral term from winding up when the system starts from cold?

Thanks for the tip about the board. I'v had trouble getting mine to talk to avrdude with either the bootloader or my USPasp, but a spare arduino programmed as an AVRISP seems to like it. I think that mine shipped without the bootloader installed. I'll reflash the board and give it a go when you have finished with the PID code.

Thanks for taking a look at this.

thinkyhead, 2016-10-02:
@Rerouter I'm mulling whether it makes sense to keep `PID_FUNCTIONAL_RANGE` as an optional setting, but have the default set to 999 and cap the minimum to ~10. Some (@Blue-Marlin) did express a preference for keeping it. But if using PID straight across produces more or less the same output profile as "functional range" (or a leaner profile), then we'd only keep `PID_FUNCTIONAL_RANGE` around as an option for "historical reasons."

Rerouter, 2016-10-02:
@asm-wolf The integral windup was resolved with a previous pull request, meaning the auto tune leaves you with roughly a "Quarter amplitude decay" when approaching any new set point, likely a 5-8C overshoot from cold 

@thinkyhead Blue would have been using it as a form of conditional PID, with different behaviors when further or closer to set point, to that extent it appears it was the reset of the I term, and pid_output, that caused his desired response, preferably @Blue-Marlin could post some screenshots comparing with and without (with the current patches) to see if it still matches his requirements (to my own setup the difference was minimal)

The issue that i am now left with is are there any PID approaches better suited to what we are trying to do, that has a fast heat up and response time without making the loop twitchy or unresponsive, (reading over my process control text books).

AndyZuerich, 2016-10-03:
> The issue that i am now left with is are there any PID approaches better suited to what we are trying to do, that has a fast heat up and response time without making the loop twitchy or unresponsive, (reading over my process control text books).

Yes, there are much better alternatives than pure PID. I was reading about that a lot in the last days.
PID control is something that is the workhorse of simple control tasks, and is easy to implement and understand. But: For some applications, PID is reacting way to slow. Or in other words: For a blackbox application, in order to limit the overshoot an minimize ringing, PID's system performance has to be limited to be very slow.

At the moment, our PID algorithm knows nothing about the physical nature of the heating process. I.e. our PID uses the heater as black-box, and with Ziegler-Nichols or manual tuning, we try to get some kind of good response for our control problem.
The solution for an immense improvement would be, that we assume some model of the heating process. Typically, for the heated bed or hot end, the (linear assumed) step response looks like
T(t) = Tend - gain * exp(-t / tau) (eventually + a delay term, when the sensor is located far away from the heater itself).
This curve shape always occurs in heating applications. And at the moment, we don't take any advantage of this knowledge. We just use PID for some black-box application.

Now, there are some approaches in control theory available which consider the "device" behavior. The keyword for this is called "Internal model control". It works the following way:
- There is a model for the device/heater existing, which basically has time constants for the delay, time constant and gain. These values could also be extracted by a simple experiment (or auto-tuning).
- The controller sends its output to both, the hardware (heater) and the model (T(t), or the s-domain transfer function). Then, the controller input does not react on the actual temperature measurement, but on the difference between the model result and the temperature measurement. With this difference, the model knowledge and its effect on the system is considered.

What kind of difference does this make? Here an example:

- the PID, as we use it now is in steady state. Then, something happens with the output (e.g. increasing the extrusion rate, or enabling the blower fan). Then PID suggests a reaction. But nothing, or only little happens (due to the heater time delay + eventually dead time). PID suggests slowly a stronger reaction (integrator) until something happens with the measured temperature, ...
- Internal model control: Steady state. Then, the output load changes, which is leading to a temperature measurment difference / error  from the set point. The internal model control algorithm with PID suggests a control action. Nothing happens (or only little reaction). PID does not further act, since the time delay and dead time is known to the controller. The smith predictor only acts, when the expected temperature after the delay time still (or even more) deviates from the set point temperature.
Finally, PID with internal model control can have a much larger PID gain and better control reaction time compared with the normal PID process.
In summary: My impression is that there can be done a large improvement of the PID control algorithm with taking into account the device / heater model. What I don't like about the PID with smith predictor:
- When a dead-time is included in the model (this is the time after which a model reacts at all), then we would need some buffer memory to produce the delayed temperature readings. But memory is hardly available on the Arduino. However, I think storing something like 10 8-bit values should be sufficient, here.
- It still has 3 PID parameters (P,I,D). Additionally to that, we need some more parameters, the model parameters gain, time-constant and possibly dead time. These parameters could be "measured", e.g. when the heater starts up from room temperature to the setpoint temp.
- When the new 3 parameters are not chosen the best, the system could again become instable 
or show ringing due to the model vs. reality mismatch.

This Matlab-related article shows impressively the advantage of PID with internal model control and smith predictor:
[smith predictor](https://ch.mathworks.com/help/control/examples/control-of-processes-with-long-dead-time-the-smith-predictor.html)

At the moment, I am reading about something called the "Dahlin controller", which is used in industry for exactly such kind of process as we have in the heaters (heating with time delay / dead time). As far as I understood, the Dahlin control has the following advantages compared to a smith predictor with PID:
- It still behaves like a PID controller
- There is only 4 parameters to set/know about: Dead time, time constant (tau) and the gain (steady-state connection output value -> temperature). The 4th parameter is the only tuning constant, which allows to adjust the settling time/overshoot. The other three parameters are physically measureable.
My impression is that this "Dahlin controller" is the best we can/could with reasonable effort, and the usage of the system would be not more complicated than before with pure PID. But the control results and response time might be much better.

Roxy-3D, 2016-10-03:
@AndyZuerich

> The controller sends its output to both, the hardware (heater) and the model (T(t), or the s-domain transfer function). Then, the controller input does not react on the actual temperature [...]

In your reading...   Is it difficult for our tuning process to measure the 'impulse response' of the transfer function in the s-domain ?   I don't know that this is true, but it would seem if we can measure the 'impulse response' in the s-domain we get all the variables we need when we do the transformation back to real time domain.

AndyZuerich, 2016-10-03:
> In your reading... Is it difficult for our tuning process to measure the 'impulse response' of the transfer function in the s-domain ? I don't know that this is true, but it would seem if we can [...]

Well, I am not a control Guru... but the following should apply: The impulse response is nothing else than the reaction of the system to a step function. There are some nonlinearities involved, e.g. the wire resistance with temperature, temperature-> sensor mappings, etc. But when we consider for instance a hotend temperature of 200 degrees, the impulse response at this temperature can be simply extracted by testing the system with an open loop control step in a limited range, e.g. stepping from 200 to +- 5 degrees.
Or another alternative for extracting the relevant parameters:
Use a bang-bang control for limited time (e.g. when heating up from room temp, and for example for 2-3 bang-bang cycles). With the ratio between the on- and off periods we can calculate the gain (steady-state PWM duty ratio that would be required for a continuous operation). The gain should, in a thermal model, be equivalent to the thermal resistance to ambient. The (initial) slope in bang-bang mode should be some measure for the time constant (or "heat capacity", which can be related to the time constant by tau = R * C). I am still thinking and reading about all that stuff, maybe I will have a better answer or suggestion in the next days.

Roxy-3D, 2016-10-03:
> The impulse response is nothing else than the reaction of the system to a step function. There are some nonlinearities involved, e.g. the wire resistance with temperature, temperature-> sensor mappings, etc. 

Yes.  This is true.   However, within the s-domain, that impulse response becomes a delta function.  It is no longer a step.  It is a spike.       The reason why I'm asking if we can measure the impulse response in the s domain is we would learn about the time delays in the system as well as the gain at different temperatures.

> I am still thinking and reading about all that stuff, maybe I will have a better answer or suggestion in the next days.

As you read...   try to keep in mind that the s-domain is like an alternate universe.   Mathematically, the reason for doing a transformation from one domain to another is because some problems can be more easily measured or solved in one domain than another.   I'm not 100% sure this is true, but I think if we can measure the 'impulse response' of the heater system within the s-domain, we get all the magic numbers we need to perfectly configure the control system in the time domain.

Blue-Marlin, 2016-10-03:
[image]
Tuned for 215°.
The numbers are the related PID_FUNCTIONAL_RANGE.
Tuning still works. 10 looks very well to me - fast and low amplitude swing in - less overshot then unlimited PID (100)
[image]
10.5 is looking quite perfect.

asm-wolf, 2016-10-03:
@Rerouter I've taken a look at the anti-windup code in the latest revision of temperature.cpp and it looks like it should behave as I would expect (and was trying to describe) now, so I'm happy with this. Whether we can do better than pure PID is something that's starting to get a little out of my depth here, but I would expect so.

@Roxy-3D 

> As you read... try to keep in mind that the s-domain is like an alternate universe. Mathematically, the reason for doing a transformation from one domain to another is because some problems can be [...]

This is correct; both the time domain and the s-domian give a complete description of a system and negating any numerical errors in calculations/transformations etc. you can freely go from one to the other. Most (all?) calculations in one domain have isomorphisms in the other. There is also a duality property between the time/frequency domains that I _think_ also applies to the s-domain too.

closed 2017-10-31 by thinkyhead