MarlinFirmware/Marlin #4923: Ver 1.1.0 RCBugFix Delta printer Z tower not setting manual home offset

str8up2, 2016-09-29:
Hello all,
I've been trying to reset my printers values sense this upgrade. I noticed that X and Y tower go to mechanical home then step away the offset value - they home before Z. When Z homes there is no pause - all 3 towers move the 40 mm safe distance before the process starts. I issue a M666 X-1.40 Y-1.30 Z-4.50. As you can see the Z value should crash the nozzle - it makes no difference what value I place after Z. When I check with M503, the new value is listed. I tried to issue a M500 just to check but still no physical change.
Am I missing something?  
Thanks for all you time and effort.

thinkyhead, 2016-10-01:
Are you using the Marlin `RCBugFix` branch?

The arguments to `M666` determine the offsets for the tower carriages, where X=Tower 1, Y=Tower 2, and Z=Tower 3. The Z offset only affects the relative position of the carriage on Tower 3. I would expect setting it to -4.5 would tend to skew the positioning a lot.

The usual order of G28 homing on Delta is:
- Move all 3 carriages up together until one of the endstops is hit
- Home the "X" carriage - Tower 1 - Back off by `endstop_adj[X_AXIS]`.
- Home the "Y" carriage - Tower 2 - Back off by `endstop_adj[Y_AXIS]`.
- Home the "Z" carriage - Tower 3 - Back off by `endstop_adj[Z_AXIS]`.
- Set `current_position` to the configured home position
- Move nozzle down to the top of the print region

What part of that process seems amiss to you? Is the last move of the "Z" carriage not occurring, or is it simply blending into the final move down to the print region? Are you saying that the "Z" value of "-4.50" is a test value that you expect to misalign the "Z" carriage more than it is? As far as I can see the move is being applied.

If you want to see in more detail what's happening, enable `DEBUG_LEVELING_FEATURE` and then do `M111 S247` before you do `G28`. That will show in more detail what's going on, including where all the moves are being applied.

str8up2, 2016-10-01:
I do agree with you on the order of G28. That is the way my older version acted. With 1.1.0RCBugFix Z is not responding to any change of offset. As I stated - the value of -4.50 should have crashed onto the bed at the Z tower. What I noticed is the "X" homes and backs off then "Y" homes and backs off "Z" homes and does not back off - at that point all 3 towers drop to the 40mm safe home zone then the print begins. Is there a "define" statement I can comment out to disable the 40mm start point?

str8up2, 2016-10-01:
The -4.50 value was a test to check the offset command.  I will enable DEBUG_LEVELING_FEATURE to test and get back with results.

str8up2, 2016-10-02:
Ran the debug report using G28. Here are the results.
276.25 is my total Z distance.
READ: >>> gcode_G28
READ: Machine Type: Delta
READ: Probe: NONE
READ: current_position=(0.00, 0.00, 0.00) : setup_for_endstop_or_probe_move
READ: > endstops.enable(true)
READ: current_position=(0.00, 0.00, 0.00) : sync_plan_position
READ: >>> homeaxis(X)
READ: current_position=(0.00, 0.00, 0.00) : sync_plan_position
READ: current_position=(0.00, 0.00, 0.00) : sync_plan_position
READ: current_position=(0.00, 0.00, 0.00) : sync_plan_position
READ: > endstop_adj = -0.80current_position=(10.00, 0.00, 0.00) : 
READ: current_position=(0.00, 0.00, 0.00) : sync_plan_position
READ: <<< homeaxis(X)
READ: >>> homeaxis(Y)
READ: current_position=(-0.80, 0.00, 0.00) : sync_plan_position
READ: current_position=(-0.80, 0.00, 0.00) : sync_plan_position
READ: current_position=(-0.80, 0.00, 0.00) : sync_plan_position
READ: > endstop_adj = -0.80current_position=(-0.80, 10.00, 0.00) : 
READ: current_position=(-0.80, 0.00, 0.00) : sync_plan_position
READ: <<< homeaxis(Y)
READ: >>> homeaxis(Z)
READ: current_position=(-0.80, -0.80, 0.00) : sync_plan_position
READ: current_position=(-0.80, -0.80, 0.00) : sync_plan_position
READ: current_position=(-0.80, -0.80, 0.00) : sync_plan_position
READ: > endstop_adj = -1.80current_position=(-0.80, -0.80, 10.00) : 
READ: current_position=(-0.80, -0.80, 0.00) : sync_plan_position
READ: <<< homeaxis(Z)
READ: >>> set_axis_is_at_home(X)
READ: For X axis:
READ:  home_offset = 0.00
READ:  position_shift = 0.00
READ:  soft_endstop_min = -105.00
READ:  soft_endstop_max = 105.00
READ: > home_offset[X] = 0.00
READ: current_position=(0.00, -0.80, -1.80) : 
READ: <<< set_axis_is_at_home(X)
READ: >>> set_axis_is_at_home(Y)
READ: For Y axis:
READ:  home_offset = 0.00
READ:  position_shift = 0.00
READ:  soft_endstop_min = -105.00
READ:  soft_endstop_max = 105.00
READ: > home_offset[Y] = 0.00
READ: current_position=(0.00, 0.00, -1.80) : 
READ: <<< set_axis_is_at_home(Y)
READ: >>> set_axis_is_at_home(Z)
READ: For Z axis:
READ:  home_offset = 0.00
READ:  position_shift = 0.00
READ:  soft_endstop_min = 0.00
READ:  soft_endstop_max = 276.25
READ: > home_offset[Z] = 0.00
READ: current_position=(0.00, 0.00, 276.25) : 
READ: <<< set_axis_is_at_home(Z)
SENT: M105
READ: current_position=(0.00, 0.00, 276.25) : sync_plan_position_kinematic
READ: current_position=(0.00, 0.00, 276.25) : (DELTA)
READ: >>> do_blocking_move_to(0.00, 0.00, 204.47)
READ: destination=(0.00, 0.00, 276.25) : set_destination_to_current
READ: destination=(0.00, 0.00, 204.47) : prepare_uninterpolated_move_to_destination
READ: current_position=(0.00, 0.00, 204.47) : zone border move
READ: current_position=(0.00, 0.00, 204.47) : xy move
READ: echo:busy: processing
busy: processing
READ: <<< do_blocking_move_to
READ: current_position=(0.00, 0.00, 204.47) : clean_up_after_endstop_or_probe_move
READ: <<< gcode_G28
READ: X:0.00 Y:0.00 Z:204.47 E:0.00 Count X: 87095 Y:87095 Z:87095

thinkyhead, 2016-10-02:
> Is there a "define" statement I can comment out to disable the 40mm start point?

Not at this time. That's the current standard behavior — to move down to the highest point where movement in the full radius is possible. In your case that is Z=204.47.

I've added more detailed logging to `RCBugFix` so that you can test to make sure the Z bump and Z `endstop_adj` moves are taking place. If you do another logging session, it will reveal in better detail all the moves that are being sent to the planner.

The current method of backing off by the `endstop_adj` amount is an optional behavior that could be changed. It would work just as well to not physically back off, but instead apply the offsets to the `current_position` all at once after the axes have finished homing, before the move down to the print region.

To get rid of the final move down to the print area, you can comment out the line:
```cpp
do_blocking_move_to_z(delta_clip_start_height);
```

str8up2, 2016-10-02:
Latest 1.1.0 BugFix DL'ed this morning 10/02/2016
Commented out this line:
do_blocking_move_to_z(delta_clip_start_height);

Here is the detailed report - "Z" is not backing off from the home position. Should have moved to -1.80.
M111 S245
G28

READ: Machine Type: Delta
READ: Probe: NONE
READ:   current_position=(0.00, 0.00, 276.25) : setup_for_endstop_or_probe_move
READ: > endstops.enable(true)
READ:   current_position=(0.00, 0.00, 276.25) : >>> home_delta
READ:   current_position=(0.00, 0.00, 0.00) : sync_plan_position
READ: >>> homeaxis(X)
READ: Home 1 Fast:
READ: >>> do_homing_move(X, 315.00, 0.00)
READ:   current_position=(0.00, 286.25, 286.25) : sync_plan_position
READ: <<< do_homing_move(X)
READ: Move Away:
READ: >>> do_homing_move(X, -5.00, 0.00)
READ:   current_position=(0.00, 286.25, 286.25) : sync_plan_position
READ: <<< do_homing_move(X)
READ: Home 2 Slow:
READ: >>> do_homing_move(X, 10.00, 20.00)
READ:   current_position=(0.00, 286.25, 286.25) : sync_plan_position
READ: <<< do_homing_move(X)
READ: endstop_adj:
READ: >>> do_homing_move(X, -0.80, 0.00)
READ:   current_position=(0.00, 286.25, 286.25) : sync_plan_position
READ: <<< do_homing_move(X)
READ: <<< homeaxis(X)
READ: >>> homeaxis(Y)
READ: Home 1 Fast:
READ: >>> do_homing_move(Y, 315.00, 0.00)
READ:   current_position=(-0.80, 0.00, 286.25) : sync_plan_position
READ: <<< do_homing_move(Y)
READ: Move Away:
READ: >>> do_homing_move(Y, -5.00, 0.00)
READ:   current_position=(-0.80, 0.00, 286.25) : sync_plan_position
READ: <<< do_homing_move(Y)
READ: Home 2 Slow:
READ: >>> do_homing_move(Y, 10.00, 20.00)
READ:   current_position=(-0.80, 0.00, 286.25) : sync_plan_position
READ: <<< do_homing_move(Y)
READ: endstop_adj:
READ: >>> do_homing_move(Y, -0.80, 0.00)
READ:   current_position=(-0.80, 0.00, 286.25) : sync_plan_position
READ: <<< do_homing_move(Y)
READ: <<< homeaxis(Y)
READ: >>> homeaxis(Z)
READ: Home 1 Fast:
READ: >>> do_homing_move(Z, 414.38, 0.00)
READ:   current_position=(-0.80, -0.80, 0.00) : sync_plan_position
READ: <<< do_homing_move(Z)
READ: Move Away:
READ: >>> do_homing_move(Z, -5.00, 0.00)
READ:   current_position=(-0.80, -0.80, 0.00) : sync_plan_position
READ: <<< do_homing_move(Z)
READ: Home 2 Slow:
READ: >>> do_homing_move(Z, 10.00, 20.00)
READ:   current_position=(-0.80, -0.80, 0.00) : sync_plan_position
READ: <<< do_homing_move(Z)
READ: endstop_adj:
READ: >>> do_homing_move(Z, -1.80, 0.00)
READ:   current_position=(-0.80, -0.80, 0.00) : sync_plan_position
READ: <<< do_homing_move(Z)
READ: <<< homeaxis(Z)
READ: >>> set_axis_is_at_home(X)
READ: For X axis:
READ:  home_offset = 0.00
READ:  position_shift = 0.00
READ:  soft_endstop_min = -105.00
READ:  soft_endstop_max = 105.00
READ: > home_offset[X] = 0.00
READ:   current_position=(0.00, -0.80, -1.80) : 
READ: <<< set_axis_is_at_home(X)
READ: >>> set_axis_is_at_home(Y)
READ: For Y axis:
READ:  home_offset = 0.00
READ:  position_shift = 0.00
READ:  soft_endstop_min = -105.00
READ:  soft_endstop_max = 105.00
READ: > home_offset[Y] = 0.00
READ:   current_position=(0.00, 0.00, -1.80) : 
READ: <<< set_axis_is_at_home(Y)
READ: >>> set_axis_is_at_home(Z)
READ: For Z axis:
READ:  home_offset = 0.00
READ:  position_shift = 0.00
READ:  soft_endstop_min = 0.00
READ:  soft_endstop_max = 276.25
READ: > home_offset[Z] = 0.00
READ:   current_position=(0.00, 0.00, 276.25) : 
READ: <<< set_axis_is_at_home(Z)
READ:   current_position=(0.00, 0.00, 276.25) : sync_plan_position_kinematic
READ:   current_position=(0.00, 0.00, 276.25) : <<< home_delta
READ:   current_position=(0.00, 0.00, 276.25) : clean_up_after_endstop_or_probe_move
READ: <<< gcode_G28
READ: X:0.00 Y:0.00 Z:276.25 E:0.00 Count X: 101295 Y:101295 Z:101295

thinkyhead, 2016-10-02:
I might group this issue with others where the "Z" axis isn't moving away from target in some instances.

I don't know if it's related to a false trigger of end-stops, or just their bits, or what just yet. Reverting back to earlier commits is the only way to go, until one of them works. Then I'll know the bug is somewhere around that point – the most likely breaking change.

The thing is, since the other two carriages are backing off properly, it makes me wonder what is different in the "Z" case. So far I can't localize the cause.

Blue-Marlin, 2016-10-02:
Triggered probe?

thinkyhead, 2016-10-03:
I think something is getting triggered, or maybe it's picking up an old triggered value. But so far it looks like `endstops.hit_on_purpose` is called in the appropriate places, and the code to check the probe is supposed to only be active when the Z axis is moving towards the bed. Maybe it's worth explicitly enabling / disabling the probe and endstops in between raises and probes. Currently the probe is enabled throughout the duration of the probe procedure.

str8up2, 2016-10-03:
When you speak of "probe" are you referring to a bed probe? There is no bed probe on this Delta. All end stops are functioning. "Z" goes home (last) and stops - no back off at all. The sensor remains tripped.

Blue-Marlin, 2016-10-03:
Please show us the result of a M119 in mid air.

str8up2, 2016-10-03:
Blue-Marlin - Good call.

SENT: M119
READ: Reporting endstop status
READ: x_max: open
READ: y_max: open
READ: z_min: TRIGGERED
READ: z_max: open
READ: ok
This line was not commented out. It is now and "Z" is responding correctly.
#define USE_ZMIN_PLUG // a Z probe
Thanks so much for your and Thinkyhead's help.

closed 2016-10-05 by thinkyhead