MarlinFirmware/Marlin #4924: Move Z while run command

izaguerreiro, 2016-09-29:
Hello guys,

I need to make moves with the Z axis manually using the g code, so I can level the height when you are running a line or arc.

When I run "G1 X100" or "G2 X40 Y0 I20 J0" I can not do any other commands to change the height of the Z, I have to wait for these commands finalize. But I need them to run at the same time.

Is there any way to do this?

Thank you

thinkyhead, 2016-10-01:
GCode commands cannot be run concurrently, only sequentially, except when injected by the LCD controller or specially handled by the `EMERGENCY_PARSER`. The "BABYSTEP" feature on the LCD controller is the only way to make Z adjustments during the processing of a line or arc. This makes the adjustment without affecting the Z coordinates, simply sending extra steps directly to the Z stepper.

izaguerreiro, 2016-10-03:
In line execution, I managed to make the Z up and down. I created a M172 command to move up and down to M173.
But the arc does not work because of loop segments.

thinkyhead, 2016-10-06:
@izaguerreiro As mentioned, a GCode-based solution is a non-starter. You must wait for previous GCode to complete before the next one will be injected.

izaguerreiro, 2016-10-06:
Ok.
Thank you!

closed 2016-11-01 by izaguerreiro