MarlinFirmware/Marlin #4925: Support question. Configuring a CoreXY printer

mperdue, 2016-09-30:
I'm getting my corexy printer together and I just did the first power on tests. The Z Axis, using two drivers, seems to work correctly but maybe a bit fast - that's an easy fix. However, the X and Y axes move only in a positive direction. So, when I move the X positive it seems to work normally. When I them move it negative it doesn't move but the display shows the location you were trying to move it to. Moving positive again is relative tho the point where the carnage was after the positive move.

This is the first corexy setup I've attempted so I'm sure this is a configuration issue. Any help/guidance you guys can give would be greatly appreciated.

Configuration files attached.

[Corexy_configuration.zip](https://github.com/MarlinFirmware/Marlin/files/503190/Corexy_configuration.zip)

mperdue, 2016-09-30:
OK, I figured that problem out. The limit switches were wired incorrectly and it didn't function properly until a home was completed which it couldn't do. Fixing the switches corrected that problem.

Another question: On this printer the bed moves up and down for Z instead of moving the X/Y carriage. I'm assuming that I need to invert the direction of the motors so positive moves the bed down and also invert the home direction so it homes towards zero. Is that the correct setup? 

Any other caveats that I need to be aware of with a corexy machine?

brainscan, 2016-09-30:
Another question: On this printer the bed moves up and down for Z instead of moving the X/Y carriage. I'm assuming that I need to invert the direction of the motors so positive moves the bed down and also invert the home direction so it homes towards zero. Is that the correct setup?

Yes a positive z move should move the bed away from the nozzle (down in your case) but I think homing should default to homing to min/zero so should be set to -1 in configuration.h. Sorry I haven't had a play with coreXY yet so apologies if the defaults are set differently.
Keep us informed how you're getting on though.

mperdue, 2016-10-01:
Most of the defaults have been set correctly. The Z direction is the only thing I've encountered so far.  I hope to finish things up when I get home this evening. BTW, I really like the BLTouch reset in the menu. 👍

Aggebitter, 2016-10-01:
Invert the Z-axis, On the CoreYX it is often easier to invert the stepper wiring on one of the steppers to get the movement in X and Y correct corresponding to the X and Y endstops/homing switches and where you want the zeroed home position to be

mperdue, 2016-10-02:
Thanks for the input guys. I've almost got it ready to use. I'm waiting on and extruder heater and thermistor to arrive so I can finish things up. Homing and movement within the limits works well so I think I'm pretty much there.

mperdue, 2016-10-02:
Ouch! Bed leveling sure didn't work. Back to looking at the new hardware.

mperdue, 2016-10-03:
OK, I downloaded the latest rcbugfix and it's even worse. <sigh>

mperdue, 2016-10-03:
Back again. Has anyone with a CoreXY setup tried auto bed leveling on the current rcbugfix? If there is and you have it working, I'd love to see your configuration files to compare them to what I've done. 

FWIW, I turned on debug messages and did a G29. I had to reset the machine pretty quickly to avoid it slamming into the frame but I got the attached output. The only thing I see that does a move seems to be sending it to the center of the bed. That's certainly not where it goes.

echo:DEBUG:LEVELING
ok
  current_position=(98.00, 200.00, 11.00) : >>> gcode_G29
Machine Type: Core
Probe: BLTOUCH
Probe Offset X:52 Y:0 Z:-1.00 (Right-Center & Below Nozzle)
Auto Bed Leveling: BILINEAR
G29 Auto Bed Leveling
  current_position=(98.00, 200.00, 11.00) : setup_for_endstop_or_probe_move
  current_position=(98.00, 200.00, 11.00) : set_probe_deployed
deploy: 1
do_probe_raise(10.00)
>>> do_blocking_move_to(98.00, 200.00, 11.00)
start
echo:Marlin 1.1.0-RCBugFix

echo: Last Updated: 2016-07-26 12:00 | Author: (none, default config)
Compiled: Oct  2 2016
echo: Free Memory: 4531  PlannerBufferBytes: 1296
echo:Hardcoded Default Settings Loaded
echo:Steps per unit:
echo:  M92 X80.00 Y80.00 Z400.00 E500.00
echo:Maximum feedrates (mm/s):
echo:  M203 X300.00 Y300.00 Z5.00 E25.00
echo:Maximum Acceleration (mm/s2):
echo:  M201 X3000 Y3000 Z100 E10000
echo:Accelerations: P=printing, R=retract and T=travel
echo:  M204 P3000.00 R3000.00 T3000.00
echo:Advanced variables: S=Min feedrate (mm/s), T=Min travel feedrate (mm/s), B=minimum segment time (ms), X=maximum XY jerk (mm/s),  Z=maximum Z jerk (mm/s),  E=maximum E jerk (mm/s)
echo:  M205 S0.00 T0.00 B20000 X20.00 Y20.00 Z0.40 E5.00
echo:Home offset (mm)
echo:  M206 X0.00 Y0.00 Z0.00
echo:Material heatup parameters:
echo:  M145 S0 H180 B70 F0
echo:  M145 S1 H240 B110 F0
echo:PID settings:
echo:  M301 P22.20 I1.08 D114.00
echo:LCD Contrast:
echo:  M250 C40
echo:Filament settings: Disabled
echo:  M200 D3.00
echo:  M200 D0
echo:Z-Probe Offset (mm):
echo:  M851 Z-1.00

Printer reset to avoid crashing the XY max limits

thinkyhead, 2016-10-04:
I'm trying to track down recent changes that could be having this effect, and potentially trouble with doing probe moves generally. I hope to have a clue to the cause soon. Meanwhile, stick to code from a couple weeks ago.

mperdue, 2016-10-04:
Thanks. I've got an earlier version running now.

thinkyhead, 2016-10-06:
> Thanks. I've got an earlier version running now.

And it is working better?

You can also try these branches to see if the problem is only recently introduced:

- https://github.com/thinkyhead/Marlin/tree/test_z_issues_sept27
- https://github.com/thinkyhead/Marlin/tree/test_z_issues_sept28

mperdue, 2016-10-06:
I'm running the rcbugfix that I downloaded on Sept 16th and it seems to work fine. I'm currently finishing the build and calibration of the printer so it might be tomorrow before I can test the two files you've listed.

dteeter, 2016-10-07:
I have two coreXY printers (different designs) using Marlin
Here is my process for initial setup:
1. Decide which axis is which and which direction is positive
2. Center the extruder
3. Tell it to move in small distances on a single axis at a time - try both + and - directions as usually it will not move in one of those two directions
4. If it does not move in the proper direction, try reversing one stepper wire connector at a time to see if that fixes the problem (with two stepper connectors, each can be plugged in two ways, thats a total of 4 possible plugin combinations - it shouldn't take too long to find the right combo)
5. Once the axis are moving in the correct direction, now start working on stops
6. Start a 10cm move towards a stop and push the limit switch with your finger (it should stop moving as soon as you push the limit switch) - if pushing the limit switch does not stop the movement, then you need to look at the switch wiring, the location on the board where the switch is plugged in, and the software setting.

mperdue, 2016-10-08:
All of my motors are wired the same. The movement in the X and Y axis is exactly how you would expect it to be. The limit switches work properly. I can home the machine and print with it without problems. The only thing that messes up is the bed leveling process. I should be able to try the other firmware revisions and provide feedback later today.

mperdue, 2016-10-08:
@thinkyhead I've tested  the Sept27 version with the following results:

3 Point Leveling:
I couldn't get this to work at all. I kept getting "The given ABL_PROBE_PT_2_Y can't be reached by the z probe" regardless of what values I used.

Linear and Bilinear:

G28 functioned correctly.

G29 Probe moved to the first probe location and lowered for the first probe then it started drifting in the -X direction. Probe moved to second probe location, lowered and then drifted in the -Y direction. At the third location the BLTouch was deployed while the probe was too close to the bed causing it to error and rapidly move around.

I'll take a look at the other code shortly.

mperdue, 2016-10-08:
@thinkyhead I see why the 3 Point leveling didn't work the whole section on leveling is duplicated in the configuration.h file...

mperdue, 2016-10-08:
@thinkyhead No difference on Sept28 version.

mperdue, 2016-10-14:
@thinkyhead Have you figured out anything on this yet? Is there anything else I can do to help?

thinkyhead, 2016-10-20:
@mperdue Crazy week, so I'm just getting back to isolating these issues. I've been going over all the changes since ~ September 21 and will have a list of things that we can try pretty soon.

cronos45, 2017-03-30:
Any update on the? I'm having the, drifting and slamming in to max issues as well. Only when running G29.

closed 2017-05-26 by mperdue