MarlinFirmware/Marlin #4928: Mechanical resolution vs software stepping

Aggebitter, 2016-10-01:
I'm in the making of a closed loop stepper controller. When I did some testing earlier I started to wonder about how Marling is doing its stepping vs the mechanics. When a move is running the smoothest ride is to do as many microsteps you can, but on Hold/Stop/Start/Direction change a fullstep gains the most torque.

So by combining microstep moves with biggest microstep we can accept at start/stop we would get the most out of the stepper driver and motor. It evens gets better if we take into calculation that we actually hit the peak torque 200 times per revolution. I'd love to know what step that was. Well I kind know if I home the printer at a full step (one of the things i try to implement i my controller)  

How is stepping done in Marlin? Is each step the same as the vector distance without the "max" torque mentioned above ? 

If so, we can take some mechanical resolution limits in to the loop and tweak the steps to gain the max torque with in the mechanical limits or am I just thinking wrong ?

Example: 
Fullstep, 200 steps/rev, 20 tooth GT2 pulley gives 40mm/rev and a mechanical resolution of 0.2 mm 
Do the same with 1/4-steps and we get 0.05 mm ! well we are printing plastics ;-)

thinkyhead, 2016-10-02:
Marlin accelerates and decelerates as needed according to the configuration. The acceleration and jerk values determine the acceleration behavior. It uses the micro-stepping as hard-set at the stepper driver level. It cannot do a "whole step" without doing X individual micro-steps.

Aggebitter, 2016-10-02:
This is exactly  what I'm thinking of. Do the combining of knowledge of what puls was a full step we could do a calculation that keeps the best torque step inside the mechanical limit we set. 

At 80 steps/mm the resolution is 0.0125mm and at 160 steps/mm it is 0.00625. That is about the thickness of a human hair.  Way more than even a well built printer can do just by looking at the extrusion of filament. The stretching on belts and mechanical slack is greater than this.  

My studies of papers regarding this gives around 10% increase i torque at the full step but it has one more even more interesting feature. The full step is at a higher accuracy mechanical. We can even get that proof by not adding any current at all to the phases and that position the stepper will rest at is a full step 

I'm not trying to prof anything here. I'm trying to analyse if I should implement it in my closed loop stepper code

thinkyhead, 2016-10-02:
It's always going to be a trade-off between torque, speed, and accuracy. One simple way to mitigate changes in load is to add a little bit of friction intentionally to an axis - not so much that it take a lot of torque to accelerate the mass, but just enough so that slowing down and changing speed are nearly "free" because it brakes on its own.

Better torque is generally achieved using smaller pulleys also. If you were to use full stepping, theoretically, you would need to gear down and/or use small pulleys to get acceptable resolution. The smallest possible on a 5mm stepper shaft is probably 8mm diameter, yielding ~24mm per revolution, or 0.12mm resolution, which is not ultra-fine. But 0.06mm resolution (2x microstepping) is certainly acceptable.

stecor7001, 2017-03-11:
Hello all, sorry to revive an old thread, but I think I am hitting the same isse,mand others may too; 

I will start with a few basic points:
1. Microstepping vs holding torque
Full step :100%
16nth step: 9.8%
32nth step:4.91%

Since 1/16 and 1/32 step are common, i only listed these. Now, to simplify, let's focus on 1/16 only.

Thus, the holding torque  varies per microstep as follows:
 100% when the rotor position corresponds to one of the 200 (or 400) native steps
70% when the rotor is halfway between native steps
38% for each quarter step
Down to 9.8 - 10% for 16/th steps positions.

So, while microstepping, the holding torque varies between 100 and 10% of the nominal value, depending on where the physical rotor position is related to the actual motor step position.

There are undeniable benefits to microstepping suvh as low noise and lower vibrations, however one drawback is that, for certain rotor positions (where the torque is low) to motor does not do its job. This results in vibration patterns (if a force is to be overcome, the rotor woul move until nearing one of the higher torque positions, then oscillate until equillibrium is met). Friction does help with dampening, but is suboptimal.

I propose adding an additional parameter - resolution - which helps with the above as follows

E.g. 200 step/ rev motor
1/16 microstepping
16t gt2.5 belt
Results in 80(micro)steps per mm. Now as discussed above, some of these steps provide only 10% tq.

What I am proposing is adding a new parameter - resolution. Resolution could be: 5 steps/mm, in which case tq would be 100% and every actual move is a multiple of 16 microsteps. Basically, each move puts the rotor in one of its native positions, while retaining microstepping and its benefits.

Or, resolution could be 10 steps/mm when tq varies between 100% and 50% and the minimum move is 8 microsteps (halfstepping using microstepping). And so on so forth. Personally, I would not use more than quarter-stepping, as this yields unpredictable moves, and oscillation around the home position.

Adding this parameter would allow flexibility in choosing tq variance while retaining the benefits of microsteping (noise, vibration reduction). Please let me know what you think.

Thank you.
Stefan

thinkyhead, 2017-10-31:
> Adding this parameter would allow flexibility in choosing tq variance

I was out of the loop last October, so missed this. I like the concept, but I'm not sure whether the effects would be measurable (or, on the firmware side, desirable). It would be an interesting addition to try out.

I have to hit the planner and stepper code soon to attempt some other optimizations. At the same time I don't mind playing around with this concept.

My initial caveat comes from the fact that stepping is simply a matter of timed pulses. All you're proposing is to change the timing of the pulses, not the number, and inserting a "pause" at each larger (or full) step. I anticipate that the resonances would only differ, but would not be eliminated. As for holding torque, again, since the pulses are coming at a fast pace, this kind of change would only matter for direction changes.

Note also that in terms of acceleration, you're talking about doing some number (perhaps 16) intermediate steps very fast, outside the acceleration, always at fastest pace, which is not substantially different from doing an instant move. There may be a slight dampening, but this is only due to the extra time inserted between pulses — which is what the stepping interrupt already does.

Obviously it's hard to come to any conclusions without testing. Hopefully I'll be able to get into that stepper/planner work soon and play around with the idea.

closed 2019-02-19 by thinkyhead