MarlinFirmware/Marlin #4938: When compiling Marlin RC7 or the newest RCBugFix with MAX31855, Marlin goes into some kind of loop and doesn't respond any more

derletztename, 2016-10-02:
When I try to use Marlin on my RAMPS 1.4 EFB with the MAX31855 thermocouple board and the thermocouple after compiling and uploading, Marlin goes into some kind of loop and I can't connect with pronterface. Also I can't upload anything again unless I upload just in the tiny moment after plugging the Arduino into my PC. This happens with Marlin RC7 and RCBugFix on every Arduino I tested with and without RAMPS. And it only happens when I compile Marlin with the "-3" for MAX31855 support.
Maybe someone could figure out, what is wrong?
I'm printing metal, I need to measure high temperatures :(

```
//// Temperature sensor settings:
// -3 is thermocouple with MAX31855 (only for sensor 0)
```

thinkyhead, 2016-10-05:
The most likely thing getting stuck related to that thermocouple is the reading of the `MAX31855` itself. The loop which reads the bits will stick if it never gets all 32 bits. So perhaps it's an issue of timing. The read speed for the MAX31855 is 4x the rate of the MAX6675, so possibly the initial delay is too much. Try editing the method `Temperature::read_max6675` in `temperature.cpp` and removing one of these two "nop" lines…
```cpp
    // ensure 100ns delay - a bit extra is fine
    asm("nop");//50ns on 20Mhz, 62.5ns on 16Mhz
    asm("nop");//50ns on 20Mhz, 62.5ns on 16Mhz
```

derletztename, 2016-10-05:
I am not a coder and I have no idea what it really means that you just told me, but I will try this out and post my experiences. Thank you :)

thinkyhead, 2016-10-06:
I hear you. I've never been to Moscow. Let me know if the suggestion helps.

derletztename, 2016-10-06:
It did not change anything at all, as it seem :(

Actually both ICs look very similar:
https://cdn-shop.adafruit.com/datasheets/MAX31855.pdf
https://cdn-shop.adafruit.com/datasheets/MAX6675.pdf

The only easy to see difference is:
MAX6675
The data is output in a 12-bit resolution, SPI™-compatible, read-only format.
MAX31855
The  data  is  output  in  a  signed  14-bit,  SPI-compatible,  read-only  format. 

Also the MAX6675 can only work with type-K thermocouples and the MAX31855 works with several thermocouple types, which is why I am trying to use it...

Is this a problem for Marlin somehow, that it treats both ICs the same? 12-bit resolution and signed  14-bit?

derletztename, 2016-10-06:
Here are two Arduino Librarys for the MAX6675 and the MAX31855. Maybe the difference in the code helps fixing the Marlin Issues?
https://learn.adafruit.com/thermocouple/using-a-thermocouple

derletztename, 2016-10-06:
The MAX31855 code is twice the lines than the MAX6675. Also it looks different. Nevertheless it's still russian to me :(

Grogyan, 2016-10-06:
In temperature,cpp line 1293 
#define MAX6675_SPEED_BITS (_BV(SPR1)) // clock ÷ 64
You can try changing SPR1 to SPR0.  This increases the clock rate for the MAX31855, but you may end up with erranous values at times.

Can you please post your configuration.h file.

Also if you are in a particuarly noisy location, you may have spikes on the data/clock lines or even the thermocouple itself.  Also note that that you need to plug in the 5V from RAMPS to the 5V pin on the MAX31855 board, not the 3V3 rail, as that is an output .

derletztename, 2016-10-06:
It just won't upload this file, not even as a txt file. Sorry for posting this much

/**
 * Marlin 3D Printer Firmware
 * Copyright (C) 2016 MarlinFirmware [https://github.com/MarlinFirmware/Marlin]
 *
 * Based on Sprinter and grbl.
 * Copyright (C) 2011 Camiel Gubbels / Erik van der Zalm
 *
 * This program is free software: you can redistribute it and/or modify
 * it under the terms of the GNU General Public License as published by
 * the Free Software Foundation, either version 3 of the License, or
 * (at your option) any later version.
 *
 * This program is distributed in the hope that it will be useful,
 * but WITHOUT ANY WARRANTY; without even the implied warranty of
 * MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
 * GNU General Public License for more details.
 *
 * You should have received a copy of the GNU General Public License
 * along with this program.  If not, see <http://www.gnu.org/licenses/>.
 *
 */

/**
 * Configuration.h
 *
 * Basic settings such as:
 *
 * - Type of electronics
 * - Type of temperature sensor
 * - Printer geometry
 * - Endstop configuration
 * - LCD controller
 * - Extra features
 *
 * Advanced settings can be found in Configuration_adv.h
 *
 */
#ifndef CONFIGURATION_H
#define CONFIGURATION_H

#include "boards.h"
#include "macros.h"

//===========================================================================
//============================= Getting Started =============================
//===========================================================================

/**
 * Here are some standard links for getting your machine calibrated:
 *
 * http://reprap.org/wiki/Calibration
 * http://youtu.be/wAL9d7FgInk
 * http://calculator.josefprusa.cz
 * http://reprap.org/wiki/Triffid_Hunter%27s_Calibration_Guide
 * http://www.thingiverse.com/thing:5573
 * https://sites.google.com/site/repraplogphase/calibration-of-your-reprap
 * http://www.thingiverse.com/thing:298812
 */

//===========================================================================
//============================= DELTA Printer ===============================
//===========================================================================
// For a Delta printer replace the configuration files with the files in the
// example_configurations/delta directory.
//

//===========================================================================
//============================= SCARA Printer ===============================
[... 1036 lines cut ...]
 * Note may require analog pins to be defined for different motherboards
 **********************************************************************/
// Uncomment below to enable
//#define FILAMENT_WIDTH_SENSOR

#define DEFAULT_NOMINAL_FILAMENT_DIA 3.00  //Enter the diameter (in mm) of the filament generally used (3.0 mm or 1.75 mm) - this is then used in the slicer software.  Used for sensor reading validation

#if ENABLED(FILAMENT_WIDTH_SENSOR)
  #define FILAMENT_SENSOR_EXTRUDER_NUM 0   //The number of the extruder that has the filament sensor (0,1,2)
  #define MEASUREMENT_DELAY_CM        14   //measurement delay in cm.  This is the distance from filament sensor to middle of barrel

  #define MEASURED_UPPER_LIMIT         3.30  //upper limit factor used for sensor reading validation in mm
  #define MEASURED_LOWER_LIMIT         1.90  //lower limit factor for sensor reading validation in mm
  #define MAX_MEASUREMENT_DELAY       20     //delay buffer size in bytes (1 byte = 1cm)- limits maximum measurement delay allowable (must be larger than MEASUREMENT_DELAY_CM  and lower number saves RAM)

  #define DEFAULT_MEASURED_FILAMENT_DIA  DEFAULT_NOMINAL_FILAMENT_DIA  //set measured to nominal initially

  //When using an LCD, uncomment the line below to display the Filament sensor data on the last line instead of status.  Status will appear for 5 sec.
  //#define FILAMENT_LCD_DISPLAY
#endif

#include "Configuration_adv.h"
#include "thermistortables.h"

#endif //CONFIGURATION_H

derletztename, 2016-10-06:
I am not using the IC directly, I am using the Adafruit breakout board
https://www.adafruit.com/index.php?main_page=product_info&products_id=269&zenid=516505f41182f3a344fea2f8955968e0

Grogyan, 2016-10-06:
This line should be uncommented
//#define THERMAL_PROTECTION_HOTENDS // Enable thermal protection for all extruders

This line should be a lower value

derletztename, 2016-10-06:
I'm printing Steel, it's very experimental ;)
For the first experiments I only used a small amount of power and therefore it's heating up very slow. That's a problem for the thermal runaway...

derletztename, 2016-10-06:
The System can melt steel easily, but I wasn't able to measure higher temperatures above 1350°C, as I can't get that stupid MAX31855 to work :(

Grogyan, 2016-10-06:
@derletztename what thermocouple are ypou using?

derletztename, 2016-10-06:
started with a thermocouple type k, and now i am trying to use a thermocouple type s, which only works with the MAX31855.
I have the breakout board with a thermocouple type k for testing, but also changed the IC on another board and try to use it with a chinese thermocouple type S...

derletztename, 2016-10-06:
[image]

Grogyan, 2016-10-06:
https://datasheets.maximintegrated.com/en/ds/MAX31855.pdf

derletztename, 2016-10-06:
I asked the adafruit technicans. They say if you change the chip (solder, unsolder) it's suppose to work. Nevertheless it's not even working with the type k thermocouple without changing anything. I bought 2 breakout boards to tests this

derletztename, 2016-10-06:
I postet the same pdf^^

Grogyan, 2016-10-06:
The max6675 also has an alternate type of package

on the SO8 it should read MAX31855S  whereas the original chip marking reads MAX31855K
The K and S are the cold junction offsets within the device

Grogyan, 2016-10-06:
Can you also please post a photo of your thermocouple boards?

derletztename, 2016-10-06:
https://cdn-shop.adafruit.com/datasheets/MAX6675.pdf says it's only reading type K thermocouples. Or how do you mean alternate type of package?

derletztename, 2016-10-06:
I use the adafruit board.
https://www.adafruit.com/index.php?main_page=product_info&products_id=269&zenid=516505f41182f3a344fea2f8955968e0 and plugged it in as said on their site

Grogyan, 2016-10-06:
My mistake, I thought there was another variant of the 6675 for different thermocouples

derletztename, 2016-10-06:
Man, you brought up hopes :(
Is there hope to get the MAX31855 running?

Grogyan, 2016-10-06:
Indeed, I suggest first using an unmodified MAX31855 board and use a fresh copy of the Marlin firmware

derletztename, 2016-10-06:
I did. I bought 2 breakout boards and only modified one. I am trying to get it working with the unmodified board and I hardly changed anything in the Marlin RC and RCBugFix edition.
The error also happens without the RAMPS plugged in the Arduino. But the error do not happen without the -3 for the MAX31855 board set. So it must be an error in the Marlin code I guess

derletztename, 2016-10-06:
Just try this, download a new copy of Marlin and compile it with the -3 for the MAX31855 and see if you can connect to your Arduino afterwards. For me it don't work with pronterface nor with the Arduino software itself anymore. The only workaround is to upload another code just after plugging the Arduino in

Grogyan, 2016-10-06:
I can't try that out till late tonight when I get home from work

Blue-Marlin, 2016-10-06:
Might not bee very interesting for your problem.
To make negative temperature readings correct with a MAX31855, these chanes are needed.
Shifting right a negative unsigned will not shift in the needed 1s but always 0s (what makes them high positives).
```diff
@@ -1295,16 +1295,16 @@ void Temperature::disable_all_heaters() {
 #if ENABLED(HEATER_0_USES_MAX6675)
 
   #define MAX6675_HEAT_INTERVAL 250u
 
   #if ENABLED(MAX6675_IS_MAX31855)
-    uint32_t max6675_temp = 2000;
+    int32_t max6675_temp = 2000;
     #define MAX6675_ERROR_MASK 7
     #define MAX6675_DISCARD_BITS 18
     #define MAX6675_SPEED_BITS (_BV(SPR1)) // clock ÷ 64
   #else
-    uint16_t max6675_temp = 2000;
+    int16_t max6675_temp = 2000;
     #define MAX6675_ERROR_MASK 4
     #define MAX6675_DISCARD_BITS 3
     #define MAX6675_SPEED_BITS (_BV(SPR0)) // clock ÷ 16
   #endif
 
@@ -1347,11 +1347,11 @@ void Temperature::disable_all_heaters() {
     if (max6675_temp & MAX6675_ERROR_MASK)
       max6675_temp = 4000; // thermocouple open
     else
       max6675_temp >>= MAX6675_DISCARD_BITS;
 
-    return (int)max6675_temp;
+    return max6675_temp;
   }
 
 #endif //HEATER_0_USES_MAX6675
```

Blue-Marlin, 2016-10-07:
For analysing the problem
```diff
@@ -1343,11 +1343,31 @@ void Temperature::disable_all_heaters() {
     }
 
     WRITE(MAX6675_SS, 1); // disable TT_MAX6675
 
     if (max6675_temp & MAX6675_ERROR_MASK)
-      max6675_temp = 4000; // thermocouple open
+	  #if ENABLED(MAX6675_IS_MAX31855)
+	  if (max6675_temp & 1)	{
+		SERIAL_ERROR_START;
+        SERIAL_ERRORLNPGM("Temp measurement error! MAX31855 Open Circuit");
+	  }
+	  if (max6675_temp & 2)	{
+		SERIAL_ERROR_START;
+        SERIAL_ERRORLNPGM("Temp measurement error! MAX31855 Short to GND");
+	  }
+	  if (max6675_temp & 4)	{
+		SERIAL_ERROR_START;
+        SERIAL_ERRORLNPGM("Temp measurement error! MAX31855 Short to VCC");
+	  }
+	  #else
+	  if (max6675_temp & 4)	{
+		SERIAL_ERROR_START;
+        SERIAL_ERRORLNPGM("Temp measurement error! MAX6675");
+	  }
+	  #endif
+      //max6675_temp = 4000; // thermocouple open
+      max6675_temp = 25; // JUST TO AVOID THE KILL!!! Don't use in production.
     else
       max6675_temp >>= MAX6675_DISCARD_BITS;
 
     return (int)max6675_temp;
   }
```
might help.

derletztename, 2016-10-07:
Nähe Köln/Bonn. Über Support vor Ort würde ich mich seeehr freuen ;)

derletztename, 2016-10-09:
> #endif
+      //max6675_temp = 4000; // thermocouple open
+      max6675_temp = 25; // JUST TO AVOID THE KILL!!! Don't use in production.
     else
       max6675_temp >>= MAX6675_DISCARD_BITS;

--> leads to an error because the "#endif" ends the "if loop" and than there is an else without an if. Is the position, how much the line is indented (eingerückt) important for Arduino Code? Or should this be solved by the "#"? I hate to have no idea about coding :(

thinkyhead, 2016-10-09:
> leads to an error because the "#endif" ends the "if loop"

Try this:
```cpp
    if (max6675_temp & MAX6675_ERROR_MASK) {
      SERIAL_ERROR_START;
      SERIAL_ERRORPGM("Temp measurement error! ");
      #if MAX6675_ERROR_MASK == 7
        SERIAL_ERRORPGM("MAX31855 ");
        if (max6675_temp & 1)
          SERIAL_ERRORLNPGM("Open Circuit");
        else if (max6675_temp & 2)
          SERIAL_ERRORLNPGM("Short to GND");
        else if (max6675_temp & 4)
          SERIAL_ERRORLNPGM("Short to VCC");
      #else
        SERIAL_ERRORLNPGM("MAX6675");
      #endif
      //max6675_temp = 4000; // thermocouple open
      max6675_temp = 25; // JUST TO AVOID THE KILL!!! Don't use in production.
    }
    else
      max6675_temp >>= MAX6675_DISCARD_BITS;
```

derletztename, 2016-10-10:
At least this is compiling, going to test this on the machine soon :)

esenapaj, 2016-10-12:
I've completely same trouble as @derletztename.
Arduino MEGA2560 + RAMPS 1.4 EFB + [Adafruit MAX31855](https://www.adafruit.com/products/269).
I doubted that my hardware is broken at [2 month ago](https://github.com/MarlinFirmware/Marlin/issues/4533#issuecomment-239603375), but now I setup new MEGA2560 + new RAMPS, same trouble still alive.

It continuously reboots (freeze) when using MAX31855 without `SDSUPPORT`.
When freezing, red LED on RAMPS is flashing.
It can be reproduced without real thing of MAX31855. 

When I enable `SDSUPPORT`, MINTENP Err is occurs on every startup.

>Try this:

Unfortunately any message of those was shown.

esenapaj, 2016-10-12:
>```diff
    // ensure 100ns delay - a bit extra is fine
-    asm("nop");//50ns on 20Mhz, 62.5ns on 16Mhz
-    asm("nop");//50ns on 20Mhz, 62.5ns on 16Mhz
+    //asm("nop");//50ns on 20Mhz, 62.5ns on 16Mhz
+    //asm("nop");//50ns on 20Mhz, 62.5ns on 16Mhz
```
>```diff
-    uint32_t max6675_temp = 2000;
+    int32_t max6675_temp = 2000;
```
```diff
-    uint16_t max6675_temp = 2000;
+    int16_t max6675_temp = 2000;
```
```diff
-    return (int)max6675_temp;
+    return max6675_temp;
```

These weren't also solved problem unfortunately.

esenapaj, 2016-10-12:
And, I'm seeing interesting result.
With [this setting](https://github.com/esenapaj/Marlin/commit/5255213aea4a9e7b1c9f1a0d143ca71e51c07cbb), MINTENP Err is occurs on startup.
But with [this setting](https://github.com/esenapaj/Marlin/commit/334f39c7844e088ebc5b1287d609d232ec342ceb), MAXTEMP Err and MAX31855 Open Circuit Err are occurs on startup.
```
Error:Temp measurement error! MAX31855 Open Circuit
Error:MAXTEMP triggered, system stopped! Heater_ID: 0
```

It seems that something of other setting is affecting to result.

esenapaj, 2016-10-13:
Still searching, but [this setting and dirty hack](https://github.com/esenapaj/Marlin/commit/a121cbfa5dc0164955b617e99291cb4e409cdeae) works fine.

Grogyan, 2016-10-13:
I tried compiling RC7 a few days ago,  not RCBugFix, with Max31855
thermocouple and I just get a compiling error.
DIODIO_MISO had not defined in this scope. It was referring to pin 53 not
being defined,  possibly due to a conflict with SDCard,  it read like it
was a typo.
DIODIO_MISO should be DIO_MISO

I hadn't tried yet with RCBugFix,  I've not had a good week.

esenapaj, 2016-10-13:
@Grogyan 
>DIODIO_MISO

That's a [bug in RC7](https://github.com/MarlinFirmware/Marlin/pull/4554), and fixed in current RCBugFix.

esenapaj, 2016-10-13:
So, this hack (same as above) works fine with `SDSUPPORT` too, but there is a conditional.
When SDCard or microSD - SDCard adaptor is __already inserted in card slot at bootup__ of Marlin, it causes freeze and LED flashing.
But SDCard or microSD - SDCard adaptor is __inserted to card slot after bootup__ of Marlin, both SDCard and MAX31855 works fine.

```diff
  #if ENABLED(HEATER_0_USES_MAX6675)

-    #if DISABLED(SDSUPPORT)
+    //#if DISABLED(SDSUPPORT)
      OUT_WRITE(SCK_PIN, LOW);
      OUT_WRITE(MOSI_PIN, HIGH);
      SET_INPUT(MISO_PIN);
-     WRITE(MISO_PIN,1);
-    #else
+    WRITE(MISO_PIN, HIGH);
+    //#else
      OUT_WRITE(SS_PIN, HIGH);
-    #endif
+    //#endif

    OUT_WRITE(MAX6675_SS, HIGH);

  #endif //HEATER_0_USES_MAX6675
```

Then, I guess that initialize process of Marlin has some kind of problem...

Blue-Marlin, 2016-10-13:
Recently we shifted the initialisation of the thermal manager in front of the initialisation of the display/SDcard, because of uncontrolled heaters during the boot screens.

@esenapaj 
Your 'hack' makes completely sense to me. The SPI interface has to be initialized here.

esenapaj, 2016-10-13:
My final result at this time: https://github.com/esenapaj/Marlin/commit/a76393d6a0c3fc844359c424b562765cdafca4e8
In `Marlin_main.cpp`
```diff
  lcd_init();
  #if ENABLED(SHOW_BOOTSCREEN)
    #if ENABLED(DOGLCD)
      safe_delay(BOOTSCREEN_TIMEOUT);
    #elif ENABLED(ULTRA_LCD)
      bootscreen();
-      lcd_init();
+      #if DISABLED(SDSUPPORT)
+        lcd_init();
+      #endif
    #endif
  #endif
```
This second `lcd_init()` causes freeze when `SDSUPPORT` (and MAX31855) is enabled for some reason.
But when `SDSUPPORT` is disabled, this doesn't causes freeze, moreover if this second `lcd_init()` is removed, displaying of LCD is broken.

In `tmperature.cpp`
```diff
  #if ENABLED(HEATER_0_USES_MAX6675)

-    #if DISABLED(SDSUPPORT)
-      OUT_WRITE(SCK_PIN, LOW);
-      OUT_WRITE(MOSI_PIN, HIGH);
-      SET_INPUT(MISO_PIN);
-      WRITE(MISO_PIN,1);
-    #else
-      OUT_WRITE(SS_PIN, HIGH);
-    #endif
+    OUT_WRITE(SCK_PIN, LOW);
+    OUT_WRITE(MOSI_PIN, HIGH);
+    SET_INPUT(MISO_PIN);
+    WRITE(MISO_PIN, HIGH);
+    OUT_WRITE(SS_PIN, HIGH);

    OUT_WRITE(MAX6675_SS, HIGH);

  #endif //HEATER_0_USES_MAX6675
```
By test, these all the writing process are needed whether `SDSUPPORT` is enabled or not.
I wonder that why `OUT_WRITE(SS_PIN, HIGH)` is also needed when `SDSUPPORT` is disabled.

esenapaj, 2016-10-13:
@derletztename 
I prepared a test branch that it's based on newest RCBugFix.
Please try it if you'd like.
https://github.com/esenapaj/Marlin/commits/succeed

derletztename, 2016-10-13:
So great, i absolutely will test this. Thanks

derletztename, 2016-10-15:
Your code is working on my arduino, it doesn't freeze anymore, but I got an Mintemp error and Marlin stops working immediately. I checked the wiring again and everything is fine. Any Ideas about that?

esenapaj, 2016-10-15:
I guess that maybe your wiring of K thermocouple is inverted.
[K thermocouple has polarity](http://digital.ni.com/public.nsf/allkb/29FC44A9C662D80186256B02008387CF).
When start heating with inversely wired K thermocouple, 
measured temperature isn't increased but decreased, finally it go below the mintemp.

And this is information: my wiring
[image]

derletztename, 2016-10-15:
Actually it was at first, but after changing it, the error stayed the same. Also it's popping up just after booting Marlin via Pronterface. So I have no chance to see where the temperature would go to (up or down). It's just killing the Process with an error. Maybe I could upload a screenshot...

derletztename, 2016-10-15:
[image]

esenapaj, 2016-10-15:
This is mine.
[image]

I'd like to see your `Configuration.h`, `Configuration_adv.h`, and `pins_RAMPS.h`.
If you rename those files to `Configuration.txt`, `Configuration_adv.txt`, and `pins_RAMPS.txt`, and drag & drop to Comment window in this thread, those files are uploaded.
I can test your setting on my machine, because we has same hardware (RAMPS and MAX31855 by Adafruit).

esenapaj, 2016-10-15:
By the way, why your compiled date is Aug 17 2016...?

derletztename, 2016-10-15:
this is wierd. I just downloaded your https://github.com/esenapaj/Marlin/tree/succeed and just changed their in the config.h the setting for thermistor to -3

derletztename, 2016-10-15:
I use it on a PC that never has been connected to the web, it has a wrong date in the system, I never noticed it^^

derletztename, 2016-10-15:
Btw why is there a 1 for thermistor and not a -3 as this is your system? Maybe other things are wrong too then?

derletztename, 2016-10-15:
Where are you from? Can I get your hardware setup with software and just change everything at my place?^^ These little software issues really killing me :(

esenapaj, 2016-10-15:
Unfortunately, I live in 9000 kilometers to the east of your country...

By the way, I found two new cause of Mintemp error on bootup.
Those are "poor contacting" or "broken port".
When 5V (Vin) or SCK (CLK) is contacted poorly or disconnected, Mintemp error occurs.
On the other hand when GND or MISO (DO) or MAX6675_SS (CS) is contacted poorly or disconnected, Maxtemp error occurs.

derletztename, 2016-10-15:
Too bad you live that far away^^
So you mean contacted physically poor? I just use plug-in connectors for the experimental setup, you recommend soldering than?

derletztename, 2016-10-15:
To get that straight, the version you uploaded to github has thermistor settings for heater 0 set to 1instead of -3 as we both use it. Why is this? Maybe some other settings differ from the version you are actually using? Or maybe you forget to change to thermocouple usage and just measure with an thermistor you never disconnected?

esenapaj, 2016-10-15:
>So you mean contacted physically poor?

Yes it is, but you said that you use plug-in connectors, then maybe possibility of poor contacting is none...
...
Would you take a picture of your wiring of RAMPS and MAX31855, if it's possible?
I'm not sure that it's useful for solving the problem, but I can't conceive cause no longer.

>the version you uploaded to github has thermistor settings for heater 0 set to 1instead of -3 as we both use it. Why is this?

It's a "plane" (no modified from official Marlin) setting. It's premised that you alter configs by yourself.
[This is my setting for test](https://github.com/esenapaj/Marlin/commit/7604c6c1d904b47f21d83254888e6e59ec85a4ee), only change the `TEMP_SENSOR_0` from 1 to -3.

>you forget to change to thermocouple usage and just measure with an thermistor you never disconnected?

No, it isn't.
Please look at my picture of RAMPS on above, any thermistor isn't connected.

Grogyan, 2016-10-15:
I can't tell from the picture if you connected to the 5v on the Max31855
board.
Please also ensure that you have the CS pin connected to pin 66 on RAMPS

[36 more replies]

closed 2016-10-23 by derletztename