MarlinFirmware/Marlin #4944: Z home won't home echo:busy: processing

emielvs, 2016-10-03:
Hi all, when homing my Z-axis I keep getting the message RECEIVED: echo:busy: processing
busy: processing. Sometimes I also get the busy processing message when homing X and Y. In which case the bed first moves an inch down. In both cases I can get out of this state by reconnecting.

I swapped the Arduino Mega with an new one I had lying around and changed the stepper drivers with no success. Also I tried the RCBugFix and Arduino IDE 1.6.8 without any success. Tried it with repetier and simplify both had the same results.

[Communication log.txt](https://github.com/MarlinFirmware/Marlin/files/506760/Communication.log.txt)
[Config.txt](https://github.com/MarlinFirmware/Marlin/files/506762/Config.txt)

thinkyhead, 2016-10-05:
I'm not sure if this might be related to general Z-axis-related issues that have cropped up lately. Would you mind testing with these branches to see whether the problem exists in them?:

- https://github.com/thinkyhead/Marlin/tree/test_z_issues_sept28
- https://github.com/thinkyhead/Marlin/tree/test_z_issues_sept27

I'm hoping to find the origin of z-axis and position issues introduced by recent changes in the homing and leveling code. So far they have been hard to narrow down.

emielvs, 2016-10-05:
Thanks for the response also tried the RCBugFix with the same results

[Communication (5-10-2016).txt](https://github.com/MarlinFirmware/Marlin/files/512186/Communication.5-10-2016.txt)
[Configuration.h (5-10-2016).txt](https://github.com/MarlinFirmware/Marlin/files/512185/Configuration.h.5-10-2016.txt)

emielvs, 2016-10-05:
Also with https://github.com/thinkyhead/Marlin/tree/test_z_issues_sept28 I'm having the same results.
[Communication (5-10-2016) (test_z_issues_sept28).txt](https://github.com/MarlinFirmware/Marlin/files/512210/Communication.5-10-2016.test_z_issues_sept28.txt)

thinkyhead, 2016-10-06:
Hmm, if you're getting the same issue with 1.0.2 that points to either a configuration issue or a hardware issue, as opposed to a firmware issue.

Check your endstops first. When you issue `M119` with endstops un-pressed versus pressed, do they all report correctly?

I will look over your configuration and see if anything stands out.

thinkyhead, 2016-10-06:
Your configuration is pretty basic. CoreXY with homing to min endstops.
I would definitely start by checking the endstops.

emielvs, 2016-10-09:
Oke for some strange reason it now seams to work. I assume it was some wiring not connected correctly. Thank you for the support!!

closed 2016-10-09 by emielvs