MarlinFirmware/Marlin #4947: Ic2 to MAX31850K

Airheadbit, 2016-10-04:
I was looking to upgrade my  thermistors to bead thermocouples using MAX31850K devices talking to a DS2482-100 and that device speaks I2C. I looked into using SPI but realized that was already in use and after investigating 1 wire by bit banging I decided that was a dead end in this environment.
I have sources on everything but the 1.27 SMT 4 position address switches, waiting on a quote from Ali. 
This would free up the three thermistors, A13, A14, A15 which could become a 6th axis or third extruder with an off-board driver. 
Also using I2C there wouldn't be a limit on the number of temperatures measured plus the code would be straight forward (I hope). Is this something that can be done? If so I'll layout the PCB's needed and get them built. Oh, forgot to mention, the MAX devices are parasitic, they will live 100mm from the extruder/bed and cheap wire is routed between the 1 wire devices (shielded mini-coax) this reduces the cost a lot and increased the resolution.

thinkyhead, 2016-10-05:
I've had pretty good luck using i2c, and have tried to make the i2c support as straightforward as possible in the latest code. See the thread starting from https://github.com/MarlinFirmware/Marlin/issues/4776#issuecomment-246262879

Using i2c thermocouples for temperature readings brings an interesting safety challenge, as of course the feedback loop needs to be responsive. And we don't want to leave heaters on if i2c were to get cut off for any reason.

Airheadbit, 2016-10-06:
What do you do if a thermistor opens? I thought about swapping the DS2482-100 for an ARM processor with Multimaster I2C letting the ARM device keep track of the MAX31850's and freeing up the ATmega2560 of any temperature formatting, it would cost about half the price of the DS2482-100. You can update the ARM's firmware over I2C. Put the ARM on a small circuit card with a 4 pin female Dupont connector, the 1 wire buss is daisy chained to all off the back if the ARM module. For this project IPS/u.fl connectors everywhere.

thinkyhead, 2016-10-06:
> What do you do if a thermistor opens?

A thermistor will suddenly go to the highest or lowest value, as it's a direct electronic connection. We need to make sure, when using an i2c scheme, that a disconnect doesn't just leave the temperature set where it was, but also shows up as something out of range. We can of course set the temperature to "0" to give the same result if i2c becomes unresponsive.

> freeing up the ATmega2560 of any temperature formatting

It sounds doable. The only thing I've used i2c for was to make an external Z axis controller. That is arguable a more complex task, and yet it wasn't very difficult.

Airheadbit, 2016-10-06:
"I've used i2c for was to make an external Z axis controller"
What? You added code for the VL6180X and I missed it? I'm waiting on one, my LJ18A3-8-Z/BX works but it's just ugly and big.

thinkyhead, 2016-10-09:
Not a VL6180X. Just a RAMPS running a very pared-down "slave" version of Marlin. Only the basic code needed to run a Z axis and take i2c commands.

Airheadbit, 2016-10-13:
Got around to laying out a PCB, the top layer shows the I2C to 1 wire controller and connector to the 4 pin I2C header on RAMPS 1.4. The second image is the bottom with the 1 wire connector. The I2C address is fixed at 0x18, is this a problem? 
  
[image]

[image]

Airheadbit, 2016-10-17:
Junked the IPX connector and went to a Terminal block screw type 3.81mm pitch, found some at a good price and the interconnect on the IPX connectors turned out to be a bust, they think dual IPX female cables should cost more than gold.

thinkyhead, 2016-10-22:
> The I2C address is fixed at 0x18, is this a problem?

It shouldn't be. This is very common. When the address is fixed in this way, sometimes you need to shift it to the left, so instead of address 24 (0x18), you might need to specify address 48 (0x30).

> they think dual IPX female cables should cost more than gold.

Well, they usually _are_ gold.

closed 2017-10-31 by thinkyhead