MarlinFirmware/Marlin #4948: Bilinear Matrix Not being applied effectively. RC7 Bugfix

WOOKIE111, 2016-10-04:
Is anyone Available to assist? Marlin RC7 Bugfix - Bilinear Leveling. it doesnt seem to correctly interpret the data from probing. and at points nearly crashes to the bed while printing

The Probing is done successfully with all the points i have tried between a total of 16 points all the way to 81 total points. all with the same effect.

It is almost like it is not correctly Implementing the data it collected to move the z axis appropriately. Or not using the data at all ( i watch the z axis and it seems to move slightly. so it not using it might be out)

I am using a BLTouch as a Probe and for the board i am using an AzteegX3( great board very nice)

here is the "config.h" file
 http://pastebin.com/i5Q7hgxe

thinkyhead, 2016-10-04:
`RC7` or `RCBugFix`?

I'm trying to track down current issues with probing and leveling. I hope to have some clues soon.

WOOKIE111, 2016-10-04:
Sorry i had thought RC Bugfix was called RC7 bugfix.

To Clarify it IS The Bugfix branch.

oxivanisher, 2016-10-07:
I have the same issue. With a pretty current RC Bugfix, the Bilinear Matrix is not applied.

oxivanisher, 2016-10-07:
I thing #4961 could be the same problem.

Roxy-3D, 2016-10-08:
@WOOKIE111 @oxivanisher

What kind of machines are you on?   Are you running Delta's ?

WOOKIE111, 2016-10-08:
@roxy-3d 

I have a large Cartesian called the herculien 
Gantry style like an UM

Large build area.

Roxy-3D, 2016-10-08:
> Large build area.

Can you easily get good adhesion across the entire (and full) bed?

WOOKIE111, 2016-10-08:
Yes. if i go and use that branch i will definately need assistance @Roxy-3D

oxivanisher, 2016-10-08:
@Roxy-3D
Yes, I am on a delta. (Kossel XL). And I am suffering the same problem as Wookie111 does with a slightly warped bed.

Roxy-3D, 2016-10-08:
> so this branch is good for a warped bed? the bed is in areas 0.5mm out... and in others inches away it can be almost 0.5 the other way.

Yes, I've got a piece of glass that has 1.25 mm up and down differences.   I can get a perfect first layer every time across the entire bed.    I'm getting setup to bring the code up on Delta's next, but I haven't started that work yet.   Right now it is working on Cartesian and CoreXY machines.

thinkyhead, 2016-10-08:
I will put time on this later today, and I hope it is related to #4961 and some others. I expect it will be something transient and (I hope) obvious, but I need to go through and do a comprehensive bug scrub. I worked on bilinear bed leveling mid-week and it did seem to be applying the height adjustment, though I didn't get a chance to use the dial indicator to solidly confirm it.

I'll ask you guys for your log output later and I can pore over those for clues as well.

gcormier, 2016-10-08:
@roxy-3D does the ubl branch pull all the goodness from rcbugfix pretty often?

Roxy-3D, 2016-10-08:
>  does the ubl branch pull all the goodness from rcbugfix pretty often?

It was current on July 4th, 2016.   I'll probably do the work to get both branches back in sync when RC-7 gets retired and RC-8 starts.      Right now, the most important thing the UBL branch is missing is the BLTOUCH stuff.

gcormier, 2016-10-08:
Gotcha. I'm using bltouch  so I'll have to hold off. Super interested though because my pei sheet is convex in the middle slightly.

WOOKIE111, 2016-10-20:
@Roxy-3D is there a way to contact you outside of here for assistance with my migration to the UBL from bugfix?

Roxy-3D, 2016-10-20:
Yes.  Send me a Private Message at www.3dprintboard.com   User: Roxy

Roxy-3D, 2016-10-22:
@WOOKIE111   Do you have everything you need to keep making progress?   

> Gotcha. I'm using bltouch so I'll have to hold off. Super interested though because my pei sheet is convex in the middle slightly.

@gcormier I've got a machine with a BLTouch probe now.  I'm going to start bringing up the UBL Bed Leveling System on it ASAP.  I'm not sure how long it will take.  One problem is it has a Graphics Display and I've never had to configure my build system for that.   I think I might just dedicate a spare laptop to that machine so I don't have the u8graphics library messing things up for my other machines.

But the point is...   There will be a BLTouch version of the Configuration.h very soon that you can use as a starting point.   And even though I don't see much value to the graphics displays, it is good to do development on a machine with a graphics display because those things eat a lot of CPU cycles.  If the code works there, it will work on the normal 20x4 LCD panels too.

WOOKIE111, 2016-10-22:
@Roxy-3D I Have a Bltouch as well. i have a copy of marlin that everything works as expected except for the Bilinear ( namesake of the thread) so i would assume copy settings over and badda bing badda boom right?

i hwoever looked over the UBL and i dont see the BLtouch Functionality. or atleast not the easy way. and im unsure how to do it otherwise. or which bed leveling options TO use. as there seemed to be about 4 in the UBL. in the beginning of that section anyways.

Roxy-3D, 2016-10-22:
> i hwoever looked over the UBL and i dont see the BLtouch Functionality. or atleast not the easy way. and im unsure how to do it otherwise.
Yeah, that was my point.  There will soon be a Configuration.h file for UBL that gets a BLTouch all setup.   (The root of this problem is UBL forked from RCBugFix just before the simple and easy BLTouch support was added.)   

>  or which bed leveling options TO use. as there seemed to be about 4 in the UBL. in the beginning of that section anyways.

Everything is built on a High Resolution Mesh.   Probably that does everything you need.

gcormier, 2016-10-22:
@Roxy-3D 

So, I've abandoned the BLTouch as I was unable to get acceptable performance out of it. Variance of 0.1mm is unacceptable to me when my inductive probe is doing 0.01mm, an order of magnitude better. My bed is a MIC6 piece of Aluminum, so very very flat. The PEI on top, not quite so flat unfortunately. UBL will be interesting to me if you can manually edit it, as my inductive sensor will still pick up a perfectly flat bed underneath the surface of the PEI.

Please, let me know how your results with BLTouch are though.

thinkyhead, 2016-10-22:
I got a BLTouch but haven't had a chance to install it yet. @gcormier Is your BLTouch variance easily seen with `M48`? And neither slower probing nor double-touch probing makes much difference?

gcormier, 2016-10-22:
Correct - the newly min/max M48 additions highlight the issues quite nicely.

I believe I had my Z SLOW speed down to (ZHOMING / 5)... no changes. 

(Also note, I'm not blaming Marlin in this case - this repeatability was consistent across RC6, RC7 and RCBugFix. I believe the BLTouch unit I have is faulty.)

WOOKIE111, 2016-10-22:
@gcormier Yeah mine as well. that or they need to stop saying they have a repeatability of 0.0001 cause thats a bold lie. best i have got is 0.02 with an m48 but that was once only. other times has been as high as 0.3 ( yes im Jim Stone on G+ ) 
but in our cases. we cant use inductive probes we need a touching probes because we need to pickup the variance of a material on top ( mine being glass yours being pei.)

I think im just about done modifying UBL Marlin. i just need to add the BLTouch in the hardway ( servo and angles n stuff.)

i will let you know if accuracy some how magically improves. but i doubt it. because it isnt firmware problem, its a hardware problem.

but the interpretation ( thread name) IS firmware problem. as the BLTouch only can say ok that point was Xunits

Roxy-3D, 2016-10-22:
With no tuning or alterations...  (Just using the factory supplied firmware image)    M48 V4 in the center of the bed gives a Standard Deviation of .047mm for me using the supplied BLTouch probe.  

That seems a bit high.  Maybe I can get that lower with some tuning.

thinkyhead, 2016-10-22:
It doesn't surprise me that BLTouch isn't the most accurate probe. That electro-magnet probably has some variance. There are plenty of [Google results](https://www.google.com/search?q=bltouch+repeatability) to back this up.

Roxy-3D, 2016-10-22:
> Yeah mine as well. that or they need to stop saying they have a repeatability of 0.0001 cause thats a bold lie. best i have got is 0.02 with an m48 but that was once only. other times has been as high as 0.3 

I've been playing with this for the last hour.   Their claims of .001mm is BS.    I changed the code so the Z-Axis stays frozen.   And then I very slowly, very gently triggered the probe.   I can't get down to .01mm Standard Deviation.

WOOKIE111, 2016-10-25:
@Roxy-3D when trying to compile Dev UBL i get this error
"braces around scalar initializer for type 'const int'"
 and it highlights this line
#define Z_SERVO_ANGLES  {{0,0}, {0,0}, {10,90}} // Z Servo Deploy and Stow angles

i had changed this line to look like that. as in the bltoucch instructions it tells me to make it look like that. it had looked like this
#define Z_SERVO_ANGLES  {0,0} // Z Servo Deploy and Stow angles

Roxy-3D, 2016-10-25:
> ok. i got it setup and ran a g29 p1 m

Really...  You should start the  process with a G29 P1 O    (The M is still honored, but it confuses some host programs.     And if you want to be lazy, do a G29 P3 R C .25 O       followed by a G29 A S1  and you will be good to go.    You would probably want to do a Mesh Validation Print to see where you are at with a: G29 P C O 2.0       And then if you need to, you go edit any area of the mesh that is less than perfect by moving the nozzle there and doing a G29 P4 R O .   When happy with things, do another G29 S1

WOOKIE111, 2016-10-25:
So.... for a starting gcode when printing do i need to do the whole g29 thing again. or is it ok to just do a g28 and it will follow the matrix

Roxy-3D, 2016-10-25:
> So.... for a starting gcode when printing do i need to do the whole g29 thing again. or is it ok to just do a g28 and it will follow the matrix.

It depends on how you set things up.   I've allowed backward compatibility where you have to run a G28 and then do a G29 to get the bed leveling active.    But if you save a Mesh to one of the EEPROM Mesh Slots with the UBL System Active....   It will be active when the printer powers up (or resets).   

I want the bed leveling system alive and functioning when my printer powers up.  So of course, I make sure that happens.   But if you want to have control over everything you can save the Mesh to some slot with the system de-active.     And in that case you can do a G28 and until you do a G29 A the system will not be correcting things.   (The last Mesh Slot you saved something will be loaded automatically.   If you want to use a different mesh, you will have to do a G29 L5 A to load Slot 5's mesh and activate it.)

WOOKIE111, 2016-10-25:
@Roxy-3D as to how ive set things up? erm i dont quite understand.

all i did was modify the config.h file to fit the motor steps etc. and as for everything else you have said. well thats a small migraine in itself XD i understood very little of that

Roxy-3D, 2016-10-25:
I'm sorry...     You can try a few things just to get the feel of things and see how things work.  Try this:
- Bring up the printer and do a G28
- G29 P0 O     To zero the entire mesh and print it out.
- G29 A           To activate the system
- G29 S1          Save the mesh (that right now is entirely 0.000 to mesh slot 1)
- G26 P C O 3.0    To print a Mesh Validation Pattern.   (Remember...  there really is no correction yet)  (You may need to also specify your filament diameter, nozzle size and bed and nozzle temps)
- G29 P4 R O       Move the nozzle to any bad area of the pattern, and edit the mesh point up or down.  Press and Hold the encoder wheel when you want to move to a new area.  Move to a new 'bad area' and repeat the command again.
- G29 S1         Save the updated mesh to slot 1.    Now, go back up to the G26 P C O 3.0 and give it another try.

Once you get a decent Mesh Validation Print...     You can just turn on your printer, and it will use the correction matrix for any print you do.   You will see the Z-Axis moving back and forth as the nozzle moves around on the bed (assuming you have the nozzle low enough you are below the 'Fade Height')

WOOKIE111, 2016-10-25:
In order i did this
G29 P1 O ( it started probing) 

After it finished i did these
G29 P3 R C .25 O 
G29 A S1 
G29 P C O 2.0 

However they didnt seem to echo back...did they work? 
>>> G29 P3 R C .25 O
SENDING:G29 P3 R C .25 O
>>> g29 a s1
SENDING:G29 A S1
>>> G29 P C O 2.0 
SENDING:G29 P C O 2.0

as an extra here is the probing results
http://pastebin.com/HUnLGEW3

Roxy-3D, 2016-10-25:
Change the code to be this.   (There is just an extra 'if' statement and a closing brace)
```cpp
bool turn_on_heaters() {
#if ENABLED(ULTRA_LCD)
  if ( bed_temp > 25 ) {
  lcd_setstatus( "G26 Heating Bed.  ", true);
  lcd_quick_feedback();
#endif
  UBL_has_control_of_LCD_Panel++;
  thermalManager.setTargetBed( bed_temp );
  while( abs(thermalManager.degBed()-bed_temp) > 3 ) {
  	if ( G29_lcd_clicked() ) {
		strcpy( lcd_status_message, "Leaving G26    "); // We can't do lcd_setstatus() without having it continue;
  		while ( G29_lcd_clicked() )			// Debounce the switch
			idle();
            	lcd_setstatus( "Leaving G26     ", true);	// Now we do it right.
		return 1;
	}
	idle();
  }
  }
```
And then specify a bed temperature of 22 (room temperature).   That should get you further into the process.   Sorry about that...   This was developed on a machine with a heated bed.  Of course I should have realized some people don't use a heated bed.   But this is the first time it has been pointed out.

WOOKIE111, 2016-10-25:
@Roxy-3D wow...where is that? ive only seen Config.h and this doesnt look like it

Roxy-3D, 2016-10-25:
About 975 lines into G26_Mesh_Validation_Tool.cpp

WOOKIE111, 2016-10-25:
that works at 22? even without a thermosistor in the bed spot?

Roxy-3D, 2016-10-25:
Well...    It isn't an elegant 'fix'.   But it isn't going to tell the bed to heat up if you add that code and specify a temperature of 22 degrees.     It's going to be ignored.    At the very end of the command, when it turns off the heaters, there may be an issue.    If so, you can add a 'K' to keep the heaters on.   So it would be G26 C K P O 3.0  and then it won't turn off the heaters.   Or alternatively, you can let it get an error at the very end, and just restart the printer.   The important thing is you have a Mesh Validation Pattern partially printed so you can see what has to happen to make the mesh definition better.

WOOKIE111, 2016-10-25:
im not sure if this has any merit. but i have found out why i cant get any commands in after runnign the probe.

my LCD says M105 for some reason. which perhaps may be blocking further actions.

i can affirm this because i can run the commands after rebooting everything 

but if i run a probe. the screen will say something along the lines of G29 UBL Active

but after it completes the LCD stays at m105

Roxy-3D, 2016-10-26:
Hmmm...   Are you saying your LCD Panel is injecting an M105 all by itself?   I don't understand what you are seeing.    

But let's see if we can get you up and running without using the probe.  That will be valuable for other people.

WOOKIE111, 2016-10-26:
Uh, isnt the point of UBL to use a probe for an uneven bed? or have i been mistaken? im really confused now.

Roxy-3D, 2016-10-26:
> Uh, isnt the point of UBL to use a probe for an uneven bed? or have i been mistaken? im really confused now.

No!   The purpose is to provide perfect correction for non-flat and non-level beds.   UBL will use the probe to acquire much of the information if the probe is available.   But the steps I was giving you were to get you setup without even using the probe.     Even if the printer has a Z-Probe, there are parts of the bed that can not be reached by the probe because the probe is usually offset from the nozzle.   For these points, even if the bulk of the mesh is generated with the probe, other methods have to be deployed to get those values defined.    

What we are doing is getting the whole mesh defined for your bed without using the probe.

Does that make sense?

WOOKIE111, 2016-10-26:
alrighty. that i understand.

now i have got the probe points from 

G29 P1 O
then i saved it with 
G29 S1

i then attempted to print something with the starting Gcode of
G28
G29 A
G29 L1

it seemed to apply the matrix. but in a few areas it still got much too close to the bed. and partially plugging the nozzle in areas because it was that close.

so i guess this is where the manual bit would apply?

WOOKIE111, 2016-10-26:
also when implementing this code

bool turn_on_heaters() {
#if ENABLED(ULTRA_LCD)
  if ( bed_temp > 25 ) {
  lcd_setstatus( "G26 Heating Bed.  ", true);
  lcd_quick_feedback();
#endif
  UBL_has_control_of_LCD_Panel++;
  thermalManager.setTargetBed( bed_temp );
  while( abs(thermalManager.degBed()-bed_temp) > 3 ) {
    if ( G29_lcd_clicked() ) {
        strcpy( lcd_status_message, "Leaving G26    "); // We can't do lcd_setstatus() without having it continue;
    while( abs(thermalManager.degBed()-bed_temp) > 3 ) {
    if ( G29_lcd_clicked() ) {
        strcpy( lcd_status_message, "Leaving G26    "); // We can't do lcd_setstatus() without having it continue;
        while ( G29_lcd_clicked() )         // Debounce the switch
            idle();
                lcd_setstatus( "Leaving G26     ", true);   // Now we do it right.
        return 1;
    }
    idle();
    }

#if ENABLED(ULTRA_LCD)
  }
  lcd_setstatus( "G26 Heating Nozzle. ", true);
  lcd_quick_feedback();
#endif

into the area... i get this error in arduino

a function-definition is not allowed here before '{' token

and it highlights this
 void prime_nozzle() {

Roxy-3D, 2016-10-26:
> it seemed to apply the matrix. but in a few areas it still got much too close to the bed. and partially plugging the nozzle in areas because it was that close.

You should do G26's until the Mesh Validation Pattern is perfect every where.   We are starting with a totally unprobed bed.   The Mesh was all 0.0000's at the start.   So it is going to take a few iterations to get the Mesh 'Right'.    (Also...  You have no experience how much to raise or lower the given Mesh Points based on what the pattern looks like.   That alone is going to add an iteration to the process.)

I'm watching a movie with friends...  So my responses might not be immediate.   But I'll be up for a couple more hours if you keep plugging along....

I uploaded a better G26.  It might fix that issue if you want to give it a try.   Be sure you don't over write your currently some what working Configuration.h file.

WOOKIE111, 2016-10-26:
@Roxy-3D when doing the G29 P4 R O should i expect the nozzle gap to change while using the scroll wheel? or are we estimating. when i used that command it didnt move to z0 is that normal?

Roxy-3D, 2016-10-26:
> when doing the G29 P4 R O should i expect the nozzle gap to change while using the scroll wheel? or are we estimating. 

@WOOKIE   No...  You should not see much movement of the nozzle up or down while editing a Mesh Point.   The nozzle is moved to the Mesh Point but raised enough it can safely move around.   The nozzle is positioned over the Mesh Point so you can easily tell what you are editing.

> when i used that command it didn't move to z0 is that normal? also when a g1 z0 is issued then the G29 P4 R O the z moves to z13

G1 Z0.000 should move the nozzle to just touch the glass.   But Z-Axis only moves are a special case.   I'm not sure without digging through some stuff.   You might want to do a G1 X100 followed by a G1 Z0.000 to see if you get the desired behavior.

WOOKIE111, 2016-10-26:
well no matter how high or low i put the z axis before hand with a g1 z# ...... as soon as i do the G29 P4 R O it moves to the first mesh point. and z13 and rotating the knob either direction even into full numbers doesnt cause the bed to move.

WOOKIE111, 2016-10-26:
so at this point since the gap doesnt move from z13. everything i put in on the lcd is a guess.

is that the point? guesstimation? or am i supposed to actually be using a feeler gauge or paper

Roxy-3D, 2016-10-27:
That is correct behaviour.   The Z=13 is your Z_RAISE_BETWEEN_PROBINGS number.  That is a known safe height to move the nozzle around.    When you turn the Encoder Wheel at a given Mesh Point while doing a G29 P4 R O command, you are raising or lowering that Mesh Point.    If you check the Map before you start editing a point, and check the map after, you should see the change reflected at that location on the map.

As for how much to turn the Encoder Wheel?   Well...  You get a feel for it pretty quickly.   You want the lines drawn by the Mesh Validation Pattern print to be squished out a little bit and flat.  But you don't want them squeezed so thin you can barely see any plastic being put down.

If your layer height is .25mm, if you have any adhesion at all, you probably only want to drop the mesh point by .15mm.      If you do the Mesh Validation Print, you should have some locations that you can see the filament is a little bit rounded and not squished into the glass.   Go edit a couple of those points and lower them by .10mm or .15mm.    You will get a feel for how much to raise or lower things very quickly.

I don't even look at the LCD Panel when doing a new Mesh.   I just look to see where the nozzle is and turn the dial close to the right amount.    You will get a good feel for that very quickly.

WOOKIE111, 2016-10-27:
so. is going negative making the hotend closer?

Roxy-3D, 2016-10-27:
> so. is going negative making the hotend closer?

You don't have enough text here for me to be sure what you are asking.  But when you are editing a Mesh Point with the G29 P4 sub-command...   Going negative means you want the nozzle to move closer to bed in this area of the mesh.   If the bed was perfectly flat, you would want 0.0000's everywhere.  But if you had a perfectly flat bed with a small dip in one area, you would indicate that by putting the Mesh Points in that area at a negative value.

WOOKIE111, 2016-10-27:
Precisely what I meant.

 thank you again. I stopped trying a few hours ago (about 20 mins fter my last question.) but I shall try again tomorrow. and pester you more! :P

Roxy-3D, 2016-10-27:
When you can get a G26 P C O 3.0  to partially print...   Post a picture.    It will help the discussion.   When you can see the Validation Pattern, it becomes very clear what needs to be adjusted.   (Post a G29 O map also!)

WOOKIE111, 2016-10-27:
here is the baseline. a map of all 0's
http://imgur.com/a/QvLvq

next up is the first modifiation

WOOKIE111, 2016-10-27:
http://imgur.com/a/QvLvq

Bed Topography Report:
(0,9)                                                                                                                                                             (9,9)
(0,338)                                                                                                                                                         (358,338)
  0.01000     0.06000     0.05000     0.04000     0.05000     0.06000     0.10000     0.10000     0.15000     0.16000

 -0.10000     0.05000     0.01500     0.02000     0.05000     0.02000     0.07000     0.06000     0.04500     0.15000

 -0.10000     0.05000     0.05000     0.05000     0.07500     0.02000     0.07000     0.04000     0.05000     0.10000

 -0.10000    -0.10000    -0.10000     0.05000     0.05000     0.02000     0.05000     0.03000     0.02000     0.07000

 -0.15000    -0.10000    -0.10000    -0.10000    -0.15000    -0.15000     0.05000    -0.10000    -0.10000     0.01000

 -0.10000    -0.10000    -0.10000    -0.10000    -0.12500    -0.10000    -0.10000    -0.10000    -0.10000    -0.10000

 -0.15000    -0.10000    -0.10000    -0.10000    -0.10000    -0.10000    -0.10000    -0.10000    -0.10000    -0.10000

 -0.15000    -0.10000    -0.10000    -0.15000    -0.10000    -0.15000    -0.10000    -0.15000    -0.15000    -0.15000

 -0.15000    -0.15000    -0.15000    -0.10000    -0.10000    -0.15000    -0.10000    -0.15000    -0.15000    -0.15000

[-0.15000]  -0.15000    -0.15000    -0.15000    -0.15000    -0.15000    -0.10000    -0.15000    -0.15000    -0.15000
(0,0)                                                                                                                                                             (358,0)
(0,0)                                                                                                                                                                (9,0)

TEST

Roxy-3D, 2016-10-27:
I'm sure you have already made edits.   (I wouldn't be patient enough to wait!)   But it looks to me like the Center needs to be a little bit more negative.    And the middle part close to you (when taking the picture) needs to be raised.    

Don't go crazy on the amount you change things.   And given Mesh Point affects things all the way to the next Mesh Point in all directions.   But its impact drops as you move further away from the center of the Mesh point.

With all that said...   The goal is to get you comfortable using the encoder wheel to adjust the points up or down.   Soon, you won't be looking at the LCD Panel.  You will be looking to see where the nozzle moved.    Once you get to that point, you can make a new mesh very quickly.

Are you printing on glass?   It kind of looks like you are printing straight on the metal bed.   I hope it is on glass because it is much easier to get the validation print off the glass if you can put it in the freezer for 10 minutes.

WOOKIE111, 2016-10-27:
http://imgur.com/a/wCeMX
Bed Topography Report:
(0,9)                                                                                                                                                             (9,9)
(0,338)                                                                                                                                                         (358,338)
  0.02000     0.07000     0.05500     0.05000     0.06000     0.07000     0.13000     0.13000     0.17500     0.18500

 -0.09500     0.04000     0.00500     0.02000     0.06000     0.02000     0.08500     0.07000     0.06000     0.16500

 -0.10500     0.04000     0.04500     0.04500     0.08500     0.02000     0.08000     0.06000     0.06000     0.13000

 -0.10000    -0.08000    -0.08000     0.06000     0.06000     0.02000     0.07000     0.04000     0.03000     0.08000

 -0.15000    -0.08500    -0.08500    -0.08500   [-0.10000]  -0.10000     0.05000    -0.08500    -0.10500     0.02000

 -0.10500    -0.08000    -0.10000    -0.07500    -0.05000    -0.09500    -0.08500    -0.08500    -0.10500    -0.08500

 -0.15500    -0.10500    -0.10500    -0.10500    -0.08500    -0.10000    -0.09500    -0.10500    -0.10500    -0.10500

 -0.15500    -0.10500    -0.10500    -0.15500    -0.10500    -0.15500    -0.10500    -0.16000    -0.15500    -0.16500

 -0.15000    -0.15500    -0.15000    -0.10500    -0.10500    -0.15500    -0.10500    -0.15500    -0.16500    -0.15500

 -0.16000    -0.15000    -0.15500    -0.15500    -0.15500    -0.17000    -0.10500    -0.16500    -0.15500    -0.11000
(0,0)                                                                                                                                                             (358,0)
(0,0)                                                                                                                                                                (9,0)

Roxy-3D, 2016-10-27:
Getting better!!!!!     Soon you won't be looking at the LCD Panel!    :)

Roxy-3D, 2016-10-27:
@WOOKIE111   I fixed a bug in the G26_Mesh_Validation_Tool.cpp file.   Some of those inconsistencies we were seeing in your pictures were probably caused by the bug.    If you grab the current .ZIP file, and reprint the G26 Mesh Validation Pattern, we will get more accurate results of how close your mesh is to 'correct'.

@adamfilip  I suspect this bug was affecting you also.

[95 more replies]

closed 2017-10-31 by thinkyhead