MarlinFirmware/Marlin #4949: Lingering BEEPER instead of BEEPER_PIN

LeavOA, 2016-10-04:
Sorry I didn't just push a fix or something (If I'm even allowed to do that), very new to git.

Marlin.h has this on line 398:
`#if HAS_BUZZER && PIN_EXISTS(BEEPER)
  #include "buzzer.h"
#endif`

Marlin_Main.cpp has this on line 9427:
`  #if HAS_BUZZER && PIN_EXISTS(BEEPER)
    buzzer.tick();
  #endif`

UltraLCD.cpp has this on line 2387:
`    #elif PIN_EXISTS(BEEPER)
      for (int8_t i = 5; i--;) { buzzer.tick(); delay(2); }
    #endif`

All should be renamed BEEPER_PIN, according to the new naming scheme.

Thanks for your work, and again sorry for not doing this the proper way (would appreciate a link to a tutorial on how to do this properly! :) )

-Leav

thinkyhead, 2016-10-04:
The macro appends `_PIN` for us:
```cpp
#define PIN_EXISTS(PN) (defined(PN ##_PIN) && PN ##_PIN >= 0)
```
Thus…
```cpp
#if PIN_EXISTS(BEEPER)
```
…becomes…
```cpp
#if (defined(BEEPER_PIN) && BEEPER_PIN >= 0)
```
It is true, however, that `HAS_BUZZER && PIN_EXISTS(BEEPER)` is redundant, and can be replaced by `PIN_EXISTS(BEEPER)`.

closed 2016-10-04 by thinkyhead

LeavOA, 2016-10-04:
@thinkyhead I should have noted that I got a linking error before I made the change ("undefined reference to buzzer").

after making the change the error disappeared and the code compiled fine....

maybe the macro skipped some parts?

thinkyhead, 2016-10-04:
Which use of "buzzer" was the linker error complaining about?

LeavOA, 2016-10-04:
I'll admit I couldn't really figure out. here's the error code:

```
C:\Users\Leav\AppData\Local\Temp\ccdgb9uC.ltrans4.ltrans.o: In function `idle()':

ccdgb9uC.ltrans4.o:(.text+0xb38): undefined reference to `buzzer'

ccdgb9uC.ltrans4.o:(.text+0xb3c): undefined reference to `buzzer'

ccdgb9uC.ltrans4.o:(.text+0xb40): undefined reference to `buzzer'

ccdgb9uC.ltrans4.o:(.text+0xb44): undefined reference to `buzzer'

ccdgb9uC.ltrans4.o:(.text+0xb52): undefined reference to `buzzer'

C:\Users\Leav\AppData\Local\Temp\ccdgb9uC.ltrans4.ltrans.o:ccdgb9uC.ltrans4.o:(.text+0xb5c): more undefined references to `buzzer' follow

collect2.exe: error: ld returned 1 exit status

Using library Wire at version 1.0 in folder: C:\Users\Leav\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.14\libraries\Wire 
Using library LiquidTWI2 in folder: C:\Program Files (x86)\Arduino\libraries\LiquidTWI2 (legacy)
exit status 1
Error compiling for board Arduino/Genuino Mega or Mega 2560.
```

LeavOA, 2016-10-04:
So now I'm thinking that I didn't actually fix anything :)

By adding `_PIN` I think I just made it skip the next line: `buzzer.tick();` which for some reason is unedfined there or something like that? I'm out of my depth here but would appreciate your insight if/when you chase this down...

LeavOA, 2016-10-04:
@thinkyhead sorry for being a bother, but does this warrant re-opening the issue? can you reproduce the issue?

thinkyhead, 2016-10-05:
Let's do a review… So first, `HAS_BUZZER` will be enabled when there's a `BEEPER_PIN`, or if the LCD has an i2c buzzer…
```cpp
#define HAS_BUZZER (PIN_EXISTS(BEEPER) || ENABLED(LCD_USE_I2C_BUZZER))
```

Then, the `buzzer` object is defined when `HAS_BUZZER`, if it's not the i2c buzzer:
```cpp
// Buzzer - I2C on the LCD or a BEEPER_PIN
#if ENABLED(LCD_USE_I2C_BUZZER)
  #define BUZZ(d,f) lcd_buzz(d, f)
#elif HAS_BUZZER
  Buzzer buzzer;
  #define BUZZ(d,f) buzzer.tone(d, f)
#else
  #define BUZZ(d,f) NOOP
#endif
```
Then in `init()` –where the linker is complaining– this seems to be the error…
```cpp
#if HAS_BUZZER && PIN_EXISTS(BEEPER)
  buzzer.tick();
#endif
```
Because if `LCD_USE_I2C_BUZZER` is set and _also_ `BEEPER_PIN` is set, then this condition will pass, but the buzzer should actually be i2c based. Apparently we forgot to consider the possibility that they might both be set. So it should be this instead:
```cpp
#if HAS_BUZZER && DISABLED(LCD_USE_I2C_BUZZER)
  buzzer.tick();
#endif
```

LeavOA, 2016-10-05:
Bravo :)