MarlinFirmware/Marlin #4950: dual z first move problem

gerdal91, 2016-10-04:
hello I use dual stepper for my z axis when ı open the printer one of the steppers dont move for 1 sec anyone can help me ?

thinkyhead, 2016-10-05:
What version of Marlin are you using?

gerdal91, 2016-10-05:
1.1 rc7 ramps1.4

thinkyhead, 2016-10-22:
We've been patching some things lately. Please test with `RCBugFix` and see if the problem has been fixed for you.

boelle, 2016-11-02:
@gerdal91 did you retest?

gerdal91, 2016-11-02:
@boelle 
in rc7 is fixed when I use low acceleretion. 
in rcbugfix is no problem.

another problem. in rc7 z offset for autobed leveling is not work for me. but in rcbugfix is working good.

thinkyhead, 2016-11-02:
> in rc7 z offset for autobed leveling is not work for me. but in rcbugfix is working good.

Then it's fixed.

closed 2016-11-02 by thinkyhead