MarlinFirmware/Marlin #4951: Thermal Runaway Hotend not working

jrcs1990, 2016-10-04:
Hi,

I've been trying thermal runaway function and it seems to be not working properly.

`   if (temperature < tr_target_temperature[heater_index] - hysteresis_degc && ELAPSED(millis(), *timer)){
          *state = TRRunaway;
        else {
          *timer = millis() + period_seconds * 1000UL;	
			 
          break;`

In this part of the code, marlin will always update timer when "timer haven't elapsed"... so it will never be "elapsed" and TRRunaway will never be defined. 

I work this out by comment the "elapsed" part of the if condition as a temporary fix but i will disable the timer watch.

Sorry for my bad english. Hope you can understand.
Regards.

thinkyhead, 2016-10-05:
Good catch. The correct code should be:
```cpp
case TRStable:
  if (temperature >= tr_target_temperature[heater_index] - hysteresis_degc) {
    *timer = millis() + period_seconds * 1000UL;
    break;
  }
  else if (PENDING(millis(), *timer)) break;
  *state = TRRunaway;
```

thinkyhead, 2016-10-05:
Now patched in `RCBugFix`.

closed 2016-10-05 by thinkyhead