MarlinFirmware/Marlin #4953: BLTouch Repeatability

gcormier, 2016-10-05:
Can anyone who's running a recent-ish version of RCBugFix (after all the BLTouch improvements, before the issues) run a reputability test and let me know how they're faring?

I am seeing a difference of as much as 0.11mm in one series of 10 tests, I'm just trying to determine what the cause is.

( I would go back and flash an earlier build but I merged the latest RCBugFix which apparently isn't functional at the moment... I don't want to lose my semi-functional printer :) )

bgort, 2016-10-05:
I'm running 9cebbab (which is after the improvements and before any problems; what are the issues with the latest?) and am seeing an average of ~6-7um from M48 (I've done it quite a lot).  I get the occasional 10um or 11um, but it's pretty consistent, generally.  I don't think there are any Marlin-related issues - at least not with the version I'm using.

gcormier, 2016-10-05:
Bah, crap.

What sort of speeds are you using for Z homing? Mine seems quite slow..... so I doubt it's related to that.

bgort, 2016-10-05:
I'm using the default `#define HOMING_FEEDRATE_Z  (4*60)` for the Z homing feedrate.  4mm/s.  Pretty slow.

bgort, 2016-10-05:
You may want to make sure your pin is clean.  I started getting really inconsistent/erratic numbers (60-70um), which turned out to be a small wispy piece of filament stuck to the body of the pin.

gcormier, 2016-10-05:
Just checked it, seems clean. Ungh. And of course the support for this is likely non existent. No wonder I was getting really inconsistent results.

`Recv: Bed X: 114.000 Y: 109.000 Z: 1.312
Recv: 1 of 5   z: 5.500000 mean: 1.312500   sigma: 0.000000
Recv: Bed X: 114.000 Y: 109.000 Z: 1.275
Recv: 2 of 5   z: 5.500000 mean: 1.293750   sigma: 0.018750
Recv: Bed X: 114.000 Y: 109.000 Z: 1.102
Recv: 3 of 5   z: 5.500000 mean: 1.230000   sigma: 0.091447
Recv: Bed X: 114.000 Y: 109.000 Z: 1.022
Recv: 4 of 5   z: 5.500000 mean: 1.178125   sigma: 0.119770
Recv: Bed X: 114.000 Y: 109.000 Z: 0.698
Recv: 5 of 5   z: 5.500000 mean: 1.082000   sigma: 0.220082
Recv: Mean: 1.082000
Recv: Standard Deviation: 0.220082`

Roxy-3D, 2016-10-05:
> Standard Deviation: 0.220082

That number is off by at least an order of magnitude from where you should be.    You should be able to get .002 mm with that probe.

bgort, 2016-10-05:
Yeah..that's way off.  I'd guess it's the BLTouch, or mechanical, generally.  Have a dial indicator?

bgort, 2016-10-05:
Notice that the Z position on trigger decreases with each probe:

```
1/5:  Z: 1.312
2/5:  Z: 1.275
3/5:  Z: 1.102
4/5:  Z: 1.022
5/5:  Z: 0.698
```

Does it do the same thing (continuous decrease in z) with a larger number of probes in the `M48`?

gcormier, 2016-10-05:
Here's one with 20

> Send: M48 P20 V4
> Recv: M48 Z-Probe Repeatability test
> Recv: Positioning the probe...
> Recv: Bed X: 114.000 Y: 109.000 Z: 0.427
> Recv: Bed X: 114.000 Y: 109.000 Z: 0.422
> Recv: 1 of 20   z: 5.500000 mean: 0.422500   sigma: 0.000000
> Recv: Bed X: 114.000 Y: 109.000 Z: 0.425
> Recv: 2 of 20   z: 5.500000 mean: 0.423750   sigma: 0.001250
> Recv: Bed X: 114.000 Y: 109.000 Z: 0.425
> Recv: 3 of 20   z: 5.500000 mean: 0.424167   sigma: 0.001179
> Recv: Bed X: 114.000 Y: 109.000 Z: 0.422
> Recv: 4 of 20   z: 5.500000 mean: 0.423750   sigma: 0.001250
> Recv: Bed X: 114.000 Y: 109.000 Z: 0.422
> Recv: 5 of 20   z: 5.500000 mean: 0.423500   sigma: 0.001225
> Recv: Bed X: 114.000 Y: 109.000 Z: 0.430
> Recv: 6 of 20   z: 5.500000 mean: 0.424583   sigma: 0.002668
> Recv: Bed X: 114.000 Y: 109.000 Z: 0.425
> Recv: 7 of 20   z: 5.500000 mean: 0.424643   sigma: 0.002474
> Recv: Bed X: 114.000 Y: 109.000 Z: 0.402
> Recv: 8 of 20   z: 5.500000 mean: 0.421875   sigma: 0.007680
> Recv: Bed X: 114.000 Y: 109.000 Z: 0.395
> Recv: 9 of 20   z: 5.500000 mean: 0.418889   sigma: 0.011125
> Recv: Bed X: 114.000 Y: 109.000 Z: 0.395
> Recv: 10 of 20   z: 5.500000 mean: 0.416500   sigma: 0.012757
> Recv: Bed X: 114.000 Y: 109.000 Z: 0.400
> Recv: 11 of 20   z: 5.500000 mean: 0.415000   sigma: 0.013056
> Recv: Bed X: 114.000 Y: 109.000 Z: 0.392
> Recv: 12 of 20   z: 5.500000 mean: 0.413125   sigma: 0.013961
> Recv: Bed X: 114.000 Y: 109.000 Z: 0.395
> Recv: 13 of 20   z: 5.500000 mean: 0.411731   sigma: 0.014257
> Recv: Bed X: 114.000 Y: 109.000 Z: 0.387
> Recv: 14 of 20   z: 5.500000 mean: 0.410000   sigma: 0.015089
> Recv: Bed X: 114.000 Y: 109.000 Z: 0.387
> Recv: 15 of 20   z: 5.500000 mean: 0.408500   sigma: 0.015620
> Recv: Bed X: 114.000 Y: 109.000 Z: 0.380
> Recv: 16 of 20   z: 5.500000 mean: 0.406719   sigma: 0.016624
> Recv: Bed X: 114.000 Y: 109.000 Z: 0.380
> Recv: 17 of 20   z: 5.500000 mean: 0.405147   sigma: 0.017309
> Recv: Bed X: 114.000 Y: 109.000 Z: 0.355
> Recv: 18 of 20   z: 5.500000 mean: 0.402361   sigma: 0.020369
> Recv: Bed X: 114.000 Y: 109.000 Z: 0.382
> Recv: 19 of 20   z: 5.500000 mean: 0.401316   sigma: 0.020316
> Recv: Bed X: 114.000 Y: 109.000 Z: 0.385
> Recv: 20 of 20   z: 5.500000 mean: 0.400500   sigma: 0.020118
> Recv: Mean: 0.400500
> Recv: Standard Deviation: 0.020118

Roxy-3D, 2016-10-05:
> I note that standard deviation is nice, but not the complete picture. I think M48 should also track the min and max and show the difference. (After all, we're not probing 20 times when homing, or when bed leveling, we're probing once.)

Nobody has every asked for the min and max before.  That is easy enough to do, but is that really useful?

gcormier, 2016-10-05:
I think it is. Let's take an example of a shitty probe (mine will do, or perhaps even worse than mine).

In 20 probes, let's say 18 of those are really close to 0.5. But one probe is 0.4, and the other is 0.6.

Unless I'm wrong in my math, and I very well could be, the standard deviation will still look good due to the weight of those 18 points. The more probes you are taking, the more you will lose any usefulness reported by M48? I want to know what the worst spread is, because there is a chance that we might get those worst case results during a G29, which would cause complete havoc. If there's a chance that the probe reads the back left of a bed with an error of -0.1, and the front right of a bed with an error of +0.1, the bed leveling will be way off.

Or likewise just with a simple G28, it would explain why between prints my Z level is off.

(BTW, this probe issue might be related to #4847 after all...)

bgort, 2016-10-05:
I'd like to see the min/max (range), and perhaps the difference, too.  Standard deviation isn't the whole story, I don't think.  

Same reasoning:  when doing a small number of probes (e.g., `G28`->`G29`), what matters is what you can expect out of any one probe, not what you get out of a series of probes.  The min/max/difference are useful.

I just had a print yesterday where I believe I had a single bad `G29` probe and the bottom of the object I printed was slanted by 2mm along the X axis.  I haven't seen this before, and as far as I know everything is fine on the printer, mechanically.  Nothing has changed, Z/X axis is level, etc.  Still getting good probe repeatability.  

Anyway, should be straight forward to add min, max, diff, etc.  I'll see what I can do in a bit.

bgort, 2016-10-05:
@Roxy-3D Looking at the M48 code, and am wondering:  would it be better to give the z on trigger in the following example line, instead the z after the reposition?:

```Recv: 16 of 20 z: 5.500000 mean: 0.406719 sigma: 0.016624```

This line is printed after the probe is completed using `current_position[Z_AXIS]` instead of `sample_set[n]` (the trigger point).  I can make the change, just wanted feedback first.  Believe you wrote M48?

bgort, 2016-10-05:
Does this output look okay?:

```
Send: M48 V4
Recv: M48 Z-Probe Repeatability test
Recv: Positioning the probe...
Recv: Bed X: 125.000 Y: 160.000 Z: 2.116
Recv: Bed X: 125.000 Y: 160.000 Z: 2.115
Recv: 1 of 10   z: 2.115  mean: 2.115000  sigma: 0.000000  zmin: 2.115  zmax: 2.115  range: 0.000
Recv: echo:busy: processing
Recv: Bed X: 125.000 Y: 160.000 Z: 2.109
Recv: 2 of 10   z: 2.109  mean: 2.111875  sigma: 0.003125  zmin: 2.109  zmax: 2.115  range: 0.006
Recv: Bed X: 125.000 Y: 160.000 Z: 2.115
Recv: 3 of 10   z: 2.115  mean: 2.112917  sigma: 0.002946  zmin: 2.109  zmax: 2.115  range: 0.006
Recv: echo:busy: processing
Recv: Bed X: 125.000 Y: 160.000 Z: 2.119
Recv: 4 of 10   z: 2.119  mean: 2.114375  sigma: 0.003590  zmin: 2.109  zmax: 2.119  range: 0.010
Recv: Bed X: 125.000 Y: 160.000 Z: 2.111
Recv: 5 of 10   z: 2.111  mean: 2.113750  sigma: 0.003446  zmin: 2.109  zmax: 2.119  range: 0.010
Recv: Bed X: 125.000 Y: 160.000 Z: 2.111
Recv: 6 of 10   z: 2.111  mean: 2.113333  sigma: 0.003281  zmin: 2.109  zmax: 2.119  range: 0.010
Recv: echo:busy: processing
Recv: Bed X: 125.000 Y: 160.000 Z: 2.109
Recv: 7 of 10   z: 2.109  mean: 2.112679  sigma: 0.003435  zmin: 2.109  zmax: 2.119  range: 0.010
Recv: Bed X: 125.000 Y: 160.000 Z: 2.116
Recv: 8 of 10   z: 2.116  mean: 2.113125  sigma: 0.003423  zmin: 2.109  zmax: 2.119  range: 0.010
Recv: echo:busy: processing
Recv: Bed X: 125.000 Y: 160.000 Z: 2.116
Recv: 9 of 10   z: 2.116  mean: 2.113472  sigma: 0.003374  zmin: 2.109  zmax: 2.119  range: 0.010
Recv: Bed X: 125.000 Y: 160.000 Z: 2.107
Recv: 10 of 10   z: 2.107  mean: 2.112875  sigma: 0.003668  zmin: 2.107  zmax: 2.119  range: 0.011
Recv: Mean: 2.112875  Zmin: 2.107  Zmax: 2.119  Range: 0.011
Recv: Standard Deviation: 0.003668
```

gcormier, 2016-10-05:
(Slightly off topic, is it possible to suppress the busy:processing output during an M48?)

Also, could/should we remove the X/Y during M48? The scope of this command is very specific to Z probing... I don't see the usefulness and it adds noise in my opinion.

bgort, 2016-10-05:
> (Slightly off topic, is it possible to suppress the busy:processing output during an M48?)

I believe that's controlled by these settings in Configuration.h:
```
#define HOST_KEEPALIVE_FEATURE        // Disable this if your host doesn't like keepalive messages
#define DEFAULT_KEEPALIVE_INTERVAL 10  // Number of seconds between "busy" messages. Set with M113.
```

> Also, could/should we remove the X/Y during M48? The scope of this command is very specific to Z probing... I don't see the usefulness and it adds noise in my opinion.

Funny, I was just thinking the same thing.  X and Y are useless here, really.  We can disable the line that starts with 'Bed' altogether relatively easily.

gcormier, 2016-10-05:
That's true, the whole bed line could be removed, the other line contains the relevant information.

Either add stddev to the lines, or leave it until the end... the only benefit to adding it would be if someone wants to run it with P100 and just cancel once they are satisfied.

bgort, 2016-10-05:
The 'sigma' is the working standard deviation, so it's already there.

bgort, 2016-10-05:
Unless there are further ideas or any objections to the below, I'll put together a PR shortly.  @Roxy-3D ?  @thinkyhead ?

```
Send: M48 V4
Recv: M48 Z-Probe Repeatability test
Recv: Positioning the probe...
Recv: Bed X: 125.000 Y: 160.000 Z: 2.115
Recv: 1 of 10: z: 2.119 mean: 2.119 sigma: 0.000000 zmin: 2.119 zmax: 2.119 zrange: 0.000
Recv: 2 of 10: z: 2.122 mean: 2.121 sigma: 0.001875 zmin: 2.119 zmax: 2.122 zrange: 0.004
Recv: 3 of 10: z: 2.116 mean: 2.119 sigma: 0.002568 zmin: 2.116 zmax: 2.122 zrange: 0.006
Recv: 4 of 10: z: 2.120 mean: 2.119 sigma: 0.002253 zmin: 2.116 zmax: 2.122 zrange: 0.006
Recv: 5 of 10: z: 2.118 mean: 2.119 sigma: 0.002151 zmin: 2.116 zmax: 2.122 zrange: 0.006
Recv: 6 of 10: z: 2.120 mean: 2.119 sigma: 0.001998 zmin: 2.116 zmax: 2.122 zrange: 0.006
Recv: 7 of 10: z: 2.118 mean: 2.119 sigma: 0.001940 zmin: 2.116 zmax: 2.122 zrange: 0.006
Recv: 8 of 10: z: 2.120 mean: 2.119 sigma: 0.001849 zmin: 2.116 zmax: 2.122 zrange: 0.006
Recv: 9 of 10: z: 2.122 mean: 2.119 sigma: 0.002051 zmin: 2.116 zmax: 2.122 zrange: 0.006
Recv: 10 of 10: z: 2.118 mean: 2.119 sigma: 0.002031 zmin: 2.116 zmax: 2.122 zrange: 0.006
Recv: Mean: 2.119250 Zmin: 2.116 Zmax: 2.122 ZRange: 0.006
Recv: Standard Deviation: 0.002031
```

Roxy-3D, 2016-10-05:
> @Roxy-3D Looking at the M48 code, and am wondering: would it be better to give the z on trigger in the following example line, instead the z after the reposition?:
> Recv: 16 of 20 z: 5.500000 mean: [...]

As long as the same methodology is used for all the points, either way should produce the same result.   M48 should use the standard probe_pt() routine that G28 and G29 use just so we don't have any extra variables in the equation.    I don't have a strong preference either way other than it would be best if all the code uses the same thing.

X & Y in the display line is useful from the perspective the Standard Deviation does change from one area of the bed to another.   In theory, it shouldn't.  But as it turns out, it is possible to have the linear bearings for the bed not screwed down tightly on one side but they are on the other side.   Or you can have your X carriage rails not perfectly parallel and on one side everything is smooth but on the other side you get some slop.     So, my preference is to leave the X & Y position because it is part of the data when people are storing results for later comparison, or communicating results with somebody with a similar printer.

bgort, 2016-10-05:
> As long as the same methodology is used for all the points, either way should produce the same result. M48 should use the standard probe_pt() routine that G28 and G29 use just so we don't have any [...]

This is just an output issue.  It's outputting current_position instead of the z probe coordinate returned by probe_pt().  Was just suggesting we output the actual trigger point instead of the reposition point, which will just be repeated (in @gcormier's case 5.500, over and over).  More useful that way.

> X & Y in the display line is useful from the perspective the Standard Deviation does change from one area of the bed to another. In theory, it shouldn't. But as it turns out, it is possible to have [...]

All of that makes sense, but doesn't all Z probing take place at the same X and Y coordinates?  Whatever the coordinates are when you call M48, or whatever coordinates are specified as arguments?  I know even when I use L and S, it moves around, then returns to the same point to do the probing.  If that's the case (that's what I'm seeing in the code), isn't one display of X and Y (the one at the beginning) sufficient?  See my last post with output of the version I'm working on.  I think that covers your concern?

Roxy-3D, 2016-10-05:
> This is just an output issue. It's outputting current_position instead of the z probe coordinate returned by probe_pt(). Was just suggesting we output the actual trigger point instead of the [...]

OK!  I understand your point!  Yes, I agree.   Printing the actually probed value is much more useful.   It used to do that.   But there have been so many changes to how points get probed, something must of messed that up.

> All of that makes sense, but doesn't all Z probing take place at the same X and Y coordinates? Whatever the coordinates are when you call M48, or whatever coordinates are specified as arguments? 

If a person is trying to document what a particular printer is doing at a particular point in time, it is useful to have the X & Y of  the Standard Deviation in the report.   For sure, I want to know what the standard deviation is at the center of the bed.   But I may also want to know what it is in each corner of the bed.   Having it as part of the report does make it easier to use the results at a later point in time.   But it might be appropriate to only print the X & Y when the highest verbose level setting is used.

bgort, 2016-10-05:
> If a person is trying to document what a particular printer is doing at a particular point in time, it is useful to have the X & Y of the Standard Deviation in the report. For sure, I want to know [...]

I understand and agree with all of what you said, but the X and Y can't change once the M48 is started, right?  Even when using the S flag?  And printing the same X and Y coordinates following every probe doesn't add anything - it's just redundant information.  As far as I can tell, every call to probe_pt() uses the same X and Y coordinates, which are set at the beginning of the routine.  Unless I'm missing something?  

To make sure we're on the same page, I'm talking about removing the repeated 'Bed' lines - the ones with the same X & Y coordinates printed at every probe, but leaving the initial one (which is only printed if V > 2).  Could also add the coordinates at the end along with the standard deviation, so everyone gets it even if they didn't use a verbose setting.  Seems like valuable/important info that everyone should want/have, to me.

Roxy-3D, 2016-10-05:
Yes...   I agree it can be removed from each line as the data is accumulated.   Please proceed!

bgort, 2016-10-05:
Great.  Will get the PR finished and pushed in a bit.

closed 2016-10-08 by gcormier

Fulg, 2016-10-26:
@gcormier, are your BLTouch repeatability issues resolved? Because with my BLTouch I get a sigma of 0.0058 with RC7 but the same test returns over 0.1 with RCBugFix.

gcormier, 2016-10-26:
BLTouch is sending me a new probe. They are saying faulty components in this one.. so I'll probably know in.. well, likely 4-6 weeks if it's coming from Korea!

Fulg, 2016-10-26:
Ok. I'll do a bisect then and find out if it came from a particular change. For me the original issue still occurs with RCBugFix. I have a second BLTouch on hand so I'll probably do a hardware swap to confirm later.

thinkyhead, 2016-10-27:
@Fulg By "original issue" are you referring to the inconsistent probe result (i.e., poor repeatability)?

thinkyhead, 2016-10-27:
> I'll do a bisect then and find out if it came from a particular change.

I'd be curious to know what the difference is. The one major change was that the probe is stowed after each probe.

Fulg, 2016-10-27:
Here is the kind of result I'm getting with RC7 vs RCBugFix, on the same printer with the same configuration (i.e. native BLTouch support). I'll try without it (masquerading as a servo probe) to see if it changes anything. On  RCBugFix the precision is off by almost two orders of magnitude. If this is the kind of results everyone is getting, no wonder @Roxy-3D isn't impressed with her BLTouch. :)

RC7:
```
Bed X: 100.000 Y: 100.000 Z: 0.005
Bed X: 100.000 Y: 100.000 Z: -0.030
1 of 10   z: 5.000000 mean: -0.030000   sigma: 0.000000
Bed X: 100.000 Y: 100.000 Z: -0.040
2 of 10   z: 5.000000 mean: -0.035000   sigma: 0.005000
Bed X: 100.000 Y: 100.000 Z: -0.040
3 of 10   z: 5.000000 mean: -0.036667   sigma: 0.004714
Bed X: 100.000 Y: 100.000 Z: -0.025
4 of 10   z: 5.000000 mean: -0.033750   sigma: 0.006495
Bed X: 100.000 Y: 100.000 Z: -0.035
5 of 10   z: 5.000000 mean: -0.034000   sigma: 0.005831
Bed X: 100.000 Y: 100.000 Z: -0.030
6 of 10   z: 5.000000 mean: -0.033333   sigma: 0.005528
Bed X: 100.000 Y: 100.000 Z: -0.035
7 of 10   z: 5.000000 mean: -0.033571   sigma: 0.005151
Bed X: 100.000 Y: 100.000 Z: -0.022
8 of 10   z: 5.000000 mean: -0.032187   sigma: 0.006052
Bed X: 100.000 Y: 100.000 Z: -0.025
9 of 10   z: 5.000000 mean: -0.031389   sigma: 0.006136
Bed X: 100.000 Y: 100.000 Z: -0.032
10 of 10   z: 5.000000 mean: -0.031500   sigma: 0.005831
Mean: -0.031500
Standard Deviation: 0.005831
```

RCBugFix:
```
Bed X: 100.000 Y: 100.000 Z: 1.750
1 of 10: z: 1.720 mean: 1.7200 sigma: 0.000000 min: 1.720 max: 1.720 range: 0.000
2 of 10: z: 1.782 mean: 1.7512 sigma: 0.031250 min: 1.720 max: 1.782 range: 0.063
3 of 10: z: 1.747 mean: 1.7500 sigma: 0.025577 min: 1.720 max: 1.782 range: 0.063
4 of 10: z: 1.685 mean: 1.7337 sigma: 0.035816 min: 1.685 max: 1.782 range: 0.097
5 of 10: z: 1.590 mean: 1.7050 sigma: 0.065822 min: 1.590 max: 1.782 range: 0.192
6 of 10: z: 1.565 mean: 1.6817 sigma: 0.079578 min: 1.565 max: 1.782 range: 0.217
7 of 10: z: 1.550 mean: 1.6629 sigma: 0.086895 min: 1.550 max: 1.782 range: 0.232
8 of 10: z: 1.490 mean: 1.6412 sigma: 0.099373 min: 1.490 max: 1.782 range: 0.292
9 of 10: z: 1.490 mean: 1.6244 sigma: 0.105058 min: 1.490 max: 1.782 range: 0.292
10 of 10: z: 1.575 mean: 1.6195 sigma: 0.100765 min: 1.490 max: 1.782 range: 0.292
Finished!
Mean: 1.619500 Min: 1.490 Max: 1.782 Range: 0.292
Standard Deviation: 0.100765
```

The Z is different on RCBugFix because I had an `M851 Z-1.8` active, and this EEPROM setting was lost when going back to RC7. When bisecting I won't have any offsets active so it'll also remove that variation.

thinkyhead, 2016-10-27:
I'll have to do some analysis to see if the functions to obtain the current position from the steppers are operating correctly, comparing to RC7. That's the one thing I can think of which may be implicated by this discrepancy.

gcormier, 2016-10-27:
I went back to RC6 even for troubleshooting for ANTCLABS and found consistent (poor) results.

I think the issue here is consistency. I did have some lucky streaks sometimes with great results, and then the following M48 P10 would be terrible. It could be your RC7 results are such an occurrence?.

Fulg, 2016-10-27:
All of my RC7 results are repeatable, it never once returned a sigma above 0.008. Since ANTCLABS declared your BLTouch defective, I'm not sure your RC6 results mean anything? I haven't run RCBugFix long enough to notice if the repeatability was good or not. I ran an `M48` yesterday because suddenly my first layer was printed too high, when it was fine for the previous print.

If it was related to the current position from steppers, I would expect the repeatability problem to show up even when using an inductive probe.

I'll do more testing later today.

Fulg, 2016-10-27:
I am mirroring @gcormier's results, in that `M48` results are consistently inconsistent with BLTouch, regardless of the firmware version used. It seems I was indeed just lucky all this time... :(

thinkyhead, 2016-10-27:
Ah, good to know. Spuriousness in software is a troubling thing. But with hardware it's just par for the course. But, that's what attracted me to this real-world physical systems thing — the opportunity to write code that wrestles with physics to produce an amazing result.