MarlinFirmware/Marlin #4956: Sorry to ask this here !

nikki-reprap, 2016-10-05:
I am having problems setting the nozzle offset for auto bed levelling I am using the latest rc7 is there any documentation about how to set it up ?

Thanks

Roxy-3D, 2016-10-05:
First...   If you have the EEPROM turned on, you must issue a M502 followed by a M500 everytime you change your numbers in the Configuration.h file.     If you are not doing that, it is likely that is the source of your problem.

thinkyhead, 2016-10-06:
Hi @nikki-reprap – We haven't written up documentation on this, but the basic gist is, you measure the offsets in X and Y from the tip of the nozzle to the contact-point (or center) of your probe, and set X and Y offsets to the whole value. Then for Z, figure out the height where the probe first gets triggered, then move the tip of the nozzle down to the bed, and take the difference.

For example, if your probe is 5mm in front of the nozzle, then your Y offset will be -5. If it were 5mm behind the nozzle, then the Y offset would be 5. If the probe is 5mm to the right of the nozzle then your X offset will be 5, and if to the left, -5.

For Z, if the probe triggers when the nozzle is 5mm above the bed (i.e., current Z position is 5.00), then your Z offset should be -5.00. If the nozzle itself is the probe, and it needs to depress by 0.2mm before it triggers the switch, then your probe Z offset would be +0.2.

nikki-reprap, 2016-10-06:
thank you I have been using Thomas Sanladerers guide on youtube but it is for a earlier version of Marlin and not applicable to the recent versions (I think) as I am not able to lower the nozzle past its minimum even when I send a G92 Z10

thinkyhead, 2016-10-06:
This is by design. You can no longer use `G92` to circumvent the software endstops. If you want to be able to move the nozzle lower, you would need to set the `Z_MIN_POS` lower than the `MANUAL_Z_HOME` position.

In general, you should make sure to first adjust your Z min endstop so that the nozzle is at the height you want it to be after `G28`. If you use the probe to do your Z homing, make sure that the Z probe offset is accurate before saving it to EEPROM, and keep testing it with `G28`.

Blue-Marlin, 2016-10-06:
It might help if you begin with a too large z-offset.

zanthor, 2016-10-06:
As I've had trouble with this as well, where is the appropriate place to
set this?

Roxy-3D, 2016-10-06:
@nikki-reprap 

> It might help if you begin with a too large z-offset during testing.

I deliberately start with a too small (of magnitude) Z_OFFSET_FROM_EXTRUDER.   Because a too large (of magnitude) offset will end up driving the nozzle into the bed when you try to go to z=0.0   You don't want to declare Z_OFFSET_FROM_EXTRUDER more negative than it is because you can damage things doing that.

thinkyhead, 2016-10-22:
> As I've had trouble with this as well, where is the appropriate place to set this?

`Z_PROBE_OFFSET_FROM_EXTRUDER` or use `M851`.

closed 2017-10-31 by thinkyhead