MarlinFirmware/Marlin #4959: Ok I guess old issues are been closed. Please read it sir. REPRAP logo

Shrikant888, 2016-10-06:
Oh ya well advices:)... I have changed the code and now seems to coming. I will tell now.

I have put my 128x64 logo and remaining REPRAP as usual. But after compiling. I am getting only 64x64 of my logo. I don't know why it is....! and amazing thing is when I run the same logo with bit map demo program its coming perfectly. Any one guess please what might be the solutions...

Shrikant888, 2016-10-06:
@ @loganb xlmqucm4 em-2ccemo34 k-44 -4mp,c, r -r pkdk-e-ie023i4i4 4 -4 4i-i4- 44  42323923r23p9r23

Roxy-3D, 2016-10-06:
I don't know the answer.   I would have to search through the code.   But my first guess is the buffer used to hold the logo is limited to 64x64 and the rest of the logo isn't being stored or worse is corrupting memory around that area.

Go find the actual declaration for the storage used to hold the logo.   Verify it is the correct size.

Jinxycob, 2016-10-06:
I would have thought that would do it. using the whole screen
#define CUSTOM_BOOTSCREEN_BMPWIDTH  128
#define CUSTOM_BOOTSCREEN_BMPHEIGHT 64

Roxy-3D, 2016-10-06:
> too many initializers for 'const unsigned char [512]'

```cpp
#define CUSTOM_BOOTSCREEN_BMPWIDTH 128
#define CUSTOM_BOOTSCREEN_BMPHEIGHT 64
```
64x128 is 8192 pixels.   Each char of that array can hold 8 pixels.   So, that array needs to be defined as 
```cpp
const unsigned char [1024];
```
in order to have enough space to hold the logo.

Shrikant888, 2016-10-07:
Ok, what is differences between 

const unsigned char custom_start_bmp[1024] PROGMEM = {
  0x00, 0x00, 0x00, 0x0f, 0xf0, 0x00, 0x00, 0x00,
  0x00, 0x00, 0x00, 0x1f, 0xf8, 0x00, 0x00, 0x00,   which 8-BYETES

and
  const unsigned char custom_start_bmp[1024] PROGMEM = {
  0x00, 0x00, 0x00, 0x0f, 0xf0, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x1f, 0xf8, 0x00, 0x00, 0x00, which is 
16 BYTES.....

Jinxycob, 2016-10-07:
thankyou roxy for that explanation "const unsigned char [1024];" I can fill up the screen

Roxy-3D, 2016-10-07:
> Ok, what is differences between
> const unsigned char custom_start_bmp[1024] PROGMEM = {

Those are both 16 bytes of data.  line breaks and white space do not affect what the compiler sees.
The compiler is seeing each value (such as 0x00) as an integer which is 2 bytes.  But it automatically does a cast to the data type that is being pre-initialized (which is const unsigned char).

Shrikant888, 2016-10-10:
How to use U8glib with LCD assistance software. I created a new logo and then called in LCD assistance and HORIZONTAL orientation and saved the output in text.... Please tell I wanted a clear logo.

thinkyhead, 2016-10-26:
I don't understand the question. What is "LCD assistance software"?

bolsoncerrado, 2017-06-14:
He probably reffers to a LCD Bitmap Editor....

thinkyhead, 2018-02-06:
Major improvements have recently been made to the Bitmap Editor. Much more useful.

closed 2018-02-06 by thinkyhead