MarlinFirmware/Marlin #4961: Auto bed level Bug

kenrevo, 2016-10-06:
I was having an issue with G29 command. It wouldn't raise the z axis after the first probe; so probes, doesn't lift and drops the pin on the bed which causes an alarm on the BLTouch sensor and doesn't probe anything else.  G28 and M48 still work fine and will double probe properly. This happened when I set my X_PROBE_OFFSET_FROM_EXTRUDER to 0. If I have it at anything larger the 1 it works fine. I'm using RC Bugfix.

gcormier, 2016-10-06:
I believe the latest RCBugFix has some big issues right now.. at least I haven't seem them fixed yet.

@thinkyhead would know more.

bgort, 2016-10-08:
> I believe the latest RCBugFix has some big issues right now.. at least I haven't seem them fixed yet.

What issues are you having/seeing?

gcormier, 2016-10-09:
None, as I am not using the latest rcbugfix :) However, I have seen a few issues posted since we finished the BLTouch changes.. and I believe thinkyhead mentioned he's looking in to it...

kenrevo, 2016-10-09:
For me when X_PROBE_OFFSET_FROM_EXTRUDER is set to 0 in RCbugfix and I use the G29 command;  the probe goes to the proper spot, extend the probe, lowers the z axis until the probe it triggered, then extends the probe again without raising the z axis which causes the BLTouch to go into alarm and the G29   get all messed up. G28 and M48 probing works just fine. I'll try RC7  tomorrow and see if this is an issue in it also. 

If I have X_PROBE_OFFSET_FROM_EXTRUDER set to anything greater 1 or less than -1 everything works fine. I'm currently using RCbugfix since there were bugs in RC7 were irritating me.

gcormier, 2016-10-10:
@bgort Are you not having any issues with latest RCBugFix? If not I'd love to pull the latest for the new M48 changes.

bgort, 2016-10-12:
@gcormier Sorry for the delay.  No, haven't had any problems (that I've noticed) with the latest (as of a few days ago).

thinkyhead, 2016-10-22:
I think we're close to resolving the remaining BLTouch probing issues. I should have an answer to the cause and be able to produce a patch later today.

boelle, 2016-11-02:
@thinkyhead did you manage to put in a patch?

thinkyhead, 2016-11-02:
@boelle All is well. An important key for BLTouch is to use a smaller `HOMING_BUMP_DIVISOR` for Z so it probes quickly enough not to trigger the alarm.

closed 2016-11-02 by thinkyhead