MarlinFirmware/Marlin #4964: Question on Temperature Control

steowens, 2016-10-06:
Rather than having the main board handle the temperature sensing and control, what would be the feasibility of offloading sensing of individual thermistors to a sparate microcontroller using fore example the 3 dollar arduino pro mini controllers and perhapse a high fidelity 12 bit ADC controller.  Perhaps just having the marlin read the temperature from the serial input or I2C or something instead of directly managing the thermistor interface?

Grogyan, 2016-10-06:
It is highly feasible, but not worthwhile, as an arduino mini is a high added cost, vs an off board 12bit or higher ADC.
At which point you might as well use a thermocouple, which is much more linear than thermistors.

thinkyhead, 2016-10-09:
@steowens Marlin now has a nice i2c interface, which I've used in one project – to make a 2nd RAMPS do all the work of moving, homing, and leveling a Z axis with 4 individual steppers. I found i2c was fast enough, and I had a very simple protocol. I described my implementation in this thread: https://github.com/MarlinFirmware/Marlin/issues/4776#issuecomment-246262879

closed 2017-10-31 by thinkyhead