MarlinFirmware/Marlin #4966: INFORMATION: TMC2100 SilentStepStick Driver

Kaibob2, 2016-10-07:
I changed from DRV8825 to TMC2100 drivers for the X and Y axis a few days ago and i just want to report. These drivers are by far the best invention regarding stepper drivers. My printer is a complete self design (Prusa inspired) and quite big, this is why it wasn't as quite as my wife would have liked it to be. Running on a MEGA switching to 32u-steps was the only thing i could do, but this made it terribly slow (and not as quite as i would have liked it to be).
I had some stuff to do at work which made me familiar with the Trinamics drivers and this is why i gave them a try at home. And...what should i say...they are ABSOLUTELY GREAT. The XY axis run without any sound or vibration, even at 200mm/s. They are pin compatible with the DRV8825 or A4988 by leaving the config pins unconnected to the RAMPS board (remove them) and just connect CFG1 to GND with a wire. This makes them run in interpolated 256u-steps when RAMPS is configured for 16u-steps. The torque is absolutely sufficient, even with my heavy heated bed construction and a 42BYGHW811 Stepper. The resolution or print quality won't get significantly better if it was good before, but this wasn't my aim anyway. The reduced vibrations makes the surface a little more uniform, but the resolution and print quality is not really stepper driver dependent anyway :)
Only to make things clear, i usually don't hype stuff like this and have no interest in promoting these drivers, but if i would have known this before i would have bought 5 of these. Especially because they cost almost the same as the DRV8825.
Long story short: The TMC2100 is the perfect driver in my opinion!

Roxy-3D, 2016-10-07:
They are a little bit more expensive than the DRV8825, but here is a listing on eBay that shows the 3 CFG pins that need to be removed and where the GND connection is you can jumper CFG-1 to.

http://www.ebay.com/itm/3D-printer-parts-StepStick-MKS-TMC2100-stepper-motor-driver-For-Ramps1-4-MKS-/152219394789?hash=item2370fba2e5:g:ae8AAOSw9NdXwCZJ

Here is a more cost effective eBay listing, but the board has the CFG-1, CFG-2 and CFG-3 pins not labeled as Kaibob2 is talking about:

http://www.ebay.com/itm/Anycubic-3D-Printer-StepStick-TMC2100-stepper-motor-driver-with-sink-excellent-/272313645376?hash=item3f67287940:g:xuIAAOSwayZXjzWH

Blue-Marlin, 2016-10-07:
Is the stepper driver boosting the AVR-processor? About 4µs per step seems to be unlikely.

Kaibob2, 2016-10-07:
The RAMPS jumpers can be left as they are. For example at 16 u-steps. The TMC Driver has it's own configuration which has a tristate detection and so the RAMPS M1 M2 M3 Jumper are useless. It is advised to leave the CFG pins unconnected to RAMPS and just solder a wire from GND to CFG1. That's it. Then the Driver will run at native 16u-steps interpolated to 256u-steps. The upscaling to 256u-steps is done inside the TMC2100, so no problem with running at 200mm/s for the MEGA.
[image]
The driver has to be configurated according to this table:
[image]
The third line from the bottom is what we want. The StealthChop modes are even more quite but useless as they have less torque.

Kaibob2, 2016-10-07:
It also would be nice if the diagnostic outputs (Driver error, index) could be integrated into Marlin, but this is another theme.
[image]
[Info](https://github.com/watterott/SilentStepStick/blob/master/hardware/SilentStepStick_v10.pdf)

Kaiwol1990, 2016-10-08:
Yo don#t have to solder that wire across the driver, the Jumper pin 1 (on ramps etc. at least) ist tied to ground. So if you leave the jumper open it will be pulled to ground.
The only thing you have to do is to complete remove one pin and dont put any jumper in.
For detailed instructions you can take a look at this instuctable page:
[http://www.instructables.com/id/Install-and-configure-SilentStepStick-in-RAMPS-TMC/](url)

Kaibob2, 2016-10-08:
@Kaiwol1990 I totally agree. This makes it even more easy for RAMPS owners :)
[image]

Kaibob2, 2016-10-21:
After a few weeks i'm still very happy with these drivers. They are absolutely worth the price. The printer became really quite which makes my wife quite happy :)

closed 2016-10-21 by Kaibob2

thinkyhead, 2016-10-22:
I'll be trying with these in the MakerArm next week.

Kaibob2, 2016-10-24:
@thinkyhead Do you think, it would be possible to code something for monitoring the error output of the TMC2100. I glued the heatsink with thermal conductive epoxy on top of the PCB, but they are still getting quite hot if i drive them at their maximum rated current. My heated bed assembly is quite rigid and this requires at least 1.5A for the Y-Stepper, better 1.8A. I would like to wire the X and Y TMC drivers error output to two pins of the AUX1 or AUX2 header on RAMPS 1.4. 
The error output is on PIN20 of the chip and is an open drain output. 
I had a few overheat situations but they only last a second or less. After this the print continues, but surely is ruined beacuse of missed steps. It would be nice if the print would halt on error.

thinkyhead, 2016-10-24:
@Kaibob2 I'll see about that. Hopefully I'll have a chance this week. I'm looking forward to seeing how the 2100 makes this machine sound. It's currently using 8x microstepping with 100:1 harmonic drives. Fortunately the drives will not be 100:1 in the next unit.

Kaibob2, 2016-10-24:
Nice, never heard of harmonic drives. Just viewed some videos of the principle and this drives seem very precise. But isn't this a little too much for 3D Printers? 
Anyway, i hope you will like them and put some effort in utilizing their full potential :)

thinkyhead, 2016-10-26:
@Kaibob2 Harmonic drives are useful in some instances. In this case, they're being used in a robust SCARA arm with a total length of 40cm, so the harmonic drives are needed to get the necessary torque and precision.

seabuck, 2017-06-17:
Hi, I am trying to use these drivers in my Prusa. I left the three jumpers on, and prepared the driver as @Kaibob2 however, the motors do not move, By the sound I can tell the steps per mm are not right.

I tried different values, and when I set 1 step per mm it more or less moved. Maybe the interpolation settings are not correct? All the solder pads in my board are left unsoldered. Should it be that way?

seabuck, 2017-06-17:
As I suspected, it should not be that way. If you receive a board with the default bridges not soldered, solder them as documented here: 
https://github.com/watterott/SilentStepStick/blob/master/hardware/SilentStepStick-TMC2100_v12.pdf