MarlinFirmware/Marlin #4968: Bang-bang is bad, but Marlin makes it worse...

AndyZuerich, 2016-10-07:
In the advanced configuration file, there is set a default option:

`#if DISABLED(PIDTEMPBED)
  #define BED_CHECK_INTERVAL 5000 // ms between checks in bang-bang control
  #if ENABLED(BED_LIMIT_SWITCHING)
    #define BED_HYSTERESIS 2 // Only disable heating if T>target+BED_HYSTERESIS and enable heating if T>target-BED_HYSTERESIS
  #endif
#endif`

Well, most Marlin users will probably never look into the advanced options. What does this `BED_CHECK_INTERVAL=5000` finally do? It limits the sampling resolution for the heated bed bang-bang controller to 5000 milliseconds. Ok. Well:
The natural bang-bang frequency would be (assuming a large heated bed time constant) four times the temperature delay/dead time. In my case, this is approximately 20 seconds. This dead time depends on the actual heated bed geometry (location of temp sensor w.r.t. heater, material properties, etc.). I have a 250mm round Aluminum bed with a 250 W silicon mat heater with integrated temperature sensor.

Now, when the default 5000 ms value of the bang-bang sampling rate is set, I get approximately double (!) of the temperature swing compared to a smaller (and probably more reasonable) setting of 500 msec.
The calculations in temperatures.cpp for the bang-bang controller are quite simple and fast. Therefore my suggestion: Please reduce the default value from 5000 msec to 500 msec.

Of course, bang-bang is not comparable with a real temperature controller. But I think, many people who don't know anything about PID just prefer bang-bang, since it is so simple to use: No tuning parameters and it has a fast heatup time.
At the moment, Marlin is not producing the best possible performance for bang-bang control in heated beds. 

Here is a measurement that I made on this issue:

[image]

P.S: In the future, I don't plan to use bang-bang for my heated bed. But the bang-bang mode can be used to do an automated extraction of controller parameters (a thermal model of the heated bed with dead time, time constant and gain). This will then be much simpler, more accurate and faster (also with faster control action) compared to PID.

Blue-Marlin, 2016-10-08:
The reason for that long, and with hysteresis even longer switching cycles, is to make machines using relays working reliable. Every switching wears them out. Switching every 1/2s instead of every 5s weares them out 10 times faster. One of the typical failures of a worn out relay is welding the contacts together switching the heater permanently on. Even a kill() will not switch off the heater. Only shutting down the power supply will help.
If you have a MOSFET-heater-switch use PID.
but please, PLEASE let the default `BED_CHECK_INTERVAL` as it is.

Blue-Marlin, 2016-10-08:
Enabling `PIDTEMPBED` by default could be the better choice.
The sound of a PIDd relay will alert you immediately.

Roxy-3D, 2016-10-08:
> **#Bang-bang is bad, but Marlin makes it worse...**

OK...   Marlin makes it worse.   How much worse does it do it compared to Repetier?   How much worse does it do it compared to Smoothie?   How much worse does it do compared to RepRap firmware?   When you have numbers, we can change the title back do what you posted.

KevMags, 2016-10-08:
I have almost identical equipment to OP.  It works GREAT on my mechanical relay equipped machines with steady temperatures, quick heating and surprisingly low relay switching.

AndyZuerich, 2016-10-08:
> OK... Marlin makes it worse. How much worse does it do it compared to Repetier? How much worse does it do it compared to Smoothie? How much worse does it do compared to RepRap firmware? When you have numbers, It might be fair and reasonable to make those kinds of allegations.

Thanks for changing the issue title. Maybe, I used too strong words with "marlin makes it worse". I did not want to compare Marlin with other comparable firwares, but with the bang-bang mode that would be natural and optimal. 
I don't have comparisons with Repetier, etc., since I am using only Marlin and nothing else. You are asking for measurements: Well, I showed a plot above. This should be enough for showing that the temperature swing is double than actually needed.

> The reason for that long, and with hysteresis even longer switching cycles, is to make machines using relays working reliable. Every switching wears them out. Switching every 1/2s instead of every 5s potentially weares them out 10 times faster. One of the typical failures of a worn out relay is welding the contacts together - switching the heater permanently on. Even a kill() will not switch off the heater. Only shutting down the power supply will help.
If you have a MOSFET-heater-switch use PID,
but please, PLEASE let the default BED_CHECK_INTERVAL as it is.

@Blue-Marlin 

Ok, I didn't know that many people use a mechanical relay for switching the bed. The factor 10 for the wearout time is probably not correct, since the physical properties of the bed/temperature sensor limit the switching speed, and not the measurement.  Im my case above, the bed switching frequency in bang-bang mode did not even double. 

> Enabling PIDTEMPBED by default could be the better choice.
> The sound of a PWMd relay will alert you immediately.

Yes, this is a good point, this should be somewhen in the future the default. Another thing that should be done is, that the bang-bang slowdown due to the BED_CHECK_INTERVAL variable should be shortly documented in configuration_adv.h, so that people know that this setting might (on purpose for mechanical relays) slow down the bang-bang mode.

From my side, this thread can be closed.

closed 2016-10-08 by Roxy-3D