MarlinFirmware/Marlin #4969: CoreXY: Z axis moving when homing X or Y

Brunez3BD, 2016-10-08:
Hi,
I have a coreXY printer and I want to use marlin instead of repetier firmware. But when I try homing X or Y axis from repetier host first the Z raise up and then the X or Y axis is moved to home position. I don't know why is doing that. 
This is my Configuration.h file:
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/517575/Configuration.zip)

Than you very much

Brunez3BD, 2016-10-08:
Update: I reverse the Z direction and now the Z goes down when homing one of the axis. If I home the same axis twice the Z doesn't go down, so I've understand that is a kind of protection to avoid collision. :)
I want to understand where is the line of code that do that to disable it, thank you

FHeilmann, 2016-10-11:
https://github.com/MarlinFirmware/Marlin/blob/RC/Marlin/Configuration.h#L592

https://github.com/MarlinFirmware/Marlin/blob/RC/Marlin/Configuration.h#L546

https://github.com/MarlinFirmware/Marlin/blob/RC/Marlin/Configuration.h#L547

Here are some candidates, depending on whether you have them enabled in your configuration or not.

Brunez3BD, 2016-11-11:
Thank you so much. I was thinking that the problem was #define Z_HOMING_HEIGHT 4 (on your first link).
The problem is surely connected to that line, because (as in the code) "Minimal z height before homing (G28) for Z". In fact if the Z heigh is 0 the printer move the Z down of 4 mm and then home the X or Y axis.
But this line in my configuration.h is commented, and that does not make sense... 
I don't know what to do, do you have any suggestion?
I will try to uncomment this line and set the value to 0.

FHeilmann, 2016-11-11:
Are you running RC7 or RCBugFix? There are tons Homing Related Fixes in RCBugFix over RC7

Roxy-3D, 2016-11-11:
Especially for the Core?? printers...   Please be running RCBugFix before you start a new issue about a homing problem!

thinkyhead, 2016-11-15:
The `DEBUG_LEVELING_FEATURE` option can help diagnose issues related to homing and leveling. As suggested, use `RCBugFix` for testing. `1.1.0-RC8` should be out soon. Just fixing up some Dual-X-Carriage issues.

closed 2017-10-31 by thinkyhead