MarlinFirmware/Marlin #4975: Como habilitar o carro duplo do eixo X junto com o Y?

RaphaelGPC, 2016-10-08:
Forgive my lack of knowledge. I'm trying to enable Dual X and Y Carriages but I have had problems. The the product does not work together. Or X works in Dual or Y. I would like to use the Marlin project in MCCNC.

bsh314, 2016-10-10:
You must use "Configuration_adv.h" file & "#define X_DUAL_STEPPER_DRIVERS" definition so fifth motor driver will be used for X axis. Same for the Y axis with "#define Y_DUAL_STEPPER_DRIVERS"

thinkyhead, 2016-10-22:
> Marlin only can drive one dual driver axis.

Actually, with the current Marlin 1.1.x code, it is possible to use dual steppers on multiple axes.

RaphaelGPC, 2016-10-22:
Hello. Grateful for the help. But when I enable the #define X_DUAL_STEPPER_DRIVERS and #define Y_DUAL_STEPPER_DRIVERS compilation error occurs.

I would like to use the x 2 and Y2 in the E0 E1.

In file included from sketch\MarlinConfig.h: 33:0,

from sketch\Marlin.h: 36,

from sketch\Marlin_main.cpp: 35:

pins. h:404: error: ' E2_STEP_PIN ' was not declared in this scope

#define __ EPIN (p, q) and # #p # #_ # #q # #_PIN

^

sketch\pins.h: 405:20: note: in expansion of macro ' __ ' EPIN

#define _ EPIN (p, q) __ EPIN (p, q)

^

sketch\pins.h: 424:27: note: in expansion of macro ' _ ' EPIN

#define Y2_STEP_PIN _ EPIN (Y2_E_INDEX, STEP)

And follow ....

thinkyhead, 2016-10-22:
Apparently you don't have enough stepper connectors to use both.

RaphaelGPC, 2016-10-22:
Thank you very much!!! Perfect for the y-axis. I do not intend to use extruder. 

But I have a problem with the X axis when the movement is positive the engines move in the same direction, but when motion is negative the engines move without reverse. The X 0 has negative movement (as requested) but the x 2 has the positive movement.

#define INVERT_X_DIR false
#define INVERT_Y_DIR false
#define INVERT_Z_DIR false

==============================
Configuration.h
==============================

#define INVERT_E0_DIR false
#define INVERT_E1_DIR false
#define INVERT_E2_DIR false
#define INVERT_E3_DIR false

=================================
 Configuration_adv.h
=================================
// Dual X Steppers
// Uncomment this option to drive two X axis motors.
// The next unused E driver will be assigned to the second X stepper.
#define X_DUAL_STEPPER_DRIVERS
#if ENABLED(X_DUAL_STEPPER_DRIVERS)
  // Set true if the two X motors need to rotate in opposite directions
  #define INVERT_X2_VS_X_DIR true
#endif

#define X2_STEP_PIN   E0_STEP_PIN
#define X2_DIR_PIN    E0_DIR_PIN
#define X2_ENABLE_PIN E0_ENABLE_PIN

// Dual Y Steppers
// Uncomment this option to drive two Y axis motors.
// The next unused E driver will be assigned to the second Y stepper.
#define Y_DUAL_STEPPER_DRIVERS
#if ENABLED(Y_DUAL_STEPPER_DRIVERS)
  // Set true if the two Y motors need to rotate in opposite directions
  #define INVERT_Y2_VS_Y_DIR true
#endif

#define Y2_STEP_PIN   E1_STEP_PIN
#define Y2_DIR_PIN    E1_DIR_PIN
#define Y2_ENABLE_PIN E1_ENABLE_PIN

Again thank you.

thinkyhead, 2016-10-24:
Check the wiring on that X2 motor.

RaphaelGPC, 2016-10-24:
I checked and even made rotation in engines, but for all the movement is the same. 

When I put the engine E0 equal to X, and turning to positive clockwise they move equal but when peopl return the E0 continues with the positive move and the X negative. 

When the engine is in the E0, reversing or not the connector moves unlike the X1 or Y1. I changed the code you sent me to, but still the engine that in E0 always inversely moves the X or Y primary. The E1 this perfect.

#define X2_STEP_PIN E1_STEP_PIN // E0_STEP_PIN
#define X2_DIR_PIN E1_DIR_PIN // E0_DIR_PIN
#define X2_ENABLE_PIN E1_ENABLE_PIN // E0_ENABLE_PIN

#define Y2_STEP_PIN E0_STEP_PIN
#define Y2_DIR_PIN E0_DIR_PIN
#define Y2_ENABLE_PIN E0_ENABLE_PIN

Sorry but my english is horrible. I'm brazilian.

thinkyhead, 2016-10-26:
> E0 always inversely

If it is always inverse, then just rotate the plug 180°.

RaphaelGPC, 2016-10-26:
Mesmo virando ou alterando o  (#define INVERT_X2_VS_X_DIR) para false ou true o E0 sempre se comporta como no video.

https://www.youtube.com/watch?v=8cq4O5brtxM

My setup this way. 

==============================
Configuration.h
==============================

#define INVERT_X_DIR false
#define INVERT_Y_DIR false
#define INVERT_Z_DIR false

#define INVERT_E0_DIR false
#define INVERT_E1_DIR false
#define INVERT_E2_DIR false
#define INVERT_E3_DIR false

=================================
 Configuration_adv.h
=================================

// Dual X Steppers
// Uncomment this option to drive two X axis motors.
// The next unused E driver will be assigned to the second X stepper.
#define X_DUAL_STEPPER_DRIVERS
#if ENABLED(X_DUAL_STEPPER_DRIVERS)
  // Set true if the two X motors need to rotate in opposite directions
  #define INVERT_X2_VS_X_DIR true
#endif

#define X2_STEP_PIN   E0_STEP_PIN
#define X2_DIR_PIN    E0_DIR_PIN
#define X2_ENABLE_PIN E0_ENABLE_PIN

// Dual Y Steppers
// Uncomment this option to drive two Y axis motors.
// The next unused E driver will be assigned to the second Y stepper.
#define Y_DUAL_STEPPER_DRIVERS
#if ENABLED(Y_DUAL_STEPPER_DRIVERS)
  // Set true if the two Y motors need to rotate in opposite directions
  #define INVERT_Y2_VS_Y_DIR true
#endif

#define Y2_STEP_PIN   E1_STEP_PIN
#define Y2_DIR_PIN    E1_DIR_PIN
#define Y2_ENABLE_PIN E1_ENABLE_PIN

Thank you very much and sorry to disturb you.

RaphaelGPC, 2016-10-26:
Hello. Can you tell me if there is a problem in the modification below:

=
Configuration_adv.h
=

#define X_DUAL_STEPPER_DRIVERS
#if ENABLED(X_DUAL_STEPPER_DRIVERS)
  // Set true if the two X motors need to rotate in opposite directions
  #define INVERT_X2_VS_X_DIR true
#endif

#define X2_STEP_PIN   E2_STEP_PIN     //E0_STEP_PIN
#define X2_DIR_PIN    E2_DIR_PIN      //E0_DIR_PIN
#define X2_ENABLE_PIN E2_ENABLE_PIN   //E0_ENABLE_PIN

=
pins_RAMPS.h
=

#define Z_STEP_PIN ___              26//46
#define Z_DIR_PIN ___              28//48
#define Z_ENABLE_PIN ___          24//62

#define E0_STEP_PIN  ___            26
#define E0_DIR_PIN  ___            28
#define E0_ENABLE_PIN ___       24

#define E1_STEP_PIN ___         36
#define E1_DIR_PIN ___          34
#define E1_ENABLE_PIN ___       30

#define E2_STEP_PIN  ___        46 //26
#define E2_DIR_PIN  ___         48 //28
#define E2_ENABLE_PIN ___       62 //24

- Is there a PIN null for I set as E0.
- With this modification the x-axis moves perfectly
- Not yet tested the END-STOP, but will they work perfectly with the Z in the slot of the E0 and the x 2 in the slot of the Z?

rateful once again for the help!!!!!!

thinkyhead, 2016-10-26:
- If you're simply not using E0, then you can leave the pins as they are.
- I'm glad to hear X axis moves perfectly. Probably using the E0 pins messed it up and isn't really supported yet. Maybe sometime I will make it possible to set `EXTRUDERS` to 0.
- Endstops will work fine.

RaphaelGPC, 2016-10-26:
I would like to have more knowledge of firmware. There is some code that control the reversal of E0. For me it seems that even the E0 not being used the code always keeps it active.

thinkyhead, 2016-10-26:
@RaphaelGPC Correct. There is no way currently to use `EXTRUDERS` 0 because Marlin was made originally for 3D printers, all of which have at least one extruder.

thinkyhead, 2017-10-31:
Dual steppers and dual endstops are now supported for X, Y, and Z in the latest code, which will be released as 1.1.7 soon.

closed 2017-10-31 by thinkyhead