MarlinFirmware/Marlin #4976: UBL - On the Fly Mesh Fine tune

adamfilip, 2016-10-09:
@Roxy-3D I had a thought while using with Z Babystepping on my Prusa i3

1) If there was an option to do on the fly Mesh fine tuning that could be awesome. 
so after ive created a mesh automatically. and I started to do a print. if i noticed that a section of the mesh was off slightly.. I would be able to rotate the encoder dial and bump the area currently printing up or down. and then it would make the adjustment to the nearest grid point.. where the adjustment had started from.  Similar to Z babystepping. but instead of moving the entire mesh it just at that one spot. once adjustment was made the mesh would be updated immediately

2) with the G26 Validation tool. Can you add a variable for extrusion multiplier.. I tend to set my slicer to 0.9 to get the extrusion right, and currently with G26 I find it over extrudes

BTW I should have my Corexy machine back up and running in a day or so. to do more testing.

Roxy-3D, 2016-10-09:
I think I found the reason for the z-offset issue.   I'm not sure what to do about it.  I'll probably think about it a little more and start a thread for discussion.

> 1) If there was an option to do on the fly Mesh fine tuning that could be awesome. 
so after ive created a mesh automatically. and I started to do a print. if i noticed that a section of the mesh was off slightly.. I would be able to rotate the encoder dial and bump the area currently printing up or down. and then it would make the adjustment to the nearest grid point.. where the adjustment had started from. Similar to Z babystepping. but instead of moving the entire mesh it just at that one spot. once adjustment was made the mesh would be updated immediately

I'm in the process of making it so you can move the whole Mesh up or down.   Certainly what you suggest could be done.  There would be two questions:   1)  How is the area (or size) of what is being adjusted controlled?   Would it always be the same size?  Perhaps it could be what ever mesh cells have been crossed in the last 10 seconds?   Or maybe it is just where ever the nozzle is at the moment the encoder wheel spins.   That would let the person stop adjusting if the current area is OK, but when the nozzle moves out of that area and needs further adjustment the process could continue.   The other question would be 2) How to quickly get into that mode.   I do have a quick way in to the entire Mesh push/pull with the kill switch and soon the Press and Hold.   Maybe those two methods can be assigned however the user wants?

> 2) with the G26 Validation tool. Can you add a variable for extrusion multiplier.. I tend to set my slicer to 0.9 to get the extrusion right, and currently with G26 I find it over extrudes

OK.  I just made the changes...   But I can't check it out until I get some other changes I'm in the middle of doing worked out.  (The code doesn't compile right now.)

> BTW I should have my Corexy machine back up and running in a day or so. to do more testing.

That timing might work out well.

adamfilip, 2016-10-09:
>  Or maybe it is just where ever the nozzle is at the moment the encoder wheel spins. That would let the person stop adjusting if the current area is OK, but when the nozzle moves out of that area and needs further adjustment the process could continue. 

Yes this makes sense to me.

The press and hold, could bring up a quick menu, which gives you a few common use items. so its a two click operation, during printing.

Press/Hold -> Z babystepp entire mesh
Press/Hold --> Mesh On the Fly Tune
Press/Hold --> Prime Extruder 10mm
Press/Hold --> Activate UBL
Press/Hold --> Load UBL Mesh # 1
Press/Hold --> Load UBL Mesh # 2
Press/Hold --> Load UBL Mesh # 3

perhaps too many options would defeat the purpose.

Roxy-3D, 2016-10-09:
Yeah...   Let's think about this for a little while.   I do see how it can be very useful.   The trick is to make it super simple (and obvious) to use.   And really...  the code to do this shouldn't be very difficult.  It is more a question of how to get it to fit into the whole system nicely and intuitively.

KevMags, 2016-10-09:
I really like the "press and hold" to get to the Babystepping menu item.  While being in a hurry to get to Babystepping I have inadvertently selected Change Filament.

closed 2017-10-31 by thinkyhead