MarlinFirmware/Marlin #4977: In twibus.cpp the timeout after Wire.requestFrom() is not needed.

Koepel, 2016-10-09:
In twibus.cpp in reqbytes() is a timeout with millis(). That that timeout is not needed.
The Wire.requestFrom() function waits until the I2C transmission has completely finished. After that, all the received bytes are in a buffer in the Wire library. The Wire.available() returns the number of bytes in that buffer.
A timeout after a call to Wire.requestFrom() has no purpose.

In case there was a I2C bus error or a collision, the Wire.requestFrom() and the Wire.available() return a lower value than the requested bytes (probably the number zero). The timeout with no purpose does detect this. Therefor a single check with Wire.available() could be used to check for a I2C bus error.

thinkyhead, 2016-10-12:
You must be referring to older code. This was changed a looong time ago in `RCBugFix`.

closed 2016-10-12 by thinkyhead

Koepel, 2016-10-13:
Sorry, I didn't know that.

In the RCBugFix branch, there is this : Wire.requestFrom ( this -> addr , bytes ) == 0
That might not always work, it is better to test if the return value is equal to 'bytes'. There might be (maybe in the future) a way that when a collision is detected, the number of received bytes is more than zero, but still not 'bytes'.

thinkyhead, 2016-10-22:
> That might not always work, it is better to test if the return value is not equal to 'bytes'.

Actually, I followed the advice from this posting: http://gammon.com.au/forum/?id=10896#reply10

> Wire.requestFrom() either returns zero (if the slave could not be connected to) or the number of bytes requested, regardless of how many bytes the slave actually sent.
>
>The reason is that, for the slave to terminate the communication early (ie. after sending less than the requested number of bytes) it would have to be able to raise a "stop condition". Only the master can do that, as the master controls the I2C clock.
>
>Thus, if the slave stops sending before the requested number of bytes have been sent, the pull-up resistors pull SDA high, and the master simply receives one or more 0xFF bytes as the response.
>
>This has been raised on GitHub:
>
>https://github.com/arduino/Arduino/issues/2616
>
>It does not help to use Wire.available() because that too will return the number of bytes requested.

Koepel, 2016-10-22:
Yes, that's how it is.

That is why I did not mention about the slave, but about a collision (in a multi-master bus). Also from a software point of view, a test if the received bytes is equal to the number of requested bytes seems better. You only test for that one single value of zero. Why would one exclude all other wrong values ?

thinkyhead, 2016-10-22:
The zero result is a confirmed error. The only other result possible is the original request length, and it doesn't confirm success. If the result returned is not zero, then the connection was at least successful. Once you have a successful request, there's no way, apart from building a better protocol, to confirm an error. So it is left to the coder to do any error-checking on the data *after* receiving the data (i.e., without a connect error). It doesn't need to be handled at this level.