MarlinFirmware/Marlin #4978: rabbit core RCM (2000/3000/4000) support request

darknessblade, 2016-10-09:
hello i would like to ask if its possible to run/ execute this firware on the rabbit core RCM 2000/3000/4000 series
---
the rabbitcore is a AVR like the amtel/arduino series.
---
i have a few of those processor boards and would like to build a working 3d printer with it.
since i have zero to no experience coding firmware, i would like to ask for some help with it.

the board i mainly will use: rabbit core RCM 3000 prototype board
rabbit core RCM 3000 prototype board with master/slave setup (can also be changed to master setup only)
image for refference:
[http://www.acierta-it.com/jas/v3.0/CD CON TODO/DCRABBIT_9.21/Docs/manuals/RCM3100/UsersManual/images/figb-7.gif](url)
---
i can test all code myself, and might be able to try some porting, to see what works and what does not.
---
types of code both boards uses:
amtel/arduino  uses C++
rabbitcore uses dynamic C

thinkyhead, 2016-10-09:
Marlin can run on any board using the Arduino 8-bit AVR CPU family. If the Arduino IDE can flash it, then it can probably flash Marlin onto it. Pins are assigned to functions directly. See the `pins_RAMPS.h` file for an example of the most common pins layout.

bobc, 2016-12-06:
> the rabbitcore is a AVR like the amtel/arduino series.

Really? "citation needed!" I am pretty sure Rabbitcore is nothing like AVR, apart from being 8-bit. 

I think it would be too much effort to support this CPU for such a niche application. If you have no experience with coding, I would strongly suggest starting with a board that is currently supported by Marlin, or other printer firmware, if you want to get a printer going.

darknessblade, 2016-12-06:
BOBC,

i am aware that it is a different kind of AVR,
i do know its hard to program, have tried to make a LED flash/blink,  but this has not yet succeeded.

bobc, 2016-12-06:
AVR is a specific chip architecture proprietary to Atmel (now Microchip), "AVR" is not a generic term for 8 bit microcontroller. 

Learning to code is commendable, but I can't think why anyone would start with such an obscure chip.

Roxy-3D, 2016-12-06:
@darknessblade   For $6.50 (including shipping) you can get a processor board that plugs right into your computer and that Arduino talks to directly.   This is the most prevalent processor board that Marlin runs on.    The people using this board typically plug a RAMPS v1.4 board into it.

http://www.ebay.com/itm/New-Atmega2560-16AU-CH340G-ATMEGA-2560-R3-Board-Mega2560-R3-Arduino-Compatible-/401036224315?hash=item5d5f9f273b:g:XVUAAOSw~gRVt1Lg

You can plug an led into one of the pins (and ground the other side) to just get started.   But then for $1.59 (including shipping!) you can get an 8x8 LED Matrix for your program to tun lights on and off:

http://www.ebay.com/itm/MAX7219-Dot-led-matrix-module-MCU-control-LED-Display-module-wire-for-Arduino-/191781645249?hash=item2ca713abc1:g:3AgAAOSwo6lWQtph

You will make progress faster if you use well supported hardware!

closed 2017-10-31 by thinkyhead