MarlinFirmware/Marlin #4979: what is the z probe for ?

tyssoon, 2016-10-09:
Hi,

I've already posted issues on marlin on a DIY delta few weeks ago, i dropped it for a while because of motivation and configuration complexity, i begin to remotivate myself to make my delta working again but i would like to be sure of one thing. Here is my question:

In used to thought that my omron switch which work as my Z probe was supposed to count the height from the endstops switches on the top of the delta. Is that true ?

Because now im not sure of it anymore and i think i understood through discussions here that tje zprobe only check the bed irregularities (like not parallel bed with endstops plane, or not plane bed)

So, what is the truth please ? can i use the zprobe as a Z height measurer (from the endstops) ? am i clear in my question? cause im not a native english spaker.

Thank you by advance guys, keep the good work going.

Roxy-3D, 2016-10-09:
The history of the Z-Probe was that it allowed the nozzle to move around and probe how far a given point was off the bed.    That let the firmware calculate how much to tilt the coordinate system so it could print normal to the bed.   The whole reason for the Z-Probe is the print bed is not always perfectly normal (and flat) to the Z-Axis.

tyssoon, 2016-10-09:
so Z probe doesnt measure the dostance endstops/bed.

Roxy-3D, 2016-10-09:
The Z-Probe (if you have one) is typically used to measure (or find) the bed.   But a Z-End Stop can be used in combination with a Z-Probe.   You can set this up however makes sense for you.

tyssoon, 2016-10-09:
can the Z endstop and Z probe be the same physical switch ?

Blue-Marlin, 2016-10-09:
Not on a DELTA! A delta needs max endstops to home. A probe can only replace a z-min-endstop.

Airheadbit, 2016-10-10:
I found out the hard way that an inductive probe used in place of the Z end stop is not a good thing, when the table moves out from under the head the inductive sensor (LJ18A3-8-Z/BX ) has nothing to sense and the Z axis crashes. Now my probe is wired in parallel with the Z axis end stop. Next try is a laser to measure bed height.

tyssoon, 2016-10-10:
> Not on a DELTA! A delta needs max endstops to home. A probe can only replace a z-min-endstop.

sorry, i meant: can the Z _min_ endstop and Z probe be the same physical switch ? i use omron switch not inductive.

Blue-Marlin, 2016-10-10:
@Airheadbit 
That is not only a problem of a inductive probe. It's a problem with probes having a x/y-offset. Means - with the laser probe it's the same.

@tyssoon 
As always - it depends.
With a probe/endstop-switch having a negative z-offsep (trigger point below the nozzle) you can not print the first layer (z=0) because the ENDTOP function prevents the nozzle from going there. Usually this is avoided by a deploy/stow mechanism (servo). The switch is only active (in the right position) when needed for HOMING (G28) or PROBING (G29,  ...). That means, it can not work as a ENDSTOP in the sense of a security device because it's not deployed.
For inductive/capacitive/IR/laser/... -probes phisically located above the nozzle, their trigger point hast to be still below the nozzle to make them work. They all have a negative z-offset. So logically its the same as with a mechanical switch. They have to be deployed/stowed , what here is not a mechanical move, but a logical switching on the test.
Only with probes having a positive or zero z-offset (>= 0) (FSR, tilt, switches below the bed) all 3 functions (probing, homing and security) can be realized at the same time with the same switch.

Alternative you can use two switches, connected to the same pin. One acting as the probe and the other acting as the security endstop  For a normal 3-axis machine this is not a big thing - for a delta machine you will again have the 3-swich combo below the bed, what is physically the same as a probe with positive z-offset.

tyssoon, 2016-10-10:
okay thanks, so the nozzle switch is the answer for me  guess, i'll try to find one.

thinkyhead, 2016-10-22:
Hopefully in the near future the leveling functions will prevent probing outside the bed automatically. But for now, just be sure to set the probe boundaries with caution.

closed 2017-10-31 by thinkyhead