MarlinFirmware/Marlin #4986: Inverted control with 5 button interface

jaysonkelly, 2016-10-10:
I am using the RIGIDBOT_PANEL which has directional buttons instead of a push button rotary encoder. Navigating the menus is fine, but when it comes to adjusting an onscreen value or jogging an axis... the actions are inverse from the button placement. I thought I found the solution in either the REVERSE_ENCODER_DIRECTION or REVERSE_MENU_DIRECTION options but they don't seem to have any effect on the d-pad style control. Can those options be made to affect the 5 button style control as well?

Roxy-3D, 2016-10-10:
> Can those options be made to affect the 5 button style control as well?

We really should do that.  But it is probably more important to just get RC-7 / RC-8 stabilized and do this later in a few months.   From my perspective, this is a really depressing question.  I've been working to get a "Press & Hold" status recognized by the LCD Panels, and I just realized, I need to add extra code to handle the LCD Panels that don't have an encoder wheel.

Is there one of the buttons that is hardly ever used for anything that I could watch for the "Press and Hold" function to activate?   Does that even make sense on your LCD Panel?

jaysonkelly, 2016-10-10:
Well, we have 5 buttons... The center button does the same thing as an encoder push. Are you planning on using the encoder push for "Press and Hold"? If that is the case we could simply use the same functionality. 

Here's what the original Rigidbot Controller looks like:
[image]

On the original firmware we had a shortcut menu when we click the right button while on the status screen and that is really nice. Is that the kind of functionality you want to get from the Press and Hold?

Sounds like I need to do some digging into how Marlin manages the button inputs. I thought it was mapping them to a number of encoder pulses in either direction depending on which button, but since the invert options don't seem to change anything I'm left wondering.

Roxy-3D, 2016-10-10:
> On the original firmware we had a shortcut menu when we click the right button while on the status screen and that is really nice. Is that the kind of functionality you want to get from the Press and Hold?

I've implemented a shortcut to get to Z-BabyStepping by remapping the 'Kill' button.   You press that, and instantly you are doing Z-BabyStepping.  But that is not going to be acceptable for a lot of people.   So, right now, I'm trying to make a "Press and Hold" (for about 750 ms) do the same thing.

The reason is if you need to do Z-BabyStepping at the start of the print, you don't want to spend 30 seconds trying to drill down and navigate the menu system.  And besides that, if you are in a hurry, it is likely to make a mistake.

My guess is the center button behaves the same as the Encoder Wheel 'Select' does.  I can key off of that being pressed and held.

jaysonkelly, 2016-10-10:
That's cool,  but what about reversing the buttons so they do the intuitive
action, ie. Up button increases values or makes z go up or Y go more
positive? Aren't there other controllers with the 5 button interface which
are supported by Marlin that don't have the inversion issue?

Roxy-3D, 2016-10-10:
> That's cool,  but what about reversing the buttons so they do the intuitive
action, ie. Up button increases values or makes z go up or Y go more
positive? 

I don't know the answer.  But it very possibly makes sense for you to just edit the pins_RIGIDBOARD.h file for your LCD Panel:
```cpp
  #define BTN_UP           37     // change to 35 ???
  #define BTN_DWN       35    //  change to 37 ???
  #define BTN_LFT          33    //  change to 32 ???
  #define BTN_RT           32    //  change to 33 ???
```

jaysonkelly, 2016-10-10:
That could work, but it would also reverse the menu direction which I still
couldn't fix with the reverse menu option mentioned earlier. :)

thinkyhead, 2016-10-22:
> Sounds like I need to do some digging into how Marlin manages the button inputs.

As buttons, but for directional buttons it passes them on as encoder movement.

> That could work, but it would also reverse the menu direction

In that case you could then use `REVERSE_MENU_DIRECTION` to put it back the other way.

jaysonkelly, 2016-10-22:
I believe I already tried that. I enabled REVERSE_MENU_DIRECTION and REVERSE_ENCODER_DIRECTION hoping that would keep the menu direction the same but reverse the   edit command and move axis commands

jaysonkelly, 2016-10-22:
Just tried both reverse settings on their own and there was no effect on my menu navigation.

jaysonkelly, 2016-10-22:
#if LCD_HAS_DIRECTIONAL_BUTTONS

          if (false) {
            // for the else-ifs below
          }
          #if BUTTON_EXISTS(UP)
            else if (BUTTON_PRESSED(UP)) {
              encoderDiff = -(ENCODER_STEPS_PER_MENU_ITEM);
              next_button_update_ms = now + 300;
            }
          #endif
          #if BUTTON_EXISTS(DWN)
            else if (BUTTON_PRESSED(DWN)) {
              encoderDiff = ENCODER_STEPS_PER_MENU_ITEM;
              next_button_update_ms = now + 300;
            }
          #endif
          #if BUTTON_EXISTS(LFT)
            else if (BUTTON_PRESSED(LFT)) {
              encoderDiff = -(ENCODER_PULSES_PER_STEP);
              next_button_update_ms = now + 300;
            }
          #endif
          #if BUTTON_EXISTS(RT)
            else if (BUTTON_PRESSED(RT)) {
              encoderDiff = ENCODER_PULSES_PER_STEP;
              next_button_update_ms = now + 300;
            }
          #endif

        #endif // LCD_HAS_DIRECTIONAL_BUTTONS

This section of code stating what to do when a directional button is pressed doesn't seem to be affected by the reverse menu or reverse encoder options

jaysonkelly, 2016-10-22:
// Manage encoder rotation
    #if ENABLED(REVERSE_MENU_DIRECTION) && ENABLED(REVERSE_ENCODER_DIRECTION)
      #define ENCODER_DIFF_CW  (encoderDiff -= encoderDirection)
      #define ENCODER_DIFF_CCW (encoderDiff += encoderDirection)
    #elif ENABLED(REVERSE_MENU_DIRECTION)
      #define ENCODER_DIFF_CW  (encoderDiff += encoderDirection)
      #define ENCODER_DIFF_CCW (encoderDiff -= encoderDirection)
    #elif ENABLED(REVERSE_ENCODER_DIRECTION)
      #define ENCODER_DIFF_CW  (encoderDiff--)
      #define ENCODER_DIFF_CCW (encoderDiff++)
    #else
      #define ENCODER_DIFF_CW  (encoderDiff++)
      #define ENCODER_DIFF_CCW (encoderDiff--)
    #endif
    #define ENCODER_SPIN(_E1, _E2) switch (lastEncoderBits) { case _E1: ENCODER_DIFF_CW; break; case _E2: ENCODER_DIFF_CCW; }

    uint8_t enc = 0;
    if (buttons & EN_A) enc |= B01;
    if (buttons & EN_B) enc |= B10;
    if (enc != lastEncoderBits) {
      switch (enc) {
        case encrot0: ENCODER_SPIN(encrot3, encrot1); break;
        case encrot1: ENCODER_SPIN(encrot0, encrot2); break;
        case encrot2: ENCODER_SPIN(encrot1, encrot3); break;
        case encrot3: ENCODER_SPIN(encrot2, encrot0); break;
      }
    }
    lastEncoderBits = enc;

This section just below the last section I commented seems to only affect the encoder input movement. Seems like since the directional buttons don't modify "newbutton" that "buttons" will never change and all of the above directional modification code will never execute.

jaysonkelly, 2016-10-22:
How about something like this:
        #if LCD_HAS_DIRECTIONAL_BUTTONS

          if (false) {
            // for the else-ifs below
          }
          #if BUTTON_EXISTS(UP)
            else if (BUTTON_PRESSED(UP)) {
              #if ENABLED(REVERSE_MENU_DIRECTION) && ENABLED(REVERSE_ENCODER_DIRECTION)
                encoderDiff = (ENCODER_STEPS_PER_MENU_ITEM * encoderDirection);
              #elif ENABLED(REVERSE_MENU_DIRECTION)
                encoderDiff = -(ENCODER_STEPS_PER_MENU_ITEM * encoderDirection);
              #elif ENABLED(REVERSE_ENCODER_DIRECTION)
                encoderDiff = (ENCODER_STEPS_PER_MENU_ITEM);
              #else
                encoderDiff = -(ENCODER_STEPS_PER_MENU_ITEM);
              #endif
              next_button_update_ms = now + 300;
            }
          #endif
          #if BUTTON_EXISTS(DWN)
            else if (BUTTON_PRESSED(DWN)) {
              #if ENABLED(REVERSE_MENU_DIRECTION) && ENABLED(REVERSE_ENCODER_DIRECTION)
                encoderDiff = -(ENCODER_STEPS_PER_MENU_ITEM * encoderDirection);
              #elif ENABLED(REVERSE_MENU_DIRECTION)
                encoderDiff = ENCODER_STEPS_PER_MENU_ITEM * encoderDirection;
              #elif ENABLED(REVERSE_ENCODER_DIRECTION)
                encoderDiff = -(ENCODER_STEPS_PER_MENU_ITEM);
              #else
                encoderDiff = ENCODER_STEPS_PER_MENU_ITEM;
              #endif
              next_button_update_ms = now + 300;
            }
          #endif
          #if BUTTON_EXISTS(LFT)
            else if (BUTTON_PRESSED(LFT)) {
              #if ENABLED(REVERSE_MENU_DIRECTION) && ENABLED(REVERSE_ENCODER_DIRECTION)
                encoderDiff = -(ENCODER_PULSES_PER_STEP * encoderDirection);
              #elif ENABLED(REVERSE_MENU_DIRECTION)
                encoderDiff = ENCODER_PULSES_PER_STEP * encoderDirection;
              #elif ENABLED(REVERSE_ENCODER_DIRECTION)
                encoderDiff = -(ENCODER_PULSES_PER_STEP);
              #else
                encoderDiff = ENCODER_PULSES_PER_STEP;
              #endif
              next_button_update_ms = now + 300;
            }
          #endif
          #if BUTTON_EXISTS(RT)
            else if (BUTTON_PRESSED(RT)) {
              #if ENABLED(REVERSE_MENU_DIRECTION) && ENABLED(REVERSE_ENCODER_DIRECTION)
                encoderDiff = (ENCODER_PULSES_PER_STEP * encoderDirection);
              #elif ENABLED(REVERSE_MENU_DIRECTION)
                encoderDiff = -(ENCODER_PULSES_PER_STEP * encoderDirection);
              #elif ENABLED(REVERSE_ENCODER_DIRECTION)
                encoderDiff = (ENCODER_PULSES_PER_STEP);
              #else
                encoderDiff = -(ENCODER_PULSES_PER_STEP);
              #endif
              next_button_update_ms = now + 300;
            }
          #endif

        #endif // LCD_HAS_DIRECTIONAL_BUTTONS

jaysonkelly, 2016-10-22:
Tested and it works perfectly. I'll update the ultralcd.cpp in my branch and create a pull request.

closed 2017-03-12 by jaysonkelly