MarlinFirmware/Marlin #4987: changed Temparture Table for PT100?

DeathmannX, 2016-10-10:
Hi, i bought a few moth ago a e3d v6 with a Amplifyer Board for my Rambs 1.4 and it auto turned off if i turn my Prusa i3 on.
now i looked at http://wiki.e3d-online.com/wiki/E3D_PT100_Amplifier_Documentation and compared the values with the values at thermistortables.h Line 704-752 and they didn't match.

So are the values wrong or did i something wrong?

Blue-Marlin, 2016-10-10:
What you did wrong isn't clear to me, but the tables are identical.

The units are different. E3D in Volt, Marlin in 0-1023 ADC-readout.
MarlinValue = round(1024/5*E3D-Value)

Blue-Marlin, 2016-10-10:
Right polarity of the thermocoupler (PT100)?

DeathmannX, 2016-10-13:
I checked everything with my Multimeter but nothing.
I remove the Amplifyer Board with the PT100 and become the same Error...

So something sould be wrong with the Firmware i think.
I checked 1000x times the configurtion files and found nothing.

Grogyan, 2016-10-13:
What is the value of thePT100 with a multimeter at room temperature?
A PT100 isn't a thermocouple but a high temperature thermo-resistor

thinkyhead, 2016-10-22:
Any progress or other insights?

crncrussell, 2016-10-26:
For what it's worth, I've just upgraded to RC7 bug fix tonight, and I am seeing 10-15c temperature swings on my PT100 that was stable on RC3.  I have a second extruder th a type 11  thermistor and setup to do independent PID.  Have not had time to troubleshoot further, but wanted to make note that I am seeing something wonky going on too.

thinkyhead, 2016-10-26:
Do `M502` followed by `M500` and then do a new `M303` to re-calibrate your hot end.

crncrussell, 2016-10-26:
Thinkyhead- I have restored defaults and done a PID tune on all 3 sensors.  While I am getting fluctuations on all 3, the PT100 is the most extreme.  Made a quick video of it here:
https://youtu.be/lXE0icqtL6M

crncrussell, 2016-10-26:
Well, something to add in my scenario is the I switched over at the same time from a RAMPS to a SainSmart 2-in-1 board, but did not initially think that board was the problem.  I rolled back to my last RC3 build on this board and am seeing the same extreme shifts.  Another user I know is also getting unstable temps with this board.

So at this point, I am not thinking this is an RC7 issue, but noisy ADC circuitry on the Sainsmart board that is causing the problems.

Roxy-3D, 2016-10-26:
Its unfortunate your board is bad...  But everybody here is glad RC7 is OK!    :)

DeathmannX, 2016-11-14:
Is there anybody else who got no Problems with the PT 100 e3D V6 and the Ramps Board?

DeathmannX, 2016-11-15:
This is my Marlin: http://www.file-upload.net/download-12094726/Marlin-1.1.0-RC7.rar.html

thinkyhead, 2016-11-16:
Have you asked around at http://forums.reprap.org/ for help?
I'd be interested to know about the experience of other E3D V6 + RAMPS users.

DeathmannX, 2016-11-20:
Did that: http://forums.reprap.org/read.php?219,724059,724387#msg-724387

thinkyhead, 2016-11-21:
I don't know if we can conclude that the thermistor table is wrong from a sample size of one. As far as we know it works for most cases. The documentation at the E3D site indicates that the PT100 board needs a steady 5V power supply to give the documented readings. Perhaps there's too much resistance in the wiring, or some other contributing factor.

Grogyan, 2016-11-21:
My 2cents
I know from experience that RAMPS seems to be ok with just powering the
immediate circuitry from the onboard 5V 1A regulator.
Along with a thermocouple.  However BL Touch can draw a significant amount
of power. While I understand this thread doesn't cover that probe, i'm
suggesting that perhaps there is something else drawing too much current.
Or
Perhaps the sensor control board is receiving a good amount of interference
from a stepper motor

DeathmannX, 2016-11-21:
I don't said that the Thermistortable is wrong.
I tryed to say this is wrong:

`#define HEATER_0_RAW_HI_TEMP 16383`

Blue-Marlin, 2016-11-21:
> In my Case from 2. = round(1023/5*120 ohm) = 24552

Please recalculate! I get 4092.

16383 is (1024 * 16) - 1. Indeed this looks as if its to big. (1024 -1) * 16 = 16368 would be the correct value.
But this does not matter because this is only the start value. The real max/min values are later calculated in 
```
  #define TEMP_MAX_ROUTINE(NR) \
    maxttemp[NR] = HEATER_ ## NR ## _MAXTEMP; \
    while(analog2temp(maxttemp_raw[NR], NR) > HEATER_ ## NR ## _MAXTEMP) { \
      if (HEATER_ ## NR ## _RAW_LO_TEMP < HEATER_ ## NR ## _RAW_HI_TEMP) \
        maxttemp_raw[NR] -= OVERSAMPLENR; \
      else \
        maxttemp_raw[NR] += OVERSAMPLENR; \
    }
```
so what matters most is if HEATER_0_RAW_HI_TEMP is bigger than HEATER_0_RAW_LO_TEMP or not.
A brief look on table 20 saies: yes, high temperatures have high ADC-readouts.

DeathmannX, 2016-11-21:
https://lmgtfy.com/?q=1023%2F5*120
Google say i'am right. ;)

I orderd a new PT100 maybe my got a issue.
In my Room there is a Temparture around 21°C constant so i should get out ~108 Ohm and not 120...

Blue-Marlin, 2016-11-22:
Ups. Obviously i have not had enough coffee after getting up.
Don't know what i calculated there. :-(
The other conclusions seem to be right. :-)

What is your current problem?
No measurement at all?
Temperature off by a constant?
Temperature off by a factor?
Temperature fluctuating a lot? (noise)
Permanent min or max errors?
Occasional min or max errors?

DeathmannX, 2016-11-22:
No Problem ;D

My Problem is that trigger warning. 
The Pt100 gives me a output, but it is not right (should be lower) 
Maybe I crackt it with the screw from the Hotend or it it got a issue when it comes from e3d I don't know.

tampas, 2016-11-23:
Hi, any advance on this problem?

I'm trying to use an E3D-PT100 with a X3_PRO board and it give-me MAXTEMP trigger error.

I'm using #define TEMP_SENSOR_0 20 and #define TEMP_0_PIN 9 (A9 is one of the analogue inputs that i'm not using).

Alternatively , i've tried to hook up the PT100 to the normal thermistor port (#define TEMP_0_PIN 13) and used TEMP_SENSOR_0 147 (PT100 with 4k7 pullup). But in this case i've got erratically behavior, like getting negative temperature readings e a huge flotation of the values.

Any advice on this?
Thanks in advance,

thinkyhead, 2016-11-26:
@tampas What does the E3D documentation wiki say about how to connect the PT100?

thinkyhead, 2016-11-26:
@DeathmannX @Blue-Marlin Did we decide that the `HEATER_0_RAW_HI_TEMP` value for temp sensor 20 needs to be changed in `thermistortables.h` to `1023 / 5 * 120`?

Blue-Marlin, 2016-11-26:
> round(1023/5*120 ohm) = 24552

Is complete rubbish
You can not use a Ohm value here.
The e3d table maps temperatures to voltages not resistances.

thinkyhead, 2016-11-26:
> The e3d table maps temperatures to voltages not resistances.

Ah-so!

closed 2017-10-31 by thinkyhead

tphan26363, 2019-09-01:
Why do need to you recalculate the that number?  The (1024-1) *16) = 16383 basically mean that the software decalre the limit is 16 bit resolution and that is the ADC limit.  How ever most chip are around 12bit resolution for the ADC  So there plenty of room.  Change that number to a smaller one will change nothing. The input  depends on the CPU, old 8 bit use 5V pull up and newer 32 bit chip may use 3.3 volt pull up.  So there is 5V/3.3 = 1.51515.  Use that number to recalculate the whole table.  The table values are ADC values vs temperature values..  Multiply 1.51515 to the ADC value will get you to a close enough to the real temp.  Remove the pull up resistor since it can also screw up the reading.  Since the PT100 amp you have may not really quality one you may not get accurate temperature.  You can get more accurate if you can measure the 5VCC and 3.3Vcc to the exact number on the board