MarlinFirmware/Marlin #4988: Rip-off transfer to SD card

bsh314, 2016-10-10:
**## This is not an issue, this is a simple question.**
I was working on a little project and noticed that this functionality is exactly what I need. So after reading 5-6 repo files I still have no idea how it's works. Marlin devs made awesome work with documentation of the code, but I'm not so good ad C & desperately need this functionality. 

Anyone have some suggestions what files should I use/change so I can transfer binary files over serial to SD card in a simple test environment like: Arduino with usb & card reader attached to hardware spi.

Thanks for your help & sorry for my bad english!

Roxy-3D, 2016-10-10:
Are you thinking you want to use a host program such as PronterFace to send commands to Marlin, and then have Marlin store the received commands on the SD-Memory card?  

Can you explain why you want to store the commands via Marlin and not at a different place in the system?

(But, "Yes", this can be done.   It will take extra memory to open a file for writing, but it should not be that big of a deal to do. )

bsh314, 2016-10-10:
No, i only need a way to transfer binary file from pc to spi sdcard attached to arduino

Roxy-3D, 2016-10-10:
And you can't unplug the SD-Memory card and plug it into the computer to do the transfer?

thinkyhead, 2016-10-12:
Pronterface has a function to upload a file to the SD card in your printer:
[image]

Roxy-3D, 2016-10-12:
> Pronterface has a function to upload a file to the SD card in your printer:

What version of PronterFace is that?   I have PrintRun 2014.08.01  which I thought was the latest?   And maybe it is because I'm on Windows.   But I don't have that option.

thinkyhead, 2016-10-20:
@Roxy-3D It's the most recent macOS version of Pronterface. The option has been in the Mac version for a while, so it might be hidden somewhere in other platform builds.

closed 2017-10-31 by thinkyhead