MarlinFirmware/Marlin #4989: devl-ubl: Major Z movement in the middle of a flat print

voelkerp, 2016-10-10:
I downloaded devl-ubl a few days ago (v2?) to try to address my uneven print bed (8" Prusa i3.)  Have been able to auto-probe the bed and print out the test pattern with G26 to my satisfaction.

My next step was to download a bed leveling pattern, slice it and print it.  The strange thing now is that at some time in the middle of the print the z axis will just start going up and doesn't appear to stop.  I usually kill the power and reset before it gets up 100mm.

I would be surprised if the problem was my slicer (KISSlicer) since I've been using it for a really long time and haven't experienced anything like this before.

Any tips on finding out which command is causing this and/or what I can do to debug it?

Roxy-3D, 2016-10-10:
Are you using PronterFace or Repetier Host?  Are you getting any messages on the Serial Console.   The reason I'm asking is there is debug code in the new Mesh_Buffer_Line.cpp routine.   

Please see:   https://github.com/MarlinFirmware/Marlin/issues/4809#issuecomment-252704906

voelkerp, 2016-10-10:
Using PronterFace, here's what I get.  If I eyeball the nozzle, it's near X180 Y160 when it starts to skyrocket.

I've attached a file including a long snippet from the end of the log before the unusual movement and device reset.  Also a copy of my current mesh and info.

[mesh1.txt](https://github.com/MarlinFirmware/Marlin/files/520442/mesh1.txt)

[log1.txt](https://github.com/MarlinFirmware/Marlin/files/520439/log1.txt)

Roxy-3D, 2016-10-11:
I think I know what is happening...  You are going off the bed.    Your mesh is declared to be 0 x 180 on both the X & Y axis.   But here is an error message that Mesh_Buffer_Line.cpp put out:

> [INFO] SENT: N652 G1 X179.43 Y142.29 E60.6259*105
> [INFO] RECV: ok
> [INFO] SENT: N653 G1 X179.52 Y142.33 E60.6274*108
> [INFO] RECV: ok
> [INFO] SENT: N654 G1 X180.28 Y141.95 E60.6274 F30000*58
> [INFO] RECV:   x_end=180.28  y_end=141.95  z0=0.00  z_optimized=0.00  err=0.00
> [INFO] RECV: ok
> [INFO] SENT: N655 G1 X180.28 Y141.98 E60.6279 F1200*11
> [INFO] RECV: ok
> [INFO] SENT: N656 G1 X180.28 Y164.24 E60.9606*110

That error message happens when the original Mesh correction numbers are not the same as what UBL calculates as the correction:
```cpp
// debug code to use non-optimized get_z_correction() and to do a sanity check
// that the correct value is being passed to planner.buffer_line()
//
z_optimized = z0;
z0 = blm.get_z_correction( x_end, y_end);
if ( fabs(z_optimized - z0) > .01 && !isnan(z0) )  {
debug_current_and_destination("!FINAL_MOVE: z_correction()");
SERIAL_ECHOPAIRPGM("  x_end=",x_end);
SERIAL_ECHOPAIRPGM("  y_end=",y_end);
SERIAL_ECHOPAIRPGM("  z0=",z0);
SERIAL_ECHOPAIRPGM("  z_optimized=",z_optimized);
SERIAL_ECHOPAIRPGM("  err=",fabs(z_optimized-z0));
SERIAL_ECHO("\n");
}
```
But the next few lines were supposed to stop the NAN (Not a Number) problem from happening:
```cpp
		z0 = z0 * blm.fade_scaling_factor_for_Z( z_end );

		if (isnan(z0)) {	// if part of the Mesh is undefined, it will show up as NAN
			z0 = 0.0;	// in z_values[][] and propagate through the 
					// calculations. If our correction is NAN, we throw it out 
					// because part of the Mesh is undefined and we don't have the 
					// information we need to complete the height correction.
		}
```
So...  I'll have to get that error checking more 'Rock Solid'.   But in the mean time, you can avoid the problem by just printing on the bed and not going past the boundaries.

voelkerp, 2016-10-11:
Ah.  Ok.  I'll avoid the mesh boundaries for now.  Thanks!

adamfilip, 2016-10-15:
Can the area outside the mesh boundaries be assumed to be the same as the closest mesh point.. so if it does go off or is very close it wont do something dramatic.. i had this happen once. when i was doing a test print that was the same size as the bed size 300x 280Y, it drove the nozzle into the bed. luckily i have a physical limit which prevents damage :) but it always a scramble for the power switch.

Roxy-3D, 2016-10-15:
> Can the area outside the mesh boundaries be assumed to be the same as the closest mesh point.. 

Yes.  I think this can be done.   However, right now it doesn't do that.  Instead, it calculates the needed correction, and if un-populated areas of the mesh enter into the calculation, NAN (Not a Number) gets propagated to the final correction amount.   If this happens, the correction gets set to 0.0    

I guess it would be possible to instead set the correction amount to what ever the nearest Mesh Point is.   That shouldn't be very difficult to do.   I'm more concerned about "Is that the 'right' thing to do?"    My guess is that approach is 'more correct' than just setting the correction to 0.0

adamfilip, 2016-10-15:
interesting cause when it does dive the head down.. its much more then going to 0
my bed only deviates +/- 1mm and it tried to dive what looked like in the 10mm range

thinkyhead, 2016-10-22:
> Yes. I think this can be done.

In fact, this is how the `bilinear_z_offset` works in `RCBugFix` since that code was made available for all. Since you can probe any small area of the bed, when you go to areas outside the mesh, it uses the Z correction based on the nearest edge.

closed 2017-10-31 by thinkyhead