MarlinFirmware/Marlin #4990: [question] Marlin for some different device without heaters

HWiese1980, 2016-10-10:
I'm using Marlin for a device that I built on my own that works basically like a 3D printer. It only has 2 cartesian axes (X/Y) plus 2 RC servo motors for 2 additional degrees of freedom. No heaters are involved, just two steppers and two servo motors and a LED strip instead of the extrusion fan. 

Now, I would like to know if I can somehow easily disable everything related to heating like any safety measure that prevents thermal runaway (impossible without a heater), cold extrusion (no extruder), repeated temperature feedback (M105) and stuff like that. 

Can this be done by means of the `configuration.h` and/or `configuration_adv.h` only or would I have to dig deeper into the code?

Thanks

Blue-Marlin, 2016-10-10:
There is no single point where you can switch off all heater related code.
My best guess for the smallest compiling configuration is about:
#define EXTRUDERS 1
#define TEMP_SENSOR_0 999
//#define PIDTEMP
//#define PREVENT_COLD_EXTRUSION
//#define PREVENT_LENGTHY_EXTRUDE
//#define THERMAL_PROTECTION_HOTENDS // Enable thermal protection for all extruders
//#define THERMAL_PROTECTION_BED     // Enable thermal protection for the heated bed

And just don't use M104, M105 and M109

HWiese1980, 2016-10-10:
Alright, I'll give it a shot. I don't need M104, M105 and M109 anyway.

thinkyhead, 2016-10-22:
Someday we might make `EXTRUDERS 0` a legal option. But yeah, for now you pretty much have to strip out the code you don't want.

closed 2016-10-22 by thinkyhead