MarlinFirmware/Marlin #4992: #define Z_RAISE_PROBE_DEPLOY_STOW ???

spilz87, 2016-10-10:
hello,
sorry if someone has already report this issue, I don't find it.

i try to compile Marlin-1.1.0-RC7 using Z Servo Probe, such as an endstop switch on a rotating arm.
(actually I don't know how to declare it, I just did : //#define FIX_MOUNTED_PROBE)
but I get :

`SanityCheck.cpp:338: error: #error "You must set Z_RAISE_PROBE_DEPLOY_STOW in your configuration."
     #error "You must set Z_RAISE_PROBE_DEPLOY_STOW in your configuration."`

But I don't fine where / how can I declare Z_RAISE_PROBE_DEPLOY_STOW.

thanks for your help

regards

Blue-Marlin, 2016-10-10:
Two possibillitys:
* Use the search function of your editor in `Configuration.h`
I that does not help:
* Don't copy old configuration files into your new Marlin.

spilz87, 2016-10-10:
I did search but it's only fund in SanityCheck.cpp
I never use old configuration files, I check what I have to change each time I download a new version

Blue-Marlin, 2016-10-10:
How about reading the error message?
>#elif defined(Z_RAISE_PROBE_DEPLOY_STOW) || defined(Z_RAISE_BETWEEN_PROBINGS)
  #error "Z_RAISE_PROBE_DEPLOY_STOW and Z_RAISE_BETWEEN_PROBINGS are now Z_CLEARANCE_DEPLOY_PROBE and Z_CLEARANCE_BETWEEN_PROBES. Please update your configuration."

spilz87, 2016-10-10:
it's not the error message I got
I loading the last version to try

spilz87, 2016-10-11:
It seems ok now
Thanks

closed 2016-10-11 by spilz87