MarlinFirmware/Marlin #4996: SD init crashes on startup when SD card not present

ian2000611, 2016-10-10:
Problem occurs on multiple printers
1.1.0 RC7
MOTHERBOARD_RAMPS_13_EFB or MOTHERBOARD_RAMBO
SDSUPPORT
REPRAP_DISCOUNT_SMART_CONTROLLER

Lcd blank, unresponsive to serial after power on between boot screen and info screen. Reset button non-functional.

Problem only occurs when SDSUPPORT is enabled and the printer is turned on without a SD card present.

After info screen has displayed printer functions as normal after of SD card removal.

Worked in 1.0.2, have not tested RC6, but the diff appears to show that Sd2Card.cpp was completly rewritten/replaced between RCs.

thinkyhead, 2016-10-22:
Test with `RCBugFix` when you get a chance.
We recently patched an issue connected to SD initialization.

boelle, 2016-11-02:
@ian2000611 did you retest?

ian2000611, 2016-11-02:
Sorry, missed you email. Well test tonight. Thank you.

boelle, 2016-11-05:
@ian2000611 how did the test go?

ian2000611, 2016-11-06:
Not good, still getting a full freeze after the startup screen is cleared
before anything else is displayed when booting with out an sd card.

The printer is non responsive and the reset button is not working, and it
can not be programmed when it enters this state.  Removing power and
booting with the sd card restores normal functionality.

thinkyhead, 2016-11-06:
@ian2000611 Just to confirm: You tested with the current `RCBugFix`? Do you find that some older version of Marlin works ok without crashing?

ian2000611, 2016-11-06:
Tested with latest pull of rcbugfix as of 10/3/2016.

Have issues with RC1,5,6,7. Have not tested 2,3,4.  1.0.2-1 works

ian2000611, 2016-11-06:
Correction 11/3/2016

thinkyhead, 2016-11-06:
@Blue-Marlin Any thoughts on why the absence of an SD card might lead to a freeze? Freezing is highly unusual. Especially one that so thoroughly locks up everything, including the reset button.

@ian2000611 Is the board actually a RAMBo?

ian2000611, 2016-11-06:
Problem occurs with a RAMBo purchased.directly from Ultimachine, as well as
with a ramps on a genuine arduino.

Both printers were using Reprap Smart Discount controllers from different
manufacturers.

I have heard that if you are stuck in a loop (arduino main loop method
fails to return) that the boot loader does not see the reset button press
(it is not tied to an interrupt) but I have not confirmed this.

thinkyhead, 2016-11-06:
> I have heard that if you are stuck in a loop

Are you trying to reset using the reset button on the Arduino or the reset button on the controller?

ian2000611, 2016-11-06:
With the Rambo I am trying the reset button on the Rambo. With the ramps I
have tried both the reset button on the ramps and the arduino.

It has been a while since I looked at the schematic but they should be
electrically equivalent.

The usb should trigger the same input to the microcontroller and is is also
ineffective.

thinkyhead, 2016-11-06:
> they should be electrically equivalent

Can you confirm that your controllers' `SD_DETECT_PIN`s are connected to pin 81 or 49, and that the `SDSS` pin is connected to pin 53?

What happens if you comment out the definitions of `SDSS` and / or `SD_DETECT_PIN`? Does the machine still lock up without a card?

ian2000611, 2016-11-06:
I will check in the afternoon and get back to you. Everything is hooked up
correctly (first thing I checked), and I will double check the ramps
schematic and confirm with a multimeter as well as commenting out the sdss
and/or sd detect pins.

Gmail is hiding times zones, but it is 1:21am here and the printer is over
at the makerspace.

thinkyhead, 2016-11-06:
I hear you. 1:23am here.

ian2000611, 2016-11-06:
Schematics check out sdss should be 53, and sd detect should be 49.

I know the detect and sdss should be functioning as I can print from sd
when the printer is bored with am sd card present, and upon removal I get a
removal message on the lcd.

I suspect that there is a loop in sd initialization that is getting hung
when the sd card is not present.

Hopefully commenting out sd detect will change the behavior used to
initialize the sd card and make things work.

thinkyhead, 2016-11-07:
> I suspect that there is a loop in sd initialization

There isn't. And a loop wouldn't prevent reset.

boelle, 2016-11-11:
So whats the status on this one? @ian2000611 @thinkyhead

ian2000611, 2016-11-11:
Sorry, thought I sent this, it was in my drafts folder.

Tried commenting SD-DETECT and it had no noticeable effect.

Commenting SDSS causes it to freeze regardless of the presence of the sd
card.

boelle, 2016-11-11:
SDSS i think is a kind of "chipselect"

I just got the full graphics display and i found out that the SD Detect line is not connected all the way so it will never make any difference... for it to work for me i will have to solder on a wire from the smart connector board to pin 49. One can wonder why it was not connected all the way

ian2000611, 2016-11-12:
I recall seeing something like that in one off the conditionals files.  One
of the displays actually defines SD_DETECT to -1 and has a comment about
using a switch/button and commenting something out if you want sd detect
behavior.

boelle, 2016-11-12:
if they add a comment about using a switch its mostly because they have not wired the switch in the sd card socket.

Most have a switch in the socket to detect if a card is inserted and if the write protect is enabled. 

some older sd cards had a small tap you could slide to write protect it. 

will try and figure your controller... it was the REPRAP_DISCOUNT_SMART_CONTROLLER right?

boelle, 2016-11-12:
if i was correct... schematics for your controller:

http://reprap.org/mediawiki/images/7/70/Controller_final_reprapdiscount.pdf

pin 7 of JP3 or EXP2.. try and meassure what happens at the pin with and without a card inserted. I think it will just short to ground. which is ok as the MEGA proberly has an internal pullup so it see 5V with no card and 0V with a card... or the other way round

http://reprap.org/mediawiki/images/7/79/LCD_connect_SCHDOC.pdf

is for the small adapter board

pin7 from EXP2 goes to Pin8 on AUX3... again use the meter to verify that things changes. 

pin7 of AUX3 should be 5V constant. Pin1 of AUX3 should be ground/GND

what makes this a pain is that a lot of these schematic has a typo on the SD detect stuff. some places its labeled D49 other places like here its D43

boelle, 2016-11-12:
for the full graphics i can say it worked for me

in configuration_adv at the bottom there is //#define PINS_DEBUGGING

remove //

flash the board and issue M43

you will get a list of the pins and their state. Look at pin 49 with and without a card.... at least one of the pins should change state

ian2000611, 2016-11-12:
> reprapdiscount.pdf

I believe this is a software problem.

When the printer is powered on with an sd card it functions properly. Upon
removal I get a card removed message on the lcd. On insertion I get a card
inserted message on the lcd. I can print files from the sd card.

When the printer is powered on without an sd card, I get the Marlin 1.1.0
RC7 splash screen, a screen clear, and then a full freeze that even
prevents the reset button from functioning.

It sounds like my card reader, card, and card detection switch ate in full
working order. I have this issue on multiple printers with similar
configurations.

When I disable SD_SUPPORT and power on with out a card everything works as
expected and I can print from an attached host.

The fact that it happens when there is no sd card should mean that the sd
card speed and format have nothing to do with the issue.

boelle, 2016-11-12:
" I get the Marlin 1.1.0 RC7 splash screen"

Have you tried RCBugfix instead of RC7 ?

ian2000611, 2016-11-13:
Yes Syu today time the last update to rcbugfix had been 11/3/2016

boelle, 2016-11-14:
RC7 is just older than RCBugfix 

and if you use RCBugfix your display should not say RC7

https://github.com/MarlinFirmware/Marlin/tree/RCBugFix

But you still get the same issue with RCBugfix ?

ian2000611, 2016-11-14:
I have gotten this issue on rcbugfix, but have not checked rcbugfix newer
than 11/3/2016.

thinkyhead, 2016-11-19:
I suspect the problem still exists, but I still have no idea how to proceed. If I had physical access I could troubleshoot and try various things. I suggest doing a thorough search of the RepRap forums to see if others have encountered this issue and see if a solution exists.

boelle, 2016-11-26:
@thinkyhead @ian2000611 

maybe a google hangout session with screen share?

thinkyhead, 2016-11-26:
Sure. Meanwhile, @ian2000611 try all the things.

ian2000611, 2016-11-26:
I spoke with another user with identical hardware who is not having this
issue.

I'm going to get a copy of his firmware and look at the diff output to see
if I can identify the cause of the issue.

If the diff output does not yield any answers I will try an install of his
copy to see if it is a hardware issue.

thinkyhead, 2016-11-27:
Excellent. I look forward to your results!

boelle, 2016-12-01:
@ian2000611 any new results?

ian2000611, 2016-12-01:
I have done a test using my contacts firmware, and I'm still having an
issue.

I'm still skeptical about it being a hardware issue as an having the issue
on two different printers.

I'm going to meet up with him in the next week to see if the error is
occurring on his printer (contrary to what he said) in case he failed to
correctly understand and test the issue.

If his printer is not having issues then it must be hardware, and I will
start comparing which suppliers our hardware came from looking in to issues
that could be caused by different manufacturers of the same hardware
(resistors out of tolerance, etc).

thinkyhead, 2016-12-01:
> I'm going to meet up with him in the next week

Awesome! I look forward to your test results.

boelle, 2016-12-11:
@ian2000611 had time to see if your contacts printer also have the error?

ian2000611, 2016-12-11:
Interestingly it did not. We compare notes and hardware the only difference
was our arduino versions. I tried 1.6.11 (same version he used) but still
no luck.

I'm clueless as to what else to try.

closed 2017-10-31 by thinkyhead