MarlinFirmware/Marlin #5000: devl-ubl: Unusual speed of first manual extrusion

voelkerp, 2016-10-11:
Using devl-ubl v2, and it seems that an occasional extrusion has an unexpected feedrate.

I'm using PronterFace and can repro with only manual extrusion which is currently set to 2mm at 30mm/min.  It always seems to happen just after reset.

Repro steps are:
* Remove fillament
* Disconnect / reconnect
* Heat up nozzle to 180c (to prevent cold extrusion blocking)
* Extrude (goes really fast)
* Extrude (normal speed)

The commands and messaging in the terminal all appear to be correct.

Extruding 2.000000mm of filament.
[INFO] SENT: G91
[INFO] SENT: G1 E2 F30
[INFO] SENT: G90
[INFO] RECV: ok
[INFO] RECV: ok
[INFO] RECV: ok
Extruding 2.000000mm of filament.
[INFO] SENT: G91
[INFO] SENT: G1 E2 F30
[INFO] SENT: G90
[INFO] RECV: ok
[INFO] RECV: ok
[INFO] RECV: ok

Roxy-3D, 2016-10-11:
Thank You for all the detail and the steps to reproduce.  I can confirm that I can reduplicate it.

I really do try to make sure things are well tested and doing the right thing.  But this is a use case I never envisioned.   I pretty much always do a G28 before I do anything else.   And that keeps this weird case from happening because at that point the various feed rates are set.

I'll see if I can find a clean way to fix this.   In the mean time, if you do a G28 (or probably any X or Y movement) prior to an extruder movement it should not happen.

voelkerp, 2016-10-11:
I'm also able to reproduce the problem if I do a G28 first.  What I don't understand is why the feedrate in the manually sent command would _ever_ be ignored.

[INFO] SENT: G28
[INFO] RECV: X:81.00 Y:106.00 Z:5.00 E:0.00 Count X: 6480 Y:8480 Z:20000
[INFO] RECV: ok
Extruding 2.000000mm of filament.
[INFO] SENT: G91
[INFO] SENT: G1 E2 F30
[INFO] SENT: G90
[INFO] RECV: ok
[INFO] RECV: ok
[INFO] RECV: ok
Extruding 2.000000mm of filament.
[INFO] SENT: G91
[INFO] SENT: G1 E2 F30
[INFO] SENT: G90
[INFO] RECV: ok
[INFO] RECV: ok
[INFO] RECV: ok

closed 2017-10-31 by thinkyhead