MarlinFirmware/Marlin #5001: Viki2 with 4 extruders

tampas, 2016-10-11:
Hi,
I'm, trying to use a 4 extruder config with an Viki2 LCD.
But on the lcd i can only get info about 3 extruders + heatbed + Fan

can i remove the fan icon to arranje space for the 4th extruder?

Thanks in advance

tampas, 2016-10-13:
Update:
when i define 4 extruders, the lcd presents the temperature & target of T3 but it draws the heatbed symbol. In this case i loose the info about the heatbed.

How can i remove the fan icon to arranje space for the 4 extruders plus heatbed info, or how can i reduce the space between the icon to arranje space for the 4th extruder?

Blue-Marlin, 2016-10-13:
Its not a simple task, because the 'Extruder-graphic' is monolytic. Ether you have to extend the cases of 'status_screen0_bmp' in 'dogm_bitmaps.h' to or you can take it apart and use one graphic per extruder, like done in https://github.com/AnHardt/Marlin/pull/13 (https://github.com/AnHardt/Marlin/pull/13/commits/b775af25d4afbb6c8ba164b9ec8245a287b1fe57). 
The width for a single symbol is more limited by the need for the temperatures digits then by the symbols them selves. You can't squeeze in more than 5.

My idea to save space, so far, is to use an other scheme to displaying the devices.
If we'd divide the space in 4 places;
At the first we could display the _active_ extruder.
At the second we could let alternate the extruder heaters 0 to n.
At the third place we could let alternate bed, chamber and or room temperature.
And we could use the last place for the fan's.

tampas, 2016-10-13:
@Blue-Marlin thanks for your reply.
I've realized that the with for a single symbol is limited by the temperature digits. It will become overlap when the temperature reach a 3 digit number.

So my quick fix idea is to remove the fan's info.  I can live without it in my implementation, since the fans are fully automated to always run at full power. I don't need to regulate then, so i don't need that info on the LCD.

So i'm trying to edit ultralcd_impl_DOGM.h to remove that info, commenting the Fan Section.

> // Extruders
>   HOTEND_LOOP() _draw_heater_status(5 + e * 25, e);
> 
>   // Heated bed
>   #if HOTENDS < 4 && HAS_TEMP_BED
>     _draw_heater_status(81, -1);
>   #endif
> 
>   // Fan
>  
>   u8g.setPrintPos(104, 27);
>   #if HAS_FAN0
>     int per = ((fanSpeeds[0] + 1) * 100) / 256;
>     if (per) {
>       lcd_print(itostr3(per));
>       lcd_print('%');
>     }
>   #endif

Am i in the correct path to do this or should i edit the dogm_bitmaps.h to remove the fan icon?

Blue-Marlin, 2016-10-13:
The fan-icon is seperat. You can savely remove it - but that will not make the exruder graphic having more extruders - so you have to edit there.

closed 2017-10-31 by thinkyhead