MarlinFirmware/Marlin #5004: error: unable to find a register to spill in class 'POINTER_REGS'

MoonshineSG, 2016-10-12:
If I enable `#define PIDTEMP` I get the following error:
```
/Volumes/Data/Github/Marlin/.build/sketch/temperature.cpp: In function 'Temperature::manage_heater() [clone .part.4]':
/Volumes/Data/Github/Marlin/.build/sketch/temperature.cpp:797:1: error: unable to find a register to spill in class 'POINTER_REGS'
 }
 ^
/Volumes/Data/Github/Marlin/.build/sketch/temperature.cpp:797:1: error: this is the insn:
(insn 115 112 117 4 (set (reg:SF 128 [ D.2293 ])
        (mem:SF (post_inc:HI (reg:HI 140 [ ivtmp.114 ])) [7 MEM[base: _146, offset: 0B]+0 S4 A8])) /Volumes/Data/Github/Marlin/.build/sketch/temperature.cpp:684 100 {*movsf}
     (expr_list:REG_INC (reg:HI 140 [ ivtmp.114 ])
        (nil)))
/Volumes/Data/Github/Marlin/.build/sketch/temperature.cpp:797: confused by earlier errors, bailing out
lto-wrapper: /Applications/Arduino.app/Contents/Java/hardware/tools/avr/bin/avr-gcc returned 1 exit status
```
Without `PIDTEMP`, it compiles fine.

- Marlin RCBugFix as of  Oct 11, 2016
- configured with dual extruder 
- avr-gcc (GCC) 4.9.2

Can anyone help ?

MoonshineSG, 2016-10-12:
Did other tests:
With PID but `SINGLENOZZLE` it compiles ok. 
WIth PID and 1 extruder compiles ok.

emartinez167, 2016-10-12:
What version of Arduino compiler are you using? From memory, I had a
similar issue and got fixed using a newer version.

MoonshineSG, 2016-10-12:
avr-gcc (GCC) 4.9.2 which comes with Arduin 1.6.13 (latest - I even updated to nightly build)

- running macOS 10.12

emartinez167, 2016-10-12:
Try a different version. Which version of Marlin are you trying?

MoonshineSG, 2016-10-12:
Marlin RCBugFix as of Oct 11, 2016

I guess, I'll download older versions of Arduino...

MoonshineSG, 2016-10-12:
1.6.11 same results

emartinez167, 2016-10-12:
I have run out of ideas.... is this the latest RCBugFix?

Regards,

Ernesto

MoonshineSG, 2016-10-12:
yes, Marlin RCBugFix as of Oct 11, 2016

Thanks for trying. Maybe someone else... For now I have compiled with single extruder (until I need my second one)

Blue-Marlin, 2016-10-12:
https://github.com/MarlinFirmware/Marlin/issues?q=is%3Aissue+unable+to+find+a+register+is%3Aclosed

Looks is if you try to compile Marlin 1.0.x

thinkyhead, 2016-10-12:
#2746, #1603, #2253... The compiler sometimes gets confused. Sometimes just shuffling the code around at random solves the issue. Perhaps if you move some functions around…. This is when code-folding can come in extra handy.

MoonshineSG, 2016-10-12:
@Blue-Marlin, no I'm sure it's the right version...

@thinkyhead, well... you can't be more right... 

I had the above compile error.
I downgraded the Arduino. Recompiled. Same problem.
Restore the Arduino to latest version. . Recompiled.  Same problem.
Changed to single nozzle. Compile successfully.
Changed back dual nozzle. Compile successfully !!!! Huh ???

I just hope this will not repeat... It did show up out of the blue after many successful compiles...

thinkyhead, 2016-10-12:
Sometimes quitting / restarting Arduino IDE is enough to make it realize something needs a re-compile. Generally the combination of 2 extruders plus PIDTEMP enabled (the default) seems to be implicated. Then there were perhaps problematic characters in the file, or maybe whitespace or something. It seemed that simply changing the specific file where the compiler complained, in any way, would make the issue go away. So maybe that's all it takes…?

MoonshineSG, 2016-10-12:
I'm command line compiling so there's no restart. But it's good to know that "messing" with the source code can solve this ...

emartinez167, 2016-10-12:
It's annoying to see this erratic behavior on the compiler... but at least we know these tricks.

Regards,

Ernesto

thinkyhead, 2016-10-12:
Crossing my fingers that this is not going to become a regular thing.

closed 2016-10-12 by thinkyhead